ros2/rmw_cyclonedds #3: Talker example crashes on Dashing

juanrh, 2019-06-07:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04.2 LTS
- Installation type:
  - binaries
- Version or commit hash:
  - running Docker container for image [osrf/ros2:nightly](https://hub.docker.com/r/osrf/ros2/tags)
- DDS implementation:
  - rmw_cyclonedds
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

I'm having problems running the talker/listener example on Dashing. I'm running this on Docker using the image `osrf/ros2:nightly` that currently has ROS 2 dashing installed. I'm installing `rmw_cyclonedds_cpp` as follows:

```bash 
mkdir -p /opt/workspace/src
cd /opt/workspace

export CYCLONE_DDS_ROOT='/opt/cyclonedds'
echo "Installing Cyclone DDS at [${CYCLONE_DDS_ROOT}]"
mkdir -p "${CYCLONE_DDS_ROOT}"
mkdir -p "${CYCLONE_DDS_ROOT}-src"
pushd "${CYCLONE_DDS_ROOT}-src"
git clone https://github.com/eclipse-cyclonedds/cyclonedds.git

apt-get install maven default-jdk -y
mkdir cyclonedds/build
pushd cyclonedds/build
cmake -DCMAKE_INSTALL_PREFIX="${CYCLONE_DDS_ROOT}" ../src/
cmake --build .
cmake --build . --target install
popd
popd

export LD_LIBRARY_PATH=${LD_LIBRARY_PATH}:${CYCLONE_DDS_ROOT}/lib

pushd src
git clone https://github.com/atolab/rmw_cyclonedds.git
popd
rosdep update && rosdep install --from-paths src/rmw_cyclonedds --ignore-src -r -y
colcon build --packages-up-to rmw_cyclonedds_cpp --merge-install
```

The program shows an error message when I close the program with Ctrl+C:

```bash
source /opt/ros/dashing/setup.bash
source /opt/workspace/install/local_setup.bash
CYCLONE_DDS_ROOT='/opt/cyclonedds'
export LD_LIBRARY_PATH=${LD_LIBRARY_PATH}:${CYCLONE_DDS_ROOT}/lib

RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ros2 run demo_nodes_cpp talker
[INFO] [talker]: Publishing: 'Hello World: 1'
...
[INFO] [talker]: Publishing: 'Hello World: 16'
^C[INFO] [rclcpp]: signal_handler(signal_value=2)
[ERROR] [rmw_cyclonedds_cpp]: failed to trigger graph guard condition

# In another terminal
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ros2 run demo_nodes_py listener
[INFO] [listener]: I heard: [Hello World: 1]
...
KeyboardInterrupt
[ERROR] [rmw_cyclonedds_cpp]: failed to trigger graph guard condition
^C
[... 450 lines cut ...]
==16748==    by 0x1136ED: __shared_count<Talker, std::allocator<Talker>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (shared_ptr_base.h:637)
==16748==    by 0x1136ED: std::__shared_ptr<Talker, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<Talker>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&>(std::_Sp_make_shared_tag, std::allocator<Talker> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) (shared_ptr_base.h:1295)
==16748==    by 0x10E5CB: shared_ptr<std::allocator<Talker>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (shared_ptr.h:344)
==16748==    by 0x10E5CB: allocate_shared<Talker, std::allocator<Talker>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (shared_ptr.h:691)
==16748==    by 0x10E5CB: make_shared<Talker, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (shared_ptr.h:707)
==16748==    by 0x10E5CB: main (talker.cpp:98)
==16748== 
==16748== 
==16748== HEAP SUMMARY:
==16748==     in use at exit: 3,337,702 bytes in 846 blocks
==16748==   total heap usage: 3,493 allocs, 2,647 frees, 4,102,839 bytes allocated
==16748== 
==16748== LEAK SUMMARY:
==16748==    definitely lost: 736 bytes in 22 blocks
==16748==    indirectly lost: 1,033 bytes in 28 blocks
==16748==      possibly lost: 20,830 bytes in 22 blocks
==16748==    still reachable: 3,315,103 bytes in 774 blocks
==16748==         suppressed: 0 bytes in 0 blocks
==16748== Rerun with --leak-check=full to see details of leaked memory
==16748== 
==16748== For counts of detected and suppressed errors, rerun with: -v
==16748== Use --track-origins=yes to see where uninitialised values come from
==16748== ERROR SUMMARY: 7 errors from 4 contexts (suppressed: 0 from 0)
root@ub80c134adc2254:/opt/workspace# 
```

raghaprasad, 2019-06-07:
Adding to the above information: 
If you modify the talker listener slightly to shut down after sending and receiving X messages, we see the following error: 

`[ERROR] [rmw_cyclonedds_cpp]: failed to trigger graph guard condition`

eboasson, 2019-06-07:
Hi @raghaprasad, thanks for reporting these problems and investing a bit further.

I've found & fixed the root cause of the "failed to trigger graph guard condition" on stopping already. Just a stupid mistake in ordering of cleanup operations (the graph guard got destroyed before some readers that could trigger it were destroyed). What puzzles me is why I hadn't seen it, because it is self-evidently 100% reproducible ...

I'm looking at the "uninitialised bytes" report — my experience is that this is pretty much always padding, but I do want to be sure. If it takes too long, I'll sort this one out first:
```
==16717== Thread 1:
==16717== Invalid write of size 8
==16717==    at 0x4C367E3: memmove (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==16717==    by 0x58F3FB3: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::operator=(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&) (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25)
==16717==    by 0x9265DDC: cycdeser::deserialize(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) (in /opt/workspace/install/lib/librmw_cyclonedds_cpp.so)
==16717==    by 0x926596E: cycdeser::operator>>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) (in /opt/workspace/install/lib/librmw_cyclonedds_cpp.so)
==16717==    by 0x927387F: void rmw_cyclonedds_cpp::deserialize_field<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(rosidl_typesupport_introspection_cpp::MessageMember const*, void*, cycdeser&, bool) (in /opt/workspace/install/lib/librmw_cyclonedds_cpp.so)
==16717==    by 0x926CAF4: rmw_cyclonedds_cpp::TypeSupport<rosidl_typesupport_introspection_cpp::MessageMembers>::deserializeROSmessage(cycdeser&, rosidl_typesupport_introspection_cpp::MessageMembers const*, void*, bool) (in /opt/workspace/install/lib/librmw_cyclonedds_cpp.so)
==16717==    by 0x926CC18: rmw_cyclonedds_cpp::TypeSupport<rosidl_typesupport_introspection_cpp::MessageMembers>::deserializeROSmessage(cycdeser&, rosidl_typesupport_introspection_cpp::MessageMembers const*, void*, bool) (in /opt/workspace/install/lib/librmw_cyclonedds_cpp.so)
==16717==    by 0x9268E62: rmw_cyclonedds_cpp::TypeSupport<rosidl_typesupport_introspection_cpp::MessageMembers>::deserializeROSmessage(cycdeser&, void*, std::function<void (cycdeser&)>) (in /opt/workspace/install/lib/librmw_cyclonedds_cpp.so)
==16717==    by 0x92B2CF3: serdata_rmw_to_sample(ddsi_serdata const*, void*, void**, void*) (in /opt/workspace/install/lib/librmw_cyclonedds_cpp.so)
==16717==    by 0x955CA2A: ddsi_serdata_to_sample (ddsi_serdata.h:187)
==16717==    by 0x955CA2A: dds_rhc_take_w_qminv (dds_rhc.c:1929)
==16717==    by 0x955CA2A: dds_rhc_take (dds_rhc.c:2584)
==16717==    by 0x9564728: dds_read_impl (dds_read.c:180)
==16717==    by 0x9564D9F: dds_take (dds_read.c:489)
==16717==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
```

That crash is in the custom serializer from a ROS2 messages to Cyclone sample. I must have made a mistake in reusing/modifying/reverse-engineering the dynamic serialiser mode of FastRTPS ...

eboasson, 2019-06-07:
I've pushed fixes for both the guard condition issue & the deserializer crash.

The crash — the one I managed to occasionally produce and that has the exact same stack trace anyway — is a crash in deserializing a parameter_events message that only arrives rarely in my Ubuntu VM. It tried to deserialise into an uninitialized std::string object.

As mentioned in passing I am heavily indebted to the dynamic RMW layer of FastRTPS for the serialization and deserialization of ROS messages and the fix brings the code inline with the one in the FastRTPS RMW implementation, where the same fix went in just around the time I originally forked that bit of code ...

The "uninitialized bytes" warnings are all benign (padding bytes). I will fix Cyclone to avoid them.

Hopefully this will give you better results ...

raghaprasad, 2019-06-07:
@eboasson Thanks for the prompt fix! 
We are able to confirm that the crashes are no longer happening and the graph errors are not visible either!  We did notice some issues with the discovery process, I will open a new issue shortly with details. You can mark this issue as resolved.

closed 2019-06-10 by eboasson