ros2/rmw_cyclonedds #11: Windows 10 Compilation Fails

allenh1, 2019-06-17:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - source
- Version or commit hash:
  - master
- DDS implementation:
  - rmw_cyclonedds
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Try to build on Windows 10 with Cyclone DDS installed.

#### Expected behavior

Successful compilation and the ability to be used in runtime.

#### Actual behavior

Compilation fails

---

Note: I'm working on some patches for you, I'll file them tonight.

One issue is with `__attribute__ ((unused))` which isn't supported on Windows 10. Since the `CMakeLists.txt` file specifies C++14, I've changed these instances to `static_cast<void>(var)`. In C++17, one could use `[[maybe unused]]`, but I'm only nothing this for completeness.

I'll see if I can figure out what else is blocking the windows compilation as well, but that might take me a bit longer ;).

Glad to see a new middleware appear! Pretty exciting stuff!

eboasson, 2019-06-17:
Hi @allenh1 , I am a bit shocked to discover that I forgot to mention I had never even tried building it on Windows on account of not having a Windows box available. Surely I should have warned everyone ... (It was developed on macOS and checked on Ubuntu 18.04.)

Anyway, I am grateful that you're looking into it and have started fixing things. I browsed the ROS forums a bit, and it appears that the target really should be C++14. Casting to void would then indeed be the appropriate solution.

I'm curious to see what else you ran/run into ... and if there's anything I can do to help fix it ...

dennis-adlink, 2019-09-05:
This issue will be solved by PR [https://github.com/eclipse-cyclonedds/cyclonedds/pull/248](url) on CycloneDDS. With these fixes included I can successfully run the ROS2 examples using rmw_cyclonedds.

allenh1, 2019-09-05:
> This issue will be solved by PR https://github.com/eclipse-cyclonedds/cyclonedds/pull/248 on CycloneDDS

That's great! There seemed to be some more core adjustments to be made, glad this is all worked out.

Looking forward to trying out the new middleware.

joespeed, 2019-09-10:
@allenh1 @dennis-adlink @eboasson so this is fixed? thanks!

eboasson, 2019-09-11:
Based on @dennis-adlink's comment and my merging his PR on Cyclone, I'm presuming this is fixed (I don't have Windows to try it out myself).

closed 2019-09-11 by eboasson

allenh1, 2019-09-13:
@eboasson I'll give it a test when I get some time to. I'll post back definitively either way.