ros2/rmw_cyclonedds #18: Need colcon hook to setup CYCLONEDDS_URI envvar

rotu, 2019-07-12:
CycloneDDS provides a `cyclonedds.xml` file, but it is unused by default. You need to manually `export CYCLONEDDS_URI=file:///opt/ros/master/install/etc/CycloneDDS/cyclonedds.xml` (or wherever your `cyclonedds.xml` winds up).

See `opensplice_cmake_module` for how to do this: https://github.com/RoverRobotics-forks/rosidl_typesupport_opensplice/blob/master/opensplice_cmake_module/CMakeLists.txt

eboasson, 2019-07-15:
Thanks!

I'm about to start preparing the READMEs and the source trees to play nice with ROS2 installations and so getting some information on how to do these things is very valuable. It's not like I know the ROS2 build system and CI system well — I simply took the FastRTPS files as a starting point ...

That said, it was a deliberate in Cyclone DDS not to read any configuration files from the file system unless you tell it to. There is (at least in principle) project-dependent information in a configuration file (certainly the network partitions, asynchronous publisher configurations in the not-too-distant future) and so always reading from a standard location seems unwise. It seems to me that that argument would hold equally for a ROS2 installation.

I'd be interested in knowing what in your view would fit best with a workflow of using ROS2. To set CYCLONEDDS_URI by default via the build system, or not?

Or would the need for it disappear when some of the defaults were improved? Like multicast options that need to be set differently for WiFi than for Ethernet? Because then I would prefer to fix the defaults.

rotu, 2019-07-15:
I expect that if there's a default config in the source tree, then it will be respected.
Yes, I think the default value of <allowMulticast

rotu, 2019-07-15:
Oops. didn't mean to close the issue.

Installing `cyclonedds` creates such a config file in `<install root>/etc/cyclonedds`, and it would be best to either respect this file in `rmw_cyclonedds` or not install it at all.

I'm not sure how likely it is to need to change this file. Whatever you do, yes, the default AllowMulticast should also be changed to spdp.

eboasson, 2019-07-16:
> Oops. didn't mean to close the issue.

no worries.

> Installing cyclonedds creates such a config file in <install root>/etc/cyclonedds, and it would be best to either respect this file in rmw_cyclonedds or not install it at all.

I do agree with you there. My preference is to not install it at all. And in addition to removing it from the installation, I'd better add information on setting that variable and to the readme, along with some information on commonly required settings.

> I'm not sure how likely it is to need to change this file. Whatever you do, yes, the default AllowMulticast should also be changed to spdp.

Changing it to "spdp" has the problem that it would get me angry users that rely on wired networks. I think a reasonable compromise would be to check whether the network interface is actually a wireless network or not — something that is surprisingly complicated, but not infeasible.

That would mean a mixed network would default to using multicast even though it'd be a bad choice. Perhaps it would be possible to have a default that would fall back to unicast if the multicast packet loss rate is very high, but that'll probably have to wait a bit.

rotu, 2019-07-16:
Angry users? Why? I thought multicast is just a performance optimization and the downside of disabling it is just more traffic, which is hardly noticeable on low-bandwidth topics (basically everything but video). It might even work to use multicast only for unreliable topics (which don’t need a per-client acknowledgment).

eboasson, 2019-07-16:
Well, perhaps I exaggerated a bit with the "angry" and quite possibly it is caused in part by a background in different application fields. What I have seen quite often is systems where there is a significant fan-out. If having dozens of nodes subscribe to the same data and that gets sent at reasonably high rates, multicast is a very attractive option.

Cyclone even has some special support for mapping multicast addresses to the exact set of subscribed machines to support scenarios where you have N sources to M sinks and where all the links are saturated (UDP/IPv4 only). Addresses of the form BASE-MC-ADDR;START;COUNT;IDX automatically expand to large sets of multicast addresses to subscribe to, all derived from BASE-MC-ADDR. START gives a range of COUNT bits in which each subscriber is represented by a single bit, and IDX is the relative bit position in that set for the host. That's a bit of an extreme use case, but it works 😁

Anyway, I dug around in some header files and documentation and I think I have found a way to reliably determine whether an interface is a WiFi interface or not, and I expect that to be a pragmatic solution. PR is https://github.com/eclipse-cyclonedds/cyclonedds/pull/222.

rotu, 2019-08-29:
@eboasson closing since #222 is merged in

closed 2019-08-29 by rotu