ros2/rmw_cyclonedds #26: Following README instructions but rmw_cyclonedds doesn't build

mkhansen-intel, 2019-08-26:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

eboasson, 2019-08-26:
@mkhansen-intel, thanks for reporting this: I get exactly the same when I start with a completely clean slate. I do believe I tried it on a clean machine before I added it to the readme, so something subtle must have changed since then ...

eboasson, 2019-08-26:
@mkhansen-intel, what appeared to be an innocuous change in Cyclone DDS (last Friday), replacing a CMake variable by its value in setting the project name, changed the name under which it is known in the ROS2 build system, and that breaks the build.

Before that change, the RMW layer had to (and did) depend on ``cyclonedds``, now it needs to refer to ``CycloneDDS``. This incidentally also broke the [nightly build at OpenRobotics](http://build.ros2.org/view/Eci/job/Eci__nightly-cyclonedds_ubuntu_bionic_amd64/), which was puzzling me until this morning.

Changing the dependency fixes the problem locally for me, it fixed that build, and so presumably it fixes the problem for you, too. Instructions remain unchanged, just pull the updated repositories (both please, because of a fix in a Java-less build).

Re:
> WARNING: An illegal reflective access operation has occurred
> WARNING: Illegal reflective access by com.google.inject.internal.cglib.core.$ReflectUtils$1 (file:/usr/share/maven/lib/guice.jar) to method java.lang.ClassLoer.defineClass(java.lang.String,byte[],int,int,java.security.ProtectionDomain)
> WARNING: Please consider reporting this to the maintainers of com.google.inject.internal.cglib.core.$ReflectUtils$1
> WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
> WARNING: All illegal access operations will be denied in a future release

That's a well-known problem with Maven, but as it output shows, it is supposed to be only a warning. Perhaps you have a newer JDK that does turn it into an error? Because for me it doesn't cause any problems.

We're working on eliminating the Java dependency altogether, but that's not done yet. What is possible is building Cyclone without the Java components. For stand-alone use of Cyclone, that is of little value because it removes the IDL preprocessor, but the RMW layer doesn't rely on that preprocessor and instead uses ROS2's introspection features to dynamically serialize whatever type it is given. So for use with ROS2 it is a perfectly sensible thing to do.

If I knew how to elegantly disable the building of the Java-based components when building Cyclone as part of ROS2, I'd already done that. As it is, all I can suggest is editing ``cyclonedds/CMakeLists.txt`` and setting ``BUILD_IDLC`` and ``BUILD_CONFTOOL`` to ``OFF`` (that's lines 28 and 29).

eboasson, 2019-08-26:
Could you please try throwing away the build directory? CMake tends to cache a lot of such settings, so it is quite possible that it simply didn’t pick up the changed value.