ros2/rmw_cyclonedds #61: Setting a too-high domain ID results in an opaque error message

rotu, 2019-10-31:
When you try to run CycloneDDS with a too-high domain id, you get an uninformative error message. Setting the domain id to a negative number results in a saner message:

`ROS_DOMAIN_ID=55000 ros2 topic list`
```
Traceback (most recent call last):
  File "/opt/ros/master/install/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.8.3', 'console_scripts', 'ros2')()
  File "/opt/ros/master/install/lib/python3.6/site-packages/ros2cli/cli.py", line 69, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/master/install/lib/python3.6/site-packages/ros2topic/command/topic.py", line 43, in main
    return extension.main(args=args)
  File "/opt/ros/master/install/lib/python3.6/site-packages/ros2topic/verb/list.py", line 38, in main
    with NodeStrategy(args) as node:
  File "/opt/ros/master/install/lib/python3.6/site-packages/ros2cli/node/strategy.py", line 26, in __init__
    if is_daemon_running(args):
  File "/opt/ros/master/install/lib/python3.6/site-packages/ros2cli/node/daemon.py", line 36, in is_daemon_running
    s.bind(addr)
OverflowError: getsockaddrarg: port must be 0-65535.
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 109, in apport_excepthook
    pr.add_proc_info(extraenv=['PYTHONPATH', 'PYTHONHOME'])
  File "/usr/lib/python3/dist-packages/apport/report.py", line 543, in add_proc_info
    self.add_proc_environ(pid, extraenv)
  File "/usr/lib/python3/dist-packages/apport/report.py", line 610, in add_proc_environ
    env = _read_file('environ', dir_fd=proc_pid_fd).replace('\n', '\\n')
  File "/usr/lib/python3/dist-packages/apport/report.py", line 73, in _read_file
    with open(path, 'rb', opener=lambda path, mode: os.open(path, mode, dir_fd=dir_fd)) as fd:
  File "/usr/lib/python3/dist-packages/apport/report.py", line 73, in <lambda>
    with open(path, 'rb', opener=lambda path, mode: os.open(path, mode, dir_fd=dir_fd)) as fd:
TypeError: argument should be integer or None, not list

Original exception was:
Traceback (most recent call last):
  File "/opt/ros/master/install/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.8.3', 'console_scripts', 'ros2')()
  File "/opt/ros/master/install/lib/python3.6/site-packages/ros2cli/cli.py", line 69, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/master/install/lib/python3.6/site-packages/ros2topic/command/topic.py", line 43, in main
    return extension.main(args=args)
  File "/opt/ros/master/install/lib/python3.6/site-packages/ros2topic/verb/list.py", line 38, in main
    with NodeStrategy(args) as node:
  File "/opt/ros/master/install/lib/python3.6/site-packages/ros2cli/node/strategy.py", line 26, in __init__
    if is_daemon_running(args):
  File "/opt/ros/master/install/lib/python3.6/site-packages/ros2cli/node/daemon.py", line 36, in is_daemon_running
    s.bind(addr)
OverflowError: getsockaddrarg: port must be 0-65535.
```

`ROS_DOMAIN_ID=-10 ros2 topic list`
```
[ERROR] [rmw_cyclonedds_cpp]: rmw_create_node: domain id out of range

>>> [rcutils|error_handling.c:107] rcutils_set_error_state()
This error state is being overwritten:

  'error not set, at /opt/ros/master/src/ros2/rcl/rcl/src/rcl/node.c:336'

with this new error message:

  'rcl node's rmw handle is invalid, at /opt/ros/master/src/ros2/rcl/rcl/src/rcl/node.c:485'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [rcl]: Failed to fini publisher for node: 1
Unknown error creating node: rcl node's rmw handle is invalid, at /opt/ros/master/src/ros2/rcl/rcl/src/rcl/node.c:485
```

eboasson, 2019-11-01:
@rotu good catch: Cyclone doesn't check that the port numbers it computes based on the domain id are within range. It is not as simple as the limiting the range of domain ids, because the mapping of domain id is configured (as required to be by the spec).

https://github.com/eclipse-cyclonedds/cyclonedds/issues/303

wjwwood, 2019-11-21:
@rotu @eboasson can this be closed? Thanks!

eboasson, 2019-11-21:
I'd say Cyclone behaves quite nicely, but it still isn't really pretty. I do think the remaining lack of elegance is outside the domain of the RMW layer or DDS, so I think it is ok to close this ticket.

## Option 1: C++, domain id + port mapping resulting in port numbers that are out of range on any transport
```
#  ros2_ws ROS_DOMAIN_ID=33355000 install/examples_rclcpp_minimal_publisher/lib/examples_rclcpp_minimal_publisher/publisher_member_function
1574369453.364453 [33355000]    6489469: Invalid port mapping: port number(s) of out range: multicast discovery 8338757400, multicast data 8338757401
[ERROR] [rmw_cyclonedds_cpp]: rmw_create_node: failed to create domain, error Error

>>> [rcutils|error_handling.c:107] rcutils_set_error_state()
This error state is being overwritten:

  'error not set, at /Users/erik/ros2_ws/src/ros2/rcl/rcl/src/rcl/node.c:336'

with this new error message:

  'rcl node's rmw handle is invalid, at /Users/erik/ros2_ws/src/ros2/rcl/rcl/src/rcl/node.c:485'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [rcl]: Failed to fini publisher for node: 1
libc++abi.dylib: terminating with uncaught exception of type rclcpp::exceptions::RCLError: failed to initialize rcl node: rcl node's rmw handle is invalid, at /Users/erik/ros2_ws/src/ros2/rcl/rcl/src/rcl/node.c:485
[1]    18864 abort      ROS_DOMAIN_ID=33355000 
```

## Option 2: C++, domain id + port mapping resulting in port numbers that are potentially in range for RTPS, but out of range for the transport
```
#  ros2_ws ROS_DOMAIN_ID=55000 install/examples_rclcpp_minimal_publisher/lib/examples_rclcpp_minimal_publisher/publisher_member_function 
1574369350.324392 [55000]    6487116: Failed to create discovery multicast socket for domain 55000: resulting port number (13757400) is out of range
[ERROR] [rmw_cyclonedds_cpp]: rmw_create_node: failed to create domain, error Error

>>> [rcutils|error_handling.c:107] rcutils_set_error_state()
This error state is being overwritten:

  'error not set, at /Users/erik/ros2_ws/src/ros2/rcl/rcl/src/rcl/node.c:336'

with this new error message:

  'rcl node's rmw handle is invalid, at /Users/erik/ros2_ws/src/ros2/rcl/rcl/src/rcl/node.c:485'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [rcl]: Failed to fini publisher for node: 1
libc++abi.dylib: terminating with uncaught exception of type rclcpp::exceptions::RCLError: failed to initialize rcl node: rcl node's rmw handle is invalid, at /Users/erik/ros2_ws/src/ros2/rcl/rcl/src/rcl/node.c:485
[1]    18846 abort      ROS_DOMAIN_ID=55000 
```

## Option 3: "ros2" cli — but that error isn't coming out Cyclone, that's somewhere in the Python scripts.
```
#  ros2_ws ROS_DOMAIN_ID=55000 ros2 topic list
Traceback (most recent call last):
  File "/Users/erik/ros2_ws/install/ros2cli/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli', 'console_scripts', 'ros2')()
  File "/Users/erik/ros2_ws/build/ros2cli/ros2cli/cli.py", line 69, in main
    rc = extension.main(parser=parser, args=args)
  File "/Users/erik/ros2_ws/build/ros2topic/ros2topic/command/topic.py", line 43, in main
    return extension.main(args=args)
  File "/Users/erik/ros2_ws/build/ros2topic/ros2topic/verb/list.py", line 38, in main
    with NodeStrategy(args) as node:
  File "/Users/erik/ros2_ws/build/ros2cli/ros2cli/node/strategy.py", line 26, in __init__
    if is_daemon_running(args):
  File "/Users/erik/ros2_ws/build/ros2cli/ros2cli/node/daemon.py", line 36, in is_daemon_running
    s.bind(addr)
OverflowError: getsockaddrarg: port must be 0-65535.
```

wjwwood, 2019-11-21:
Should `rmw_cyclonedds` just check that the domain id is in a valid range and give a better error in that case? I don't think we can set a limit and check it above the rmw layer right now.

eboasson, 2019-11-21:
The problem is that a domain id in DDS is simply an unsigned 32 bit int (cyclone steals 2^32-1). The only limit you run into is that the domain id is an input for the calculation of port numbers, which are also 32 bit unsigned ints in the general case, but obviously more limited for, e.g., UDP.

You can define all the constants in that mapping (the spec actually mandates that) and 0 is a valid value, so there is no practical valid range that “just works”. The best one can do is explain that things won’t work.

Cyclone now implements the new feature of DDSI 2.3 that includes the domain id (and possibly a string as well) in the discovery data, so you can now run independent domains on the same port number, e.g., by setting the “domain gain” to 0. That in itself is a valuable trick, I think.

If you really want something that usually works, then the best thing is to limit them to [0,230] (from memory), because that results in valid UDP port numbers for the default mapping of domain id to port numbers.

wjwwood, 2019-11-21:
Well, I'll close this then. If someone wants to propose a check in this package (maybe catching common errors and extending it with note that this is likely due to a domain id that's too large) or a check for `[0,230]` in a high level of the stack, then that's fine.

closed 2019-11-21 by wjwwood