ros2/rmw_cyclonedds #63: rosdep can not resolve 'cyclonedds' dependency key

ruffsl, 2019-11-06:
This currently breaks the ROS2 night docker image build. Could we ensure all keys in the package XML are resolvable from ros-distro?

https://github.com/ros2/rmw_cyclonedds/blob/0a1770e1e27eeecface3ac9524af40b6a5d29de9/rmw_cyclonedds_cpp/package.xml#L15

``` Docketfile
Reading package lists...
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
rmw_cyclonedds_cpp: Cannot locate rosdep definition for [cyclonedds]
Removing intermediate container a9fb7f22755c
```

eboasson, 2019-11-07:
I'm hoping someone with more knowledge of rosdep is having or will have a look ...

ruffsl, 2019-11-11:
Ping @cottsay , as from a recent line commit, you may be more familiar with packages involved.  
https://github.com/ros2/rmw_cyclonedds/commit/888235e96edfa9ca49a5e074407a98ed72843e65

dirk-thomas, 2019-11-11:
What ROS distro are you seeing this problem with? The package `cyclonedds` has been released into [Dashing](https://github.com/ros/rosdistro/blob/f0945633ae7233d8221b6179269a467b82e7e923/dashing/distribution.yaml#L521) as well as [Eloquent](https://github.com/ros/rosdistro/blob/f0945633ae7233d8221b6179269a467b82e7e923/eloquent/distribution.yaml#L260).

ruffsl, 2019-11-12:
Ah, I had grepped the `rosdep/base.yaml` file and then used the github search UI to look for `cyclonedds` in repo, but didn't see any results.  Didn't think to manually grep the release distribution.yaml as well. Why is github search so bad.

closed 2019-11-12 by ruffsl