ros2/rmw_cyclonedds #65: [aarch64] failed to create domain - Precondition Not Met

dirk-thomas, 2019-11-11:
The aarch64 build shows a lot of failing tests, e.g. https://ci.ros2.org/view/nightly/job/nightly_linux-aarch64_extra_rmw_release/531/testReport/(root)/projectroot/test_publisher__rmw_cyclonedds_cpp/:

```
[ERROR] [rmw_cyclonedds_cpp]: rmw_create_node: failed to create domain, error Precondition Not Met
```

dirk-thomas, 2019-11-11:
The same error seems to be happening with x64 builds on [Ubuntu](https://ci.ros2.org/view/nightly/job/nightly_linux_extra_rmw_release/524/testReport/(root)/projectroot/test_publisher__rmw_cyclonedds_cpp/) and [CentOS](https://ci.ros2.org/view/nightly/job/nightly_linux-centos_debug/225/testReport/(root)/projectroot/test_publisher__rmw_cyclonedds_cpp/).

eboasson, 2019-11-11:
And on macOS ...

The failures appear to all be related to cases where a single process creates some nodes, deletes them all, and then attempts to create new nodes. I really was asleep at the wheel when I reviewed and merged the original implementation of ``dds_create_domain`` ...

#66 fixes it for me locally (for the test cases that checked)

dirk-thomas, 2019-11-11:
Closed by #66.

closed 2019-11-11 by dirk-thomas