ros2/rmw_cyclonedds #67: Incomplete QoS Implementation

mjcarroll, 2019-11-12:
## Bug report

There are currently a family of tests that fail in CI when using `rmw_cyclonedds` related to QoS.  This is likely due to an incomplete or incorrect implementation of some of the QoS features.

The tests are:
* `rcl/test_events` (https://github.com/ros2/rcl/blob/master/rcl/test/rcl/test_events.cpp)
  * `rcl/test_pubsub_no_deadline_missed`
  * `rcl/test_pubsub_deadline_missed`
  * `rcl/test_pubsub_liveliness_kill_pub`
* `rcl/test_get_actual_qos` (https://github.com/ros2/rcl/blob/master/rcl/test/rcl/test_events.cpp)
  * `rcl/TestPublisherGetActualQos`
  * `rcl/TestSubscriptionGetActualQos`

Example Linux CI job with these tests failing: http://build.ros2.org/view/Eci/job/Eci__nightly-extra-rmw-release_ubuntu_bionic_amd64/134/testReport/

eboasson, 2019-11-12:
@mjcarroll thanks for creating the issue. You're entirely correct that these test failures have to do with incomplete QoS support in Cyclone at the moment. There is progress in supporting them, reviewing https://github.com/eclipse-cyclonedds/cyclonedds/pull/274 is pretty high on my to do list. Deadline and lifespan are definitely simpler than liveliness.

Quite separate from that, I do have my doubts about some of the "get QoS" failures, in that this:

https://github.com/ros2/rcl/blob/9034c406c2d5bc4ec291f8e1ad4303cde0184cbc/rcl/test/rcl/test_get_actual_qos.cpp#L328-L338

looks like a wrong expectation to me. The RMW time span type has a 64-bit seconds field and so it doesn't make any sense to map ∞ to ``2^31-1``. The DDS spec _does_ map it that way, and there it makes (some) sense because it started out by storing the seconds in a 32-bit signed integer. I'm guessing this is leakage of a magic constant from the DDS spec into a ROS2 test.

But I figure the more important thing is to get Cyclone to support them in the first place 🙂

mjcarroll, 2019-11-12:
Thanks @eboasson, this was mainly to have a tracking issue, since we've had these tests failing in CI for some time.  I understand that there is work happening to address it on your end.

As to the origin of that number, I would probably defer to @mm318 or @emersonknapp who did a lot of the lifting on the original implementation.

emersonknapp, 2019-11-12:
Yes, I agree that it looks like a dds spec magic number leaking into ros2.

I think the best idea would be to change that to have a defined "infinity" constant in rmw, for ros2-side qos representations, that all rmw implementations can use

ivanpauno, 2019-11-18:
Besides the incomplete QoS implementation, https://github.com/ros2/rcl/blob/master/rcl/test/rcl/test_get_actual_qos.cpp should be updated to cover `rmw_cyclonedds` correctly (there's some rmw specific conditional code in the test, not correctly covering cyclonedds).

It could be also corrected to stop failing, until the missing features are implemented.

eboasson, 2020-02-03:
The deadline, liveliness and lifespan QoS settings are now fully supported and the PRs on cyclone and the RMW layer merged into the respective repositories. All tests now pass, so it seems like this is a good time to close this issue.

closed 2020-02-03 by eboasson