ros2/rmw_cyclonedds #70: Failures on Windows

ivanpauno, 2019-11-20:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - Tried both fat archive (https://ci.ros2.org/view/packaging/job/packaging_windows/) and from source
- Version or commit hash:
  - master
- DDS implementation:
  - rmw_cyclonedds_cpp
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Run the talker listener demo on Windows.

Terminal 1:
```bat
set RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
ros2 run demo_nodes_cpp talker
```

Terminal 2:
```bat
set RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
ros2 run demo_nodes_cpp listener
```

`demo_nodes_py` can be also used.

#### Expected behavior
It should work.

#### Actual behavior
It doesn't work.

#### Additional information

Several test failed in CI on Windows:
https://ci.ros2.org/view/nightly/job/nightly_win_extra_rmw_rel/543/#showFailuresLink

After 543, all other runs of that job failed or were aborted.
I've been able to reproduce some of those test failures locally.

Fun fact:
If combine one talker/listener from rmw_fastrtps_cpp and the other from rmw_cyclonedds_cpp, it works. But if both of them are using rmw_cyclonedds_cpp, it doesn't (which I find super strange).

nuclearsandwich, 2019-11-20:
I wasn't able to reproduce this using the binary archive from https://ci.ros2.org/view/packaging/job/packaging_windows/1564/

When I did experience issues previously it was due to ros2 daemon running with a different rmw but @ivanpauno and I have discussed that offline and we're reasonably sure that's not the case here.

sloretz, 2019-11-20:
I sort of wasn't able to reproduce this using https://ci.ros2.org/view/packaging/job/packaging_windows/1565/

The first time I ran the talker, nothing happend. There was no stdout from the talker at all. I started the listener and didn't see any output from that either. I investigated whether the talker prints anything ([it does](https://github.com/ros2/demos/blob/e1453fc25f0af51abe5aa6d75f5601f4ab69395c/demo_nodes_cpp/src/topics/talker.cpp#L47)), and by the time I figured that out the talker and listener were communicating. I would guess it took a couple minutes before the talker started printing anything, and maybe a minute before the listener started. Now when I kill/restart the talker and listener the talker starts printing stuff immediately, and the talker/listener start communicating immediately.

sloretz, 2019-11-20:
Ok this is weird, I was able to reproduce the same behavior again using `demo_nodes_py`. In the above comment, I was using `demo_nodes_cpp`. I did the same thing again with `demo_nodes_py` (started talker, waited, started listener), and saw the same behavior. The talker didn't print anything for about a minute, so I started the listener. Neither printed anything for ~15 seconds, then they started communicating. However, when I restart them they start communicating instantly.

ivanpauno, 2019-11-20:
Thanks @nuclearsandwich @sloretz !

I also observed sometimes the behavior @shane is commenting (talker slow to start). But in my case, the listener never receives any message (after waiting many minutes). I tried many times in different fresh terminals.

I really don't know what is wrong in the environment I was testing.

Besides that, the failures in https://ci.ros2.org/view/nightly/job/nightly_win_extra_rmw_rel/543/#showFailuresLink should be investigated.

If nobody can reproduce this, we can close this one and open a new issue for the ci failures.

nuclearsandwich, 2019-11-20:
I have seen Windows Defender try to scan every linked library via real-time protection or "protect me in real-time" cause slow executable startup time the first time around. I think we'd need to break out Wireshark to see whether the slow communication time is related to that or discovery issues.

ivanpauno, 2019-11-22:
Closing as nobody else can reproduce the error.
Opened https://github.com/ros2/build_cop/issues/251 to follow-up in the CI failures.

closed 2019-11-22 by ivanpauno