ros2/rmw_cyclonedds #72: failing ros1_bridge tests

dirk-thomas, 2019-11-21:
See https://ci.ros2.org/view/packaging/job/packaging_linux/1675/testReport/ which is trying to run a ROS 1 talker, the [ros1_bridge](https://github.com/ros2/ros1_bridge) `dynamic_bridge`, and ROS 2 listener.

The problem for the failure seems pretty similar to ros2/rmw_fastrtps#265. The logging messages are published as a C message containing a string. The actual message exchanged between the talker and listener is a C++ message (based on the same `.msg`). I assume the first advertised topic will "lock in" the C message type and then falsely try to use the same typesupport to read the C++ message which fails with `realloc(): invalid pointer`.

eboasson, 2019-11-25:
> I assume the first advertised topic will "lock in" the C message type and then falsely try to use the same typesupport to read the C++ message which fails with realloc(): invalid pointer.

The type support is indeed cached in the topic. This is something that needs to be changed in Cyclone for mixed-language use and not very difficult (and I'm all in favour of it).

dirk-thomas, 2020-02-11:
Resolved by #96.

Linux packaging build: [image] (one unrelated CMake warning).

closed 2020-02-11 by dirk-thomas