ros2/rmw_cyclonedds #74: Not getting service responses reliably when using CycloneDDS

hidmic, 2019-12-05:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04, both AMD64 and ARM64
- Installation type:
  - Binaries and from source
- Version or commit hash:
  - b92db52bb14ec097946bfe44d2b337359dddfbab (using https://github.com/ros2/rmw_cyclonedds/commit/4089feee815a4687b30f757ab1c87bf028804c93)
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - rclcpp **only**

#### Steps to reproduce issue

1. On one terminal, run:
```console
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ros2 run demo_nodes_cpp add_two_ints_server
```

2. On another terminal, run:
```console
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ros2 run demo_nodes_cpp add_two_ints_client_async
```

#### Expected behavior
A single service call takes place and succeeds i.e. you get logs on each terminal:

1.
```console
[INFO] [add_two_ints_server]: Incoming request
a: 2 b: 3
```

2. 
```console
[INFO] [add_two_ints_client]: Result of add_two_ints: 5
```

#### Actual behavior
The client hangs with no output though the server does log so it appears to have received the request.

#### Additional information
It does not happen on `rclpy`. It also does not happen if bringing up both nodes via `launch` e.g.:

```python
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    return LaunchDescription([
        Node(
            package='demo_nodes_cpp',
            node_executable='add_two_ints_server',
            output='screen'
        ),
        Node(
            package='demo_nodes_cpp',
            node_executable='add_two_ints_client_async',
            output='screen'
        )
    ])
```

So it's likely a race.

hidmic, 2019-12-06:
I can still reproduce on `master`.

dirk-thomas, 2019-12-06:
I can confirm the problem.

My first guess is a race that the client discovers the server and sends it request but when the server receives the request and wants to publish the reply it hasn't discovered the client yet.

With a simple hack (adding a sleep of 500ms into the server callback function) the client receives the response reliable.

I guess we might something like a "wait_for_client" in the same way as we already offer a "wait_for_service".

eboasson, 2019-12-06:
> My first guess is a race that the client discovers the server and sends it request but when the server receives the request and wants to publish the reply it hasn't discovered the client yet.

Cyclone implements what the spec calls "stateful" readers and writers, which means the server's reader for the request topic won't accept any data until it has discovered the client's writer. "wait_for_service" waits until the server's reader for the request topic has been discovered, consequently the client's writer will hold on to the request message until the service's reader acknowledges it. It is therefore pretty much impossible that service is unaware of the existence of the client.

Reader and writer discovery do run independently in RTPS and so it is technically possible that the server has discovered the client's request writer and handled the request without having discovered the client's response reader yet, but that does seem unlikely.

I haven't had a chance to try to reproduce it yet (and this weekend it's not likely to happen either because of all kinds of obligations) but if I were to try, I'd probably enable Cyclone's tracing to get some more insight in what is going on. (Not for the faint of heart, allegedly … CYCLONEDDS_URI='<Tr><V>finest</><Out>cdds.log.${CYCLONEDDS_PID}' suffices. You have to squint a bit to figure out how that relates to proper XML 🙂.)

hidmic, 2019-12-12:
> Reader and writer discovery do run independently in RTPS and so it is technically possible that the server has discovered the client's request writer and handled the request without having discovered the client's response reader yet, but that does seem unlikely.

I'm no expert here. Having said that, I _think_  that may be the case @eboasson.

The server sets up a response writer and a request reader first:

```
1576167310.543602 [0] add_two_in: new_writer(guid 8c561001:dc2ce137:5055b4d2:1303, (default).rr/add_two_intsReply/example_interfaces::srv::dds_::AddTwoInts_Response_)
1576167310.543645 [0] add_two_in: ref_participant(8c561001:dc2ce137:5055b4d2:1c1 @ 0x559c5a20f930 <- 8c561001:dc2ce137:5055b4d2:1303 @ 0x559c5a24ac84) user 20 builtin 15
1576167310.543658 [0] add_two_in: WRITER 8c561001:dc2ce137:5055b4d2:1303 QOS={topic=rr/add_two_intsReply,type=example_interfaces::srv::dds_::AddTwoInts_Response_,presentation=0:0:0,partition={},group_data=0<>,topic_data=0<>,durability=0,durability_service=0.000000000:{0:1}:{-1:-1:-1},deadline=9223372036.854775807,latency_budget=0.000000000,liveliness=0:9223372036.854775807,reliability=1:1.000000000,destination_order=0,history=1:-1,resource_limits=-1:-1:-1,transport_priority=0,lifespan=9223372036.854775807,user_data=0<>,ownership=0,ownership_strength=0,writer_data_lifecycle={1},entity_factory=1,ignorelocal=0}
1576167310.543665 [0] add_two_in: write_sample 0:0:0:3c2 #9: ST0 DCPSPublication/org::eclipse::cyclonedds::builtin::DCPSPublication:(blob)
1576167310.543668 [0] add_two_in: reader 8c561001:dc2ce137:5055b4d2:307
1576167310.543673 [0] add_two_in: match_writer_with_proxy_readers(wr 8c561001:dc2ce137:5055b4d2:1303) scanning all proxy readers
1576167310.543680 [0] add_two_in: match_writer_with_readers(wr 8c561001:dc2ce137:5055b4d2:1303) scanning all readers
1576167310.543687 [0] add_two_in: sedp: write for 8c561001:dc2ce137:5055b4d2:1303 via 8c561001:dc2ce137:5055b4d2:3c2
1576167310.543692 [0] add_two_in: write_sample 8c561001:dc2ce137:5055b4d2:3c2 #9: ST0 (null)/(null):(plist)
1576167310.543699 [0] add_two_in: new_reader(guid 8c561001:dc2ce137:5055b4d2:1404, (default).rq/add_two_intsRequest/example_interfaces::srv::dds_::AddTwoInts_Request_)
1576167310.543702 [0] add_two_in: ref_participant(8c561001:dc2ce137:5055b4d2:1c1 @ 0x559c5a20f930 <- 8c561001:dc2ce137:5055b4d2:1404 @ 0x559c5a24bd74) user 21 builtin 15
1576167310.543709 [0] add_two_in: READER 8c561001:dc2ce137:5055b4d2:1404 QOS={topic=rq/add_two_intsRequest,type=example_interfaces::srv::dds_::AddTwoInts_Request_,presentation=0:0:0,partition={},group_data=0<>,topic_data=0<>,durability=0,deadline=9223372036.854775807,latency_budget=0.000000000,liveliness=0:9223372036.854775807,reliability=1:1.000000000,destination_order=0,history=1:-1,resource_limits=-1:-1:-1,transport_priority=0,user_data=0<>,ownership=0,time_based_filter=0.000000000,reader_data_lifecycle=9223372036.854775807:9223372036.854775807,reader_lifespan={0,9223372036.854775807},subscription_keys={0,{}},entity_factory=1,ignorelocal=0}
1576167310.543714 [0] add_two_in: write_sample 0:0:0:4c2 #11: ST0 DCPSSubscription/org::eclipse::cyclonedds::builtin::DCPSSubscription:(blob)
1576167310.543716 [0] add_two_in: reader 8c561001:dc2ce137:5055b4d2:207
1576167310.543719 [0] add_two_in: match_reader_with_proxy_writers(rd 8c561001:dc2ce137:5055b4d2:1404) scanning all proxy writers
1576167310.543721 [0] add_two_in: match_reader_with_writers(rd 8c561001:dc2ce137:5055b4d2:1404) scanning all writers
1576167310.543727 [0] add_two_in: sedp: write for 8c561001:dc2ce137:5055b4d2:1404 via 8c561001:dc2ce137:5055b4d2:4c2
1576167310.543730 [0] add_two_in: write_sample 8c561001:dc2ce137:5055b4d2:4c2 #8: ST0 (null)/(null):(plist)
``` 

When the client gets brought up, it discovers the server in full:

```
1576167362.690709 [0] dq.builtin: write_sample 0:0:0:3c2 #9: ST0 DCPSPublication/org::eclipse::cyclonedds::builtin::DCPSPublication:(blob)
1576167362.690712 [0] dq.builtin: reader 1481001:ddd08e4f:524471cd:307
1576167362.690717 [0] dq.builtin: match_proxy_writer_with_readers(pwr 8c561001:dc2ce137:5055b4d2:1103) scanning all readers
1576167362.690722 [0] add_two_in: new_writer(guid 1481001:ddd08e4f:524471cd:503, (default).rr/add_two_ints_client/get_parametersReply/rcl_interfaces::srv::dds_::GetParameters_Response_)
1576167362.690728 [0] dq.builtin: SEDP ST0 8c561001:dc2ce137:5055b4d2:1303 reliable volatile writer: (default).rr/add_two_intsReply/example_interfaces::srv::dds_::AddTwoInts_Response_ NEW (as udp/239.255.0.1:7401 udp/192.168.9.2:38672 ssm=0) QOS={topic=rr/add_two_intsReply,type=example_interfaces::srv::dds_::AddTwoInts_Response_,presentation=0:0:0,partition={},group_data=0<>,topic_data=0<>,durability=0,durability_service=0.000000000:{0:1}:{-1:-1:-1},deadline=9223372036.854775807,latency_budget=0.000000000,liveliness=0:9223372036.854775807,reliability=1:1.000000000,destination_order=0,history=1:-1,resource_limits=-1:-1:-1,transport_priority=0,lifespan=9223372036.854775807,user_data=0<>,ownership=0,ownership_strength=0,writer_data_lifecycle={1},entity_factory=1,ignorelocal=0}
...
1576167362.690954 [0] dq.builtin: write_sample 0:0:0:4c2 #11: ST0 DCPSSubscription/org::eclipse::cyclonedds::builtin::DCPSSubscription:(blob)
1576167362.690957 [0] dq.builtin: reader 1481001:ddd08e4f:524471cd:207
1576167362.690961 [0] dq.builtin: match_proxy_reader_with_writers(prd 8c561001:dc2ce137:5055b4d2:1204) scanning all writers
[... cut ...]
1576167362.699742 [0] dq.builtin:   loc  1 = udp/239.255.0.1:7401            1 { 0 }
1576167362.699744 [0] dq.builtin:   best = 0
1576167362.699746 [0] dq.builtin:   simple udp/192.168.9.2:37316
1576167362.699749 [0] dq.builtin: rebuild_writer_addrset(8c561001:dc2ce137:5055b4d2:1303): udp/192.168.9.2:37316
```

I've updated both logs [here](https://gist.github.com/hidmic/f40abf773781fd4cc6893e3b9dd3407c). Mind to check if my logic is correct? Maybe I'm missing something in the midst of the ack/heartbeat dance.

eboasson, 2019-12-12:
@hidmic your analysis is entirely correct. I'm a bit surprised that the dance works out this way, but it is definitely what happens.

@dirk-thomas's "wait_for_client" would solve this, but it is a bit of tricky operation in that it must gracefully deal with the client disappearing while the service is waiting. What I would find more elegant is for the client to first wait for the server to have discovered the response reader, and only then send the request. That way the server will never have to wait and is still certain that the response will go out properly.

I think that that is possible in principle: guarantee that the endpoint discovery data doesn't get acknowledged until after it has been processed completely, and then offer a way of doing "wait_for_acknowledgements" on the endpoint discovery writers. But I don't dare predict when that'll be available ...

hidmic, 2019-12-17:
> What I would find more elegant is for the client to first wait for the server to have discovered the response reader, and only then send the request.

IMHO `client.wait_for_service()` success should imply that both service server and client have discovered each other or, in other words, that a request/response exchange may occur without risking either of the two getting lost. So I agree in principle with your idea @eboasson.  

> But I don't dare predict when that'll be available ...

Services are a core functionality in ROS 2. Meaning, I personally think we need a fix for this ASAP.

hidmic, 2020-10-21:
IIRC https://github.com/ros2/rmw_cyclonedds/pull/187 worked around this problem. @eboasson is there an ETA for a complete solution (e.g. a `DDS-RPC` like implementation for robust discovery or similar)?

hidmic, 2020-11-20:
Some re-design discussion took place in https://github.com/ros2/rmw_fastrtps/pull/418.

abhisheklolage, 2024-02-05:
Hi,
I'm reaching out regarding an issue where I am seeing in a ROS client-service setup where the number of requests that the service receives vs the number of responses that the client gets start diverging slowly. We're actually observing a memory leak where the rate of the leak increases as we increase the frequency of the client-service interactions. Are folks observing a memory leak related to this mismatch between request count vs response count? Or is there a separate known issue with ROS2 client-service with Cyclone DDS? Are there any workarounds? 

System setup: 
 - ROS2 Foxy from https://github.com/ros2/ros2/commit/75b13d3db080df51b75e5720e6f8ae043c0aa092
 - Cyclone DDS
 - Build running on arm64 

Tagging @dirk-thomas for some insight on memory leak. 

Thanks!

Kaju-Bubanja, 2025-06-23:
I'm maybe running in to this issue, has it ever been fixed?