ros2/rmw_cyclonedds #78: Build failures on Windows in TypeSupport2.hpp

nuclearsandwich, 2019-12-07:
## Bug report

**Required Info:**

- Operating System:
	- Windows 10
- Installation type:
	- from source
- Version or commit hash:
	- 88e74c3
- DDS implementation:
	- rmw_cyclonedds_cpp
- Client library (if applicable):
	- N/A

#### Steps to reproduce issue

Example build: https://ci.ros2.org/job/ci_windows/8749/

#### Expected behavior

#### Actual behavior

#### Additional information

I started looking into this and trying the obvious things in https://github.com/ros2/rmw_cyclonedds/tree/nuclearsandwich/include-functional but it's slow going without sitting in front of a Windows machine.

FYI @rotu these build issues look like they were introduced with #42

dirk-thomas, 2019-12-07:
> these build issues look like they were introduced with #42

The problem had already been brought up in that ticket: https://github.com/ros2/rmw_cyclonedds/pull/42#issuecomment-562689802

rotu, 2019-12-07:
Thanks! Yeah, it's due to windows.h defining macros called "min" and "max" which pre-empt and totally break any code using the std::min and std::max functions. The "fix" is to define NOMINMAX (as is done in several CMakeLists in the ROS2 codebase). I'm trying to track down exactly how that header got included here.

rotu, 2019-12-07:
As far as I can tell, it came in through something nuts like `serdata.cpp -> "dds/ddsi/q_radmin.h" -> "dds/ddsrt/threads.h" -> "dds/ddsrt/threads/windows.h" -> "dds/ddsrt/types.h" -> "dds/ddsrt/types/windows.h" -> <windows.h>`. Can't really confirm without a Windows machine. This code definitely could use a cleanup.

rotu, 2019-12-07:
Maybe fixed by https://github.com/eclipse-cyclonedds/cyclonedds/pull/348 (?)

eboasson, 2019-12-07:
> As far as I can tell, it came in through something nuts like `serdata.cpp -> "dds/ddsi/q_radmin.h" -> "dds/ddsrt/threads.h" -> "dds/ddsrt/threads/windows.h" -> "dds/ddsrt/types.h" -> [...]

LwIP causes similar issues within Cyclone's source.

I've been getting away with it because they are not pulled in when you simply include Cyclone's "dds.h". Limiting the platform header files to the "ddsrt" implementation is the plan, but making that happen is not so trivial because there are various types that are abstracted in name only (e.g., ``ddsrt_socket_t`` or ``ddsrt_mutex_t``). I believe that turning all of those into opaque structures will run into C's obscure "undefined behaviour" rules, which might well bite us when link-time optimisation is enabled.

Defining them as exactly the same basic types is probably the best trick (though perhaps still be illegal in C++), but that means you'll get the craziest of issues if ever there is a mismatch between the two. And of course C, not really being "fit for purpose" since the '70s, does not have a good way of checking such things. (I'm contemplating building my own checker using http://hackage.haskell.org/package/language-c-0.8.3/docs/Language-C-Syntax-AST.html or by extending Clang's static analyzer out of frustration.)

Be that as it may, I don't have a Windows box either (and the overhead of maintaining them makes it impossible for me to do so: for some reason they always crash and burn after a few days for me, and I have no clue why … perhaps this is retaliation for my conviction that Windows is suitable for purpose only if the purpose is torture 🤷‍♂). I can't really look at it until Monday ...

What I usually do as a workaround is to wrap the symbol in parentheses to prevent the pre-processor from expanding the macro (it won't treat ``(std::max)(...)`` as something to be macro-expanded).

closed 2020-03-24 by rotu