ros2/rmw_cyclonedds #82: Implement the rmw_get_publishers_info_by_topic and rmw_get_subscriptions_info_by_topic methods

mm318, 2019-12-11:
As a follow up to https://github.com/ros2/rmw_cyclonedds/pull/81, which currently returns `RMW_RET_UNSUPPORTED` for the `rmw_get_publishers_info_by_topic()` and `rmw_get_subscriptions_info_by_topic()` functions, ensure that these methods are implemented before the ROS2 Foxy release.

These methods are implemented for rmw_fastrtps in https://github.com/ros2/rmw_fastrtps/pull/336, which can be used as a reference.

dennis-adlink, 2020-01-30:
@ivanpauno have you already started implementing this? If not: I can make some time next week to implement these 2 methods.

ivanpauno, 2020-02-03:
> @ivanpauno have you already started implementing this? If not: I can make some time next week to implement these 2 methods.

I didn't start. I assigned myself because after moving the `Participant` mapping to a `Context` and not a node, the discovery info is communicated in a completly different way, and the function will have to be rewritten.
If you want to have the function available just now, please go ahead an implement it.

Also, I'm not sure if we're going to do the refactor I previously mention in `cyclonedds` or not, as cyclone doesn't have the performance problems the other vendors have when creating multiple participants. That's another discussion ... but in the case the refactor is not applied to rmw_cyclonedds, I will just unassing me.

dennis-adlink, 2020-02-06:
@ivanpauno I've implemented these two functions based on the current state wrt participant mapping. In case an update of these functions is required after the refactoring you mentioned, I'm happy to do that. In https://github.com/ros2/rcl/pull/571 I've enabled the tests for Cyclone

ivanpauno, 2020-02-06:
@dennis-adlink Thanks for opening this!
I will run CI for #97 together with ros2/rcl#571, when the comments in the later are resolved.

It would be good if somebody from adlink can review #97, as I don't have much time now for doing a thorough review right (I would be able to do it in a few weeks).

mm318, 2020-02-18:
This issue has been resolved with the completion of https://github.com/ros2/rmw_cyclonedds/pull/97.

closed 2020-02-18 by mm318