ros2/rmw_cyclonedds #83: Multiple cyclonedds failures on Windows

ivanpauno, 2019-12-12:
Unfortunately, [nightly_win_extra_rmw_rel](https://ci.ros2.org/view/nightly/job/nightly_win_extra_rmw_rel/) job have failed or been aborted 7 times in a row, but the last completed run showed many test failures:

https://ci.ros2.org/view/nightly/job/nightly_win_extra_rmw_rel/543/#showFailuresLink

Some of the aborted runs were also showing similar failures.

ivanpauno, 2019-12-12:
Transferred to `rmw_cyclone_dds` from visibility.

dirk-thomas, 2019-12-12:
A test build with CycloneDDS on Windows shows a few more than usual test failures: [image].

@eboasson FYI.

ivanpauno, 2020-03-10:
Re-triggering the same job, the situation seems to have improved in the nightlies:
[image].

ivanpauno, 2020-03-10:
Closing, as the remaining failures doesn't seem to be specific of `Windows`.

closed 2020-03-10 by ivanpauno