ros2/rmw_cyclonedds #85: CycloneDDS hangs on exit in Windows

hidmic, 2019-12-17:
## Bug report

**Required Info:**

- Operating System:
  - Windows Server 2019
- Installation type:
  - From source
- Version or commit hash:
  - [master](https://github.com/ros2/rmw_cyclonedds/commit/c0af9d898b9861cda2bf4bab31303efa53263d6e)
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

1. Spin up node, in one terminal:
```sh
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ros2 run demo_nodes_cpp talker
```

2. Try to list topics bypassing the daemon:

```sh
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ros2 topic list --no-daemon
```

#### Expected behavior
Discovered topics (zero or more) are printed on screen and `ros2 topic` exits.

#### Actual behavior
`ros2 topic` hangs and does not react to Ctrl+C anymore

#### Additional information
CycloneDDS trace for the hung process can be found [here](https://gist.github.com/hidmic/e36c149e2d655b5170e2999d4a4e0dec).

It may or may not be related to https://github.com/ros2/build_cop/issues/248 and https://github.com/ros2/rclpy/pull/470.

eboasson, 2020-01-21:
@hidmic apologies for my extreme tardiness in responding ... some of the stuff that showed up shortly before Christmas slipped off the radar over the holidays, I'm afraid.

The trace you quote here is related to this: https://github.com/ros2/ros2/pull/787#issuecomment-547024084. I think I'll change the default behaviour on Windows to run with the "Internal/MultipleReceiveThreads" set to false, that should eliminate that hang.

dirk-thomas, 2020-03-12:
@eboasson What is the status on this ticket?

eboasson, 2020-03-12:
I believe the root cause is the same as the problem that was causing all the Windows CI problems (the multicast quirks). The nightlies are almost green now (it seems to be just the wide-string interop between connext and cyclone), so I expect that this problem also no longer exists. I am ok with closing it, if @hidmic (as the creator of the issue) is ok with that, too.

hidmic, 2020-03-12:
I'd have to give it a shot a Windows VM first. Thanks for the update @eboasson.

hidmic, 2020-03-13:
Alright, no more hangs on Windows as expected. Thanks @eboasson !

closed 2020-03-13 by hidmic