ros2/rmw_cyclonedds #87: Using different typesupports on same node and topic ends in segfault

hidmic, 2020-01-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - From source
- Version or commit hash:
  - Current `master` @ https://github.com/ros2/rmw_cyclonedds/commit/aed236fb269094a361e05a230778d7f77a1431b7
- DDS implementation:
  - CycloneDDS `master` @ [`a6dc93a`](https://github.com/eclipse-cyclonedds/cyclonedds/commit/a6dc93ac03c83b57a317945cfc3ef165d4c7fb16)
- Client library (if applicable):
  - `rclcpp`

#### Steps to reproduce issue
Run `ros1_bridge` tests using https://github.com/ros2/ros1_bridge/pull/233.

#### Expected behavior
Tests pass.

#### Actual behavior
Tests segfault.

#### Additional information

<details>
<summary>Backtrace below</summary>

```
#1  0x00007ffff01d7801 in __GI_abort () at abort.c:79
#2  0x00007ffff01c739a in __assert_fail_base (fmt=0x7ffff034e7d8 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n",
    assertion=assertion@entry=0x7fffe5ca6de8 "str->capacity == str->size + 1",
    file=file@entry=0x7fffe5ca6c40 "/home/michel/Workspaces/ros2_ws/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/TypeSupport2.hpp", line=line@entry=402,
    function=function@entry=0x7fffe5ca6f00 <rmw_cyclonedds_cpp::ROSIDLC_StringValueType::data(void const*) const::__PRETTY_FUNCTION__> "virtual rmw_cyclonedds_cpp::TypedSpan<const char> rmw_cyclonedds_cpp::ROSIDLC_StringValueType::data(const void*) const") at assert.c:92
#3  0x00007ffff01c7412 in __GI___assert_fail (assertion=0x7fffe5ca6de8 "str->capacity == str->size + 1",
    file=0x7fffe5ca6c40 "/home/michel/Workspaces/ros2_ws/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/TypeSupport2.hpp", line=402,
    function=0x7fffe5ca6f00 <rmw_cyclonedds_cpp::ROSIDLC_StringValueType::data(void const*) const::__PRETTY_FUNCTION__> "virtual rmw_cyclonedds_cpp::TypedSpan<const char> rmw_cyclonedds_cpp::ROSIDLC_StringValueType::data(const void*) const") at assert.c:101
#4  0x00007fffe5c97ffd in rmw_cyclonedds_cpp::ROSIDLC_StringValueType::data(void const*) const ()
   from /home/michel/Workspaces/ros2_ws/install/rmw_cyclonedds_cpp/lib/librmw_cyclonedds_cpp.so
#5  0x00007fffe5c93ea8 in rmw_cyclonedds_cpp::CDRWriter::serialize(rmw_cyclonedds_cpp::CDRCursor*, void const*, rmw_cyclonedds_cpp::U8StringValueType const&) const ()
   from /home/michel/Workspaces/ros2_ws/install/rmw_cyclonedds_cpp/lib/librmw_cyclonedds_cpp.so
#6  0x00007fffe5c944be in rmw_cyclonedds_cpp::CDRWriter::serialize(rmw_cyclonedds_cpp::CDRCursor*, void const*, rmw_cyclonedds_cpp::AnyValueType const*) const ()
   from /home/michel/Workspaces/ros2_ws/install/rmw_cyclonedds_cpp/lib/librmw_cyclonedds_cpp.so
#7  0x00007fffe5c94862 in rmw_cyclonedds_cpp::CDRWriter::serialize(rmw_cyclonedds_cpp::CDRCursor*, void const*, rmw_cyclonedds_cpp::StructValueType const&) const ()
   from /home/michel/Workspaces/ros2_ws/install/rmw_cyclonedds_cpp/lib/librmw_cyclonedds_cpp.so
#8  0x00007fffe5c94564 in rmw_cyclonedds_cpp::CDRWriter::serialize(rmw_cyclonedds_cpp::CDRCursor*, void const*, rmw_cyclonedds_cpp::AnyValueType const*) const ()
   from /home/michel/Workspaces/ros2_ws/install/rmw_cyclonedds_cpp/lib/librmw_cyclonedds_cpp.so
#9  0x00007fffe5c9352b in rmw_cyclonedds_cpp::CDRWriter::serialize_top_level(rmw_cyclonedds_cpp::CDRCursor*, void const*) const ()
   from /home/michel/Workspaces/ros2_ws/install/rmw_cyclonedds_cpp/lib/librmw_cyclonedds_cpp.so
#10 0x00007fffe5c932f4 in rmw_cyclonedds_cpp::CDRWriter::get_serialized_size(void const*) const ()
   from /home/michel/Workspaces/ros2_ws/install/rmw_cyclonedds_cpp/lib/librmw_cyclonedds_cpp.so
#11 0x00007fffe5c8a431 in serdata_rmw_from_sample(ddsi_sertopic const*, ddsi_serdata_kind, void const*) ()
   from /home/michel/Workspaces/ros2_ws/install/rmw_cyclonedds_cpp/lib/librmw_cyclonedds_cpp.so
#12 0x00007fffe595bb57 in ddsi_serdata_from_sample (sample=0x7fffac009c90, kind=<optimized out>, topic=<optimized out>)
    at /home/michel/Workspaces/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/include/dds/ddsi/ddsi_serdata.h:179
#13 dds_write_impl (wr=wr@entry=0x7fffac00aa90, data=data@entry=0x7fffac009c90, tstamp=1579131099590500709, action=action@entry=DDS_WR_ACTION_WRITE)
    at /home/michel/Workspaces/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/core/ddsc/src/dds_write.c:171
#14 0x00007fffe595bccb in dds_write (writer=<optimized out>, data=0x7fffac009c90)
    at /home/michel/Workspaces/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/core/ddsc/src/dds_write.c:39
#15 0x00007fffe5c309b8 in rmw_publish () from /home/michel/Workspaces/ros2_ws/install/rmw_cyclonedds_cpp/lib/librmw_cyclonedds_cpp.so
#16 0x00007fffefd85197 in rmw_publish () from /home/michel/Workspaces/ros2_ws/install/rmw_implementation/lib/librmw_implementation.so
#17 0x00007ffff73c578b in rcl_publish () from /home/michel/Workspaces/ros2_ws/install/rcl/lib/librcl.so
#18 0x00007ffff53a41d4 in rclcpp::Publisher<rcl_interfaces::msg::Log_<std::allocator<void> >, std::allocator<void> >::do_inter_process_publish(rcl_interfaces::msg::Log_<std::allocator<void> > const&) () from /home/michel/Workspaces/ros2_ws/install/ros1_bridge/lib/libros1_bridge.so
```

</details>

It seems to me that [`dds_create_topic_arbitrary()`](https://github.com/eclipse-cyclonedds/cyclonedds/blob/bdb7f170531bbaeac7479936acd73cb63f529671/src/core/ddsc/src/dds_topic.c#L283) is not dealing with the fact that the same topic may be associated with different typesupports. In that sense, this issue is equivalent to https://github.com/ros2/rmw_fastrtps/issues/265.

dirk-thomas, 2020-01-16:
Duplicate of #72?

hidmic, 2020-01-16:
Indeed, and maintainers seem to know exactly how to fix it. Closing.

closed 2020-01-16 by hidmic