ros2/rmw_cyclonedds #93: Build failed on ros2 dashing release

YS1974Pizza, 2020-01-30:
## Bug report
Build failed on ros2 dashing release.
The code was download yesterday.  
Build command is "colcon build --symlink-install"

~/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/rmw_get_topic_endpoint_info.cpp:16:10: fatal error: rmw/get_topic_endpoint_info.h: No such file or directory
 #include "rmw/get_topic_endpoint_info.h"

This repo rmw_cyclonedds is set to master branch in ros2.repos file.
And the header get_topic_endpoint_info.h is only on master branch and added in 16 days ago. 
This header file is in rmw repo. In dashing release, the rmw is on dashing branch. 

Please help to confirm it. If you think it's real issue. Please help correct your ros2.repos file, and share patch file to us. Thanks a lot.

**Required Info:**

- Operating System:
  - 
Ubuntu 18.04
- Installation type:
  - 
from source
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

dirk-thomas, 2020-01-30:
Duplicate of #91.

closed 2020-01-30 by dirk-thomas

Reubenyuhe, 2020-03-29:
> after revert commit "aed236fb269094a361e05a230778d7f77a1431b7" in rmw_cyclonedds repo. The build is passed.

I've got the same issue but don't  know really how I can solve it.

eboasson, 2020-03-29:
@Reubenyuhe last Friday a mishap occurred that causes a build failure in rmw_cyclonedds_cpp for dashing & eloquent, it references a file that only very recently got added in Foxy:

https://github.com/ros2/rmw_cyclonedds/blob/c481c10f5da64a680840a4f0810d42e5bdf78491/rmw_cyclonedds_cpp/src/rmw_node.cpp#L40

If you pull the current master the problem should be gone.

YS1974Pizza, 2020-03-29:
I think the root cause of this issue is that repo rmw_cyclonedds is set to master branch in ros2.repos file. Since this release is only for dashing. To keep master branch on one repo is not reasonable. Please ask your CM team to add dashing tag on repo rmw_cyclonedds and change ros2.repos file. In this way, any master change will not impact it.