ros2/rmw_cyclonedds #103: Error when trying to create a custom service

fabmene, 2020-02-24:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Binaries
- Version or commit hash:
  - Eloquent
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Define a custom service:
``` 
bool foo
---
int64 bar
```
```
rosidl_generate_interfaces(${PROJECT_NAME}_srv
  "srv/MySrv.srv"
  DEPENDENCIES builtin_interfaces
)

rosidl_target_interfaces(my_node
  ${PROJECT_NAME}_srv "rosidl_typesupport_cpp")
```

Create service in Node

```
class MyNode : public rclcpp ::Node
{

public:
  MyNode(const rclcpp::NodeOptions& options)
      : rclcpp::Node("my_node", options)
  {
    srv = this->create_service<my_pkg::srv::MySrv>(
        "/my_srv", std::bind(&MyNode::CB, this, std::placeholders::_1, std::placeholders::_2));
  }

  void CB(const my_pkg::srv::MySrv_Request::SharedPtr pReq, const my_pkg::srv::MySrv__Response::SharedPtr pRes){}

  rclcpp::Service<my_pkg::srv::MySrv>::SharedPtr srv;
};
```

#### Expected behavior
Service is offered

#### Actual behavior
Error thrown:
```
Failed to find library 'my_pkg__rosidl_typesupport_introspection_cpp'

>>> [rcutils|error_handling.c:107] rcutils_set_error_state()
This error state is being overwritten:

  'service type support not from this implementation, at /tmp/binarydeb/ros-eloquent-rmw-cyclonedds-cpp-0.4.4/src/rmw_node.cpp:2302'

with this new error message:

  'type_support is null, at /tmp/binarydeb/ros-eloquent-rmw-cyclonedds-cpp-0.4.4/src/rmw_node.cpp:2320'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  could not create service: type_support is null, at /tmp/binarydeb/ros-eloquent-rmw-cyclonedds-cpp-0.4.4/src/rmw_node.cpp:2320, at /tmp/binarydeb/ros-eloquent-rcl-0.8.4/src/rcl/service.c:179
```

dirk-thomas, 2020-02-24:
Please share the complete code of a reproducible example (including the `package.xml` file as well as the complete CMake code).

Note: you might want to use the typedefs for the request / response instead: e.g. `my_pkg::srv::MySr::Request::SharedPtr`.

fabmene, 2020-02-25:
I updated the OP with all information.

fabmene, 2020-02-25:
Testing this issue further, I see the same problem with all other non-fastrtps middle-wares as well, meaning this is likely not a pure cyclone-dds issue.

Furthermore, once I rebuild my workspace after switching around between middlewares, even FastRTPS fails to support the custom service.
```
Failed to find library 'my_pkg__rosidl_typesupport_fastrtps_cpp'
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  could not create service: type support not from this implementation, at /tmp/binarydeb/ros-eloquent-rmw-fastrtps-cpp-0.8.1/src/rmw_service.cpp:95, at /tmp/binarydeb/ros-eloquent-rcl-0.8.4/src/rcl/service.c:179
```

Inbetween a successful launch with FastRTPS and the above error, the only things I did was starting my node, setting RMW_IMPLEMENTATION and clean-building my workspace (which only contains the one package listed in the OP).

I understand this is likely not the correct location to discuss this issue, but I'm not sure where it would belong either.

dirk-thomas, 2020-02-25:
Atm the generated library for the interfaces must have the basename `${PROJECT_NAME}`.

You have to options:

* either rename your interface target to just `${PROJECT_NAME}`:

    ```
    rosidl_generate_interfaces(${PROJECT_NAME} "srv/MySrv.srv")
    rosidl_target_interfaces(my_node ${PROJECT_NAME} "rosidl_typesupport_cpp")
    ```
* or explicitly set the library name:

    ```
    rosidl_generate_interfaces(${PROJECT_NAME}_srv "srv/MySrv.srv" LIBRARY_NAME ${PROJECT_NAME})
    ```