ros2/rmw_cyclonedds #104: CycloneDDS can hang on node destruction

rotu, 2020-02-25:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - Master
- DDS implementation:
  - `rmw_cyclonedds_cpp`
- Client library (if applicable):
  - `rclcpp`

#### Steps to reproduce issue

Very occasionally, tests in test_tf2 fail with timeouts

```colcon test --packages-select test_tf2```

This does not happen every time, and having `ros2 bag record -a` running seems to make it more likely.

#### Expected behavior

The test suite succeeds.

#### Actual behavior

The test suite fails with a timeout in some test or another. Decorating the test suite with debug statements indicates that the failure is some sort of hang in the destructor of `rclcpp::Node`.

#### Additional information
GDB backtrace:

```
#0  0x00007ffbafe089f3 in futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0xe445a4) at ../sysdeps/unix/sysv/linux/futex-internal.h:88
#1  __pthread_cond_wait_common (abstime=0x0, mutex=0xe44550, cond=0xe44578) at pthread_cond_wait.c:502
#2  __pthread_cond_wait (cond=0xe44578, mutex=0xe44550) at pthread_cond_wait.c:655
#3  0x00007ffbac999166 in ddsrt_cond_wait (cond=0xe445a4, mutex=0x80) at /opt/ros/master/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/sync/posix/sync.c:92
#4  0x00007ffbac95e0fb in gcreq_queue_drain (q=0xe44540) at /opt/ros/master/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_gc.c:217
#5  0x00007ffbac9612e9 in rtps_stop (gv=0xdf32b0) at /opt/ros/master/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_init.c:1673
#6  0x00007ffbac981fc7 in dds_domain_free (vdomain=0xdf3010) at /opt/ros/master/src/eclipse-cyclonedds/cyclonedds/src/core/ddsc/src/dds_domain.c:263
#7  0x00007ffbac984f94 in dds_entity_deriver_delete (e=0xdf3010) at /opt/ros/master/src/eclipse-cyclonedds/cyclonedds/src/core/ddsc/src/dds__types.h:189
#8  really_delete_pinned_closed_locked (e=0xdf3010, delstate=DIS_EXPLICIT) at /opt/ros/master/src/eclipse-cyclonedds/cyclonedds/src/core/ddsc/src/dds_entity.c:503
#9  0x00007ffbac984bd0 in dds_delete_impl_pinned (e=0xdf3010, delstate=DIS_EXPLICIT) at /opt/ros/master/src/eclipse-cyclonedds/cyclonedds/src/core/ddsc/src/dds_entity.c:400
#10 dds_delete_impl (entity=<optimized out>, delstate=DIS_EXPLICIT) at /opt/ros/master/src/eclipse-cyclonedds/cyclonedds/src/core/ddsc/src/dds_entity.c:375
#11 dds_delete (entity=<optimized out>) at /opt/ros/master/src/eclipse-cyclonedds/cyclonedds/src/core/ddsc/src/dds_entity.c:358
#12 0x00007ffbacbe07c3 in node_gone_from_domain_locked (did=42) at /opt/ros/master/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/rmw_node.cpp:568
#13 rmw_destroy_node (node=<optimized out>) at /opt/ros/master/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/rmw_node.cpp:833
#14 0x00007ffbb97b9d8c in rcl_node_fini (node=0xe45400) at /opt/ros/master/src/ros2/rcl/rcl/src/rcl/node.c:457
#15 0x00007ffbb9a8fd66 in rclcpp::node_interfaces::NodeBase::NodeBase(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<rclcpp::Context>, rcl_node_options_t const&, bool)::$_1::operator()(rcl_node_t*) const (this=<optimized out>, node=0xe45400) at /opt/ros/master/src/ros2/rclcpp/rclcpp/src/rclcpp/node_interfaces/node_base.cpp:125
#16 0x00007ffbb9a8fe7a in std::_Sp_counted_deleter<rcl_node_t*, rclcpp::node_interfaces::NodeBase::NodeBase(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<rclcpp::Context>, rcl_node_options_t const&, bool)::$_1, std::allocator<void>, (__gnu_cxx::_Lock_policy)2>::_M_dispose() (this=<optimized out>) at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.4.0/../../../../include/c++/7.4.0/bits/shared_ptr_base.h:470
#17 0x00007ffbb9a8f712 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=warning: RTTI symbol not found for class 'std::_Sp_counted_deleter<rcl_node_t*, rclcpp::node_interfaces::NodeBase::NodeBase(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<rclcpp::Context>, rcl_node_options_t const&, bool)::$_1, std::allocator<void>, (__gnu_cxx::_Lock_policy)2>'
0xe45000) at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.4.0/../../../../include/c++/7.4.0/bits/shared_ptr_base.h:154
#18 std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=0x7ffb94002008) at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.4.0/../../../../include/c++/7.4.0/bits/shared_ptr_base.h:684
#19 std::__shared_ptr<rcl_node_t, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=0x7ffb94002000) at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.4.0/../../../../include/c++/7.4.0/bits/shared_ptr_base.h:1123
#20 rclcpp::node_interfaces::NodeBase::~NodeBase (this=0x7ffb94001fe0) at /opt/ros/master/src/ros2/rclcpp/rclcpp/src/rclcpp/node_interfaces/node_base.cpp:153
#21 0x00007ffbb9a8f8a9 in rclcpp::node_interfaces::NodeBase::~NodeBase (this=0x7ffb94001fe0) at /opt/ros/master/src/ros2/rclcpp/rclcpp/src/rclcpp/node_interfaces/node_base.cpp:142
#22 0x00007ffbb9a8a1ed in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=warning: RTTI symbol not found for class 'std::_Sp_counted_ptr<rclcpp::node_interfaces::NodeBase*, (__gnu_cxx::_Lock_policy)2>'
0xe45060) at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.4.0/../../../../include/c++/7.4.0/bits/shared_ptr_base.h:154
#23 std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=0xdf05d0) at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.4.0/../../../../include/c++/7.4.0/bits/shared_ptr_base.h:684
#24 std::__shared_ptr<rclcpp::node_interfaces::NodeBaseInterface, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=0xdf05c8) at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.4.0/../../../../include/c++/7.4.0/bits/shared_ptr_base.h:1123
#25 rclcpp::Node::~Node (this=0xdf05b0) at /opt/ros/master/src/ros2/rclcpp/rclcpp/src/rclcpp/node.cpp:186
#26 0x000000000040a947 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=warning: RTTI symbol not found for class 'std::_Sp_counted_ptr_inplace<rclcpp::Node, std::allocator<rclcpp::Node>, (__gnu_cxx::_Lock_policy)2>'
0xdf05a0) at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.4.0/../../../../include/c++/7.4.0/bits/shared_ptr_base.h:154
#27 std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=0x7fffaa5f4ee0) at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.4.0/../../../../include/c++/7.4.0/bits/shared_ptr_base.h:684
#28 std::__shared_ptr<rclcpp::Node, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=0x7fffaa5f4ed8) at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.4.0/../../../../include/c++/7.4.0/bits/shared_ptr_base.h:1123
#29 MessageFilter_noTransformsSameFrame_Test::TestBody (this=<optimized out>) at /opt/ros/master/src/ros2/geometry2/test_tf2/test/test_message_filter.cpp:108
#30 0x0000000000444fa4 in testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void> (object=<optimized out>, method=<optimized out>, location=0x452b64 "the test body") at /opt/ros/master/install/src/gtest_vendor/./src/gtest.cc:2447
#31 testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void> (object=<optimized out>, method=<optimized out>, location=0x452b64 "the test body") at /opt/ros/master/install/src/gtest_vendor/./src/gtest.cc:2483
#32 0x0000000000425008 in testing::Test::Run (this=0xe53fa0) at /opt/ros/master/install/src/gtest_vendor/./src/gtest.cc:2522
#33 0x0000000000426150 in testing::TestInfo::Run (this=0xda8e80) at /opt/ros/master/install/src/gtest_vendor/./src/gtest.cc:2703
#34 0x0000000000426897 in testing::TestCase::Run (this=0xdedc20) at /opt/ros/master/install/src/gtest_vendor/./src/gtest.cc:2825
#35 0x0000000000433217 in testing::internal::UnitTestImpl::RunAllTests (this=<optimized out>) at /opt/ros/master/install/src/gtest_vendor/./src/gtest.cc:5216
#36 0x0000000000445b44 in testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool> (object=<optimized out>, method=<optimized out>, location=0x45331f "auxiliary test code (environments or event listeners)") at /opt/ros/master/install/src/gtest_vendor/./src/gtest.cc:2447
[... cut ...]
#40 main (argc=1, argc@entry=2, argv=argv@entry=0x7fffaa5f57f8) at /opt/ros/master/src/ros2/geometry2/test_tf2/test/test_message_filter.cpp:409
#41 0x00007ffbafa2bb97 in __libc_start_main (main=0x410510 <main(int, char**)>, argc=2, argv=0x7fffaa5f57f8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffaa5f57e8) at ../csu/libc-start.c:310
#42 0x00000000004093aa in _start () at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.4.0/../../../../include/c++/7.4.0/bits/basic_string.h:220
```

eboasson, 2020-02-25:
@rotu that's odd ... the stack trace says it is waiting for its GC process to complete, so presumably there is something blocking it from completing, but just from the stack trace I can't tell what that might be. There are a number of things one can do apart from my (so far unsuccessful) attempts at reproducing it:
* See if you can get a core dump from it (``gcore`` is your friend) and inspect the state of the process is at this point;
* The other is to guess it might be a deadlock (it need not be, but it might be) and to set the Cyclone option ``Internal/LivelinessMonitoring`` to ``true`` and to have some level of tracing enabled (``Tracing/Verbosity`` set to ``config`` is good enough, for this it need not be very detailed): that causes it to dump stack traces when there is an (apparent) lack of progress. It does this by monitoring per-thread virtual clocks that have something to do with the GC, and so it only picks up some cases, but the fact that it is waiting for the GC to finish there's a chance.

As I said, I have been trying to reproduce it on macOS for a while (running the same test in a loop) but without success so far. For some reason ``ros2 bag record -a`` fails:
```
[ERROR] [1582636732.451284000] [rosbag2_storage]: Unable to create class load instance: package 'rosbag2_storage' not found, searching: [/Users/erik/ros2_ws/install/ros2bag, /Users/erik/ros2_ws/install/tracetools_launch, /Users/erik/ros2_ws/install/topic_monitor, /Users/erik/ros2_ws/install/examples_tf2_py, /Users/erik/ros2_ws/install/sros2, /Users/erik/ros2_ws/install/ros2trace, /Users/erik/ros2_ws/install/ros2topic, /Users/erik/ros2_ws/install/ros2test, (...many more,...)]
```

So I'll try again in a Linux VM after updating that one to the latest Foxy.

eboasson, 2020-02-25:
I think I have reproduced it and managed to attach gdb to it. It gives a slightly different stack trace, but it does mean I have some more info (I've removed some of the irrelevant stuff):
```
Thread 6 (Thread 0x7f5d888cb700 (LWP 30760)):
#2  __pthread_cond_timedwait (cond=0x1a2b228, mutex=0x1a2b200, abstime=abstime@entry=0x7f5d888ca5a0) at pthread_cond_wait.c:667
#3  0x00007f5d8a9579d5 in ddsrt_cond_waituntil (cond=<optimized out>, mutex=<optimized out>, abstime=<optimized out>) at /home/erik/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/sync/posix/sync.c:116
#4  0x00007f5d8a8fca7e in threadmon_thread (sl=0x1a2b1d0) at /home/erik/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/ddsi_threadmon.c:78
#5  0x00007f5d8a93254b in create_thread_wrapper (ptr=0x1a230d0) at /home/erik/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_thread.c:220

Thread 5 (Thread 0x7f5d890cc700 (LWP 30759)):
#2  __pthread_cond_wait (cond=cond@entry=0x19fd818, mutex=mutex@entry=0x19fd7f0) at pthread_cond_wait.c:655
#3  0x00007f5d8a957959 in ddsrt_cond_wait (cond=cond@entry=0x19fd818, mutex=mutex@entry=0x19fd7f0) at /home/erik/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/sync/posix/sync.c:92
#4  0x00007f5d8a9285dd in dqueue_thread (q=0x19fd7f0) at /home/erik/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_radmin.c:2443
#5  0x00007f5d8a93254b in create_thread_wrapper (ptr=0x1a65df0) at /home/erik/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_thread.c:220

Thread 4 (Thread 0x7f5d898cd700 (LWP 30758)):
#2  __pthread_cond_wait (cond=cond@entry=0x19fba88, mutex=mutex@entry=0x19fba60) at pthread_cond_wait.c:655
#3  0x00007f5d8a957959 in ddsrt_cond_wait (cond=cond@entry=0x19fba88, mutex=mutex@entry=0x19fba60) at /home/erik/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/sync/posix/sync.c:92
#4  0x00007f5d8a9285dd in dqueue_thread (q=0x19fba60) at /home/erik/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_radmin.c:2443
#5  0x00007f5d8a93254b in create_thread_wrapper (ptr=0x1a525c0) at /home/erik/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_thread.c:220

Thread 3 (Thread 0x7f5d8a0ce700 (LWP 30757)):
#0  0x00007f5d8bda20b5 in __GI___nanosleep (requested_time=requested_time@entry=0x7f5d8a0cd5b0, remaining=remaining@entry=0x7f5d8a0cd5c0) at ../sysdeps/unix/sysv/linux/nanosleep.c:28
#1  0x00007f5d8a95850f in dds_sleepfor (n=1000000) at /home/erik/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/time.c:31
#2  dds_sleepfor (n=n@entry=1000000) at /home/erik/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/time.c:24
#3  0x00007f5d8a91f115 in gcreq_queue_thread (q=0x1a2c160) at /home/erik/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_gc.c:169
#4  0x00007f5d8a93254b in create_thread_wrapper (ptr=0x1a79890) at /home/erik/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_thread.c:220

Thread 2 (Thread 0x7f5d8a8cf700 (LWP 30724)):
#0  futex_abstimed_wait_cancelable (private=0, abstime=0x0, expected=0, futex_word=0x7f5d8bffeb60 <rclcpp::SignalHandler::signal_handler_sem_>) at ../sysdeps/unix/sysv/linux/futex-internal.h:205
#1  do_futex_wait (sem=sem@entry=0x7f5d8bffeb60 <rclcpp::SignalHandler::signal_handler_sem_>, abstime=0x0) at sem_waitcommon.c:111
#2  0x00007f5d8bda0da8 in __new_sem_wait_slow (sem=sem@entry=0x7f5d8bffeb60 <rclcpp::SignalHandler::signal_handler_sem_>, abstime=0x0) at sem_waitcommon.c:181
#3  0x00007f5d8bda0e21 in __new_sem_wait (sem=sem@entry=0x7f5d8bffeb60 <rclcpp::SignalHandler::signal_handler_sem_>) at sem_wait.c:42
#4  0x00007f5d8bfa6622 in rclcpp::SignalHandler::wait_for_signal () at /home/erik/ros2_ws/src/ros2/rclcpp/rclcpp/src/rclcpp/signal_handler.cpp:356
#5  0x00007f5d8bfa7259 in rclcpp::SignalHandler::deferred_signal_handler (this=<optimized out>) at /home/erik/ros2_ws/src/ros2/rclcpp/rclcpp/src/rclcpp/signal_handler.cpp:277

Thread 1 (Thread 0x7f5d8aa11f80 (LWP 30723)):
#2  __pthread_cond_wait (cond=cond@entry=0x1a84158, mutex=mutex@entry=0x1a84130) at pthread_cond_wait.c:655
#3  0x00007f5d8a957959 in ddsrt_cond_wait (cond=cond@entry=0x1a84158, mutex=mutex@entry=0x1a84130) at /home/erik/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/sync/posix/sync.c:92
#4  0x00007f5d8a9225a3 in rtps_stop (gv=gv@entry=0x1a83d40) at /home/erik/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_init.c:1667
#5  0x00007f5d8a9429e6 in dds_domain_free (vdomain=0x1a83aa0) at /home/erik/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/core/ddsc/src/dds_domain.c:263
...
#10 0x00007f5d8a945f57 in dds_delete (entity=<optimized out>) at /home/erik/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/core/ddsc/src/dds_entity.c:358
#11 0x00007f5d8a9afb55 in node_gone_from_domain_locked (did=<optimized out>) at /home/erik/ros2_ws/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/rmw_node.cpp:568
#12 0x00007f5d8a9afc12 in rmw_destroy_node (node=<optimized out>) at /home/erik/ros2_ws/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/rmw_node.cpp:833
#13 0x00007f5d8be72ca1 in rcl_node_fini (node=node@entry=0x1a5fb50) at /home/erik/ros2_ws/src/ros2/rcl/rcl/src/rcl/node.c:457
```

The nanosleep in thread 3 is the GC continually trying to make progress on the current GC-able item — but it has to wait until all other (Cyclone) threads have progressed enough to be certain that none of them is still using the item to be GC'd. This in turn means that it can't shut down.

[... cut ...]
diff --git a/src/core/ddsi/src/q_thread.c b/src/core/ddsi/src/q_thread.c
index a8190fa..a0a423c 100644
--- a/src/core/ddsi/src/q_thread.c
+++ b/src/core/ddsi/src/q_thread.c
@@ -216,6 +216,7 @@ static uint32_t create_thread_wrapper (void *ptr)
   struct ddsi_domaingv const * const gv = ddsrt_atomic_ldvoidp (&ctx->self->gv);
   if (gv)
     GVTRACE ("started new thread %"PRIdTID": %s\n", ddsrt_gettid (), ctx->self->name);
+  tsd_thread_state = ctx->self;
   ctx->self->tid = ddsrt_thread_self ();
   ret = ctx->f (ctx->arg);
   ddsrt_free (ctx);
```

rotu, 2020-02-27:
still haven't gotten to the bottom of this, but I am readily able to see an assertion error here:

https://github.com/eclipse-cyclonedds/cyclonedds/blob/b84eee5abbd4e935d3e47186e07ed93e4730b5f1/src/core/ddsi/src/q_thread.c#L327

Still trying to understand the code. Do note that when running with ASAN, it modifies the thread runtime, so you may need to build in debug mode (or at least NDEBUG undefined) with ASAN off.

eboasson, 2020-02-27:
> Still trying to understand the code.

... I know ... too much history, not enough time to clean it up ...

> still haven't gotten to the bottom of this

Is that with my suggested modification or without? Because that really makes a difference given my current understanding of the issue: that the system reuses thread ids on a second/third/whatever round of creating a domain and deleting it; that this can cause the lazy setting of ``tsd_thread_state`` to have multiple candidates in the ``ALIVE`` state with a matching thread id, and that this, in turn, can result in two threads setting their ``tsd_thread_state`` to the same entry in the ``thread_states.ts`` array. That breaks all kinds of assumptions, and can explain the assertion you are seeing, as well as the "impossible" value of the virtual clock I mentioned in my previous comment.

I just created a PR (https://github.com/eclipse-cyclonedds/cyclonedds/pull/411) based on my efforts yesterday. I intended to have another look at it before creating the PR, but I am now thinking it is better to share it already.

rotu, 2020-02-27:
I had not tried your fix since I thought it was a workaround and I thought you were still architecting a proper fix. I tested the PR and am confident that https://github.com/eclipse-cyclonedds/cyclonedds/pull/411 fixes this issue.

eboasson, 2020-02-28:
> I had not tried your fix since I thought it was a workaround

A change that apparently fixes a symptom but isn't based on an understanding of the cause surely is but a workaround?

In this case the change came first, the understanding later 🙂

Thanks for verifying it!

closed 2020-02-28 by rotu