ros2/rmw_cyclonedds #111: Crash when subscribing to a single topic with multiple types

pjreed, 2020-03-09:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Linux 18.04
- Installation type:
  - ROS Eloquent, from binaries
- Version or commit hash:
  - 0.4.4-1bionic.20200121.221538
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

This issue happens when a single node subscribes to a topic multiple times and each subscriber has a different message type.

It's simple to reproduce this with rviz:

```
. /opt/ros/eloquent/setup.bash
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
ros2 run rviz2 rviz2
```

1. Click Add / Pose
2. Click Add / Marker
3. Change the Pose plugin's topic to `/test`
4. Change the Marker plugin's topic to `/test`
5. rviz will crash

#### Expected behavior

The program should subscribe to the topic without crashing.  Each subscriber callback should receive messages published to that topic for the type they expect.  Note that FastRTPS behaves correctly.

#### Actual behavior

The program crashes as soon as the second `create_subscription` call is made.  Furthermore, it does not seem like it is throwing an exception; I tested this this my making my own program that subscribed to a topic multiple times, and I was unable to catch an exception, it appeared as though the program crashed due to `SIGABRT` being raised.

#### Additional information

This bug also exists with the binary version in ROS Dashing.

Console output from an example rviz session:

```
1583786612.521721 [230]      rviz2: using network interface enp21s0 (udp/x.x.x.x) selected arbitrarily from: enp21s0, docker0
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: Stereo is NOT SUPPORTED
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  could not create subscription: failed to create topic, at /tmp/binarydeb/ros-eloquent-rmw-cyclonedds-cpp-0.4.4/src/rmw_node.cpp:1446, at /tmp/binarydeb/ros-eloquent-rcl-0.8.4/src/rcl/subscription.c:168
```

eboasson, 2020-03-10:
From the description, this is a duplicate of #72, which got fixed for Foxy (https://github.com/eclipse-cyclonedds/cyclonedds/pull/382 and #96). While nothing has officially been backported to Eloquent yet, the sources are compatible and I expect the problem will be gone if you recompile cyclonedds & rmw_cyclonedds from current master.

dirk-thomas, 2020-03-10:
@eboasson the fix should be released into Eloquent (and probably Dashing).

eboasson, 2020-03-10:
Yes, we’ll do that. Do the binary package repositories get updated automatically when the sources are?

dirk-thomas, 2020-03-10:
> Do the binary package repositories get updated automatically when the sources are?

You don't update either of them yourself. You run `bloom` to create the information for the new release in the gbp repo and `bloom` then creates a pull request to `rosdistro` bumping the release version. Once that is merged the Jenkins farm picks it up and first builds the sourcedebs and then the binarydebs (and potentially rebuilds downstream packages).

eboasson, 2020-03-12:
@dirk-thomas we think we figured out how the steps tie together when ``blooming``. @ThijsSassen is now trying to run ``bloom`` after we updated and tagged our repos. It fails because he doesn't have write access to the https://github.com/ros2-gbp/cyclonedds-release and https://github.com/ros2-gbp/rmw_cyclonedds-release

Can you give him write access?

dirk-thomas, 2020-03-12:
Invited both of you to both repos.

eboasson, 2020-03-12:
Thanks @dirk-thomas!

eboasson, 2020-03-25:
@pjreed this problem should have disappeared about two weeks ago with #114 getting merged (certainly for a build from source, I don't know if the binary packages have been updated yet). I'd appreciate it if you could confirm this before closing the issue. Thanks!

pjreed, 2020-03-25:
I see in that `package.xml` file that it's version `0.5.1`, although the latest version of `ros-dashing-rmw-cyclonedds-cpp` in the apt repository is `0.4.2-1bionic.20200318.032607`, which looks like it was built a week ago but is not the latest version; I tested that and it still crashes.

The rosdistro distribution files at https://github.com/ros/rosdistro/blob/master/dashing/distribution.yaml and https://github.com/ros/rosdistro/blob/master/eloquent/distribution.yaml also respectively list `0.4.2-1` and `0.4.4-1` as the last versions in those distros.  Do updates still need to be pushed to those?

eboasson, 2020-03-25:
@pjreed yes, that fix version hasn't made it to the official dashing & eloquent versions yet. It is in the pipeline, but I don't know exactly when it'll be available via apt. If you don't mind, I'd prefer waiting for a bit over setting up everything just to try and reproduce it.

clalancette, 2023-06-13:
Given the age of this, and that we think there was a fix, I'm going to go ahead and close it out.  If you disagree, please feel free to reopen.

closed 2023-06-13 by clalancette