ros2/rmw_cyclonedds #116: Memory leak when unloading a component offering a service under Cyclone DDS

fabmene, 2020-03-19:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Binaries
- Version or commit hash:
  - Eloquent
- DDS implementation:
  - Cyclone DDS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
1. Have a node that offers a service, built as [component](https://github.com/ros2/rclcpp/tree/master/rclcpp_components)
2. Load component
3. Call service
4. Unload component

#### Expected behavior
Node is unloaded, memory is freed.

#### Actual behavior
Error message `[ERROR] [rclcpp]: Error in destruction of rcl service handle: the Node Handle was destructed too early. You will leak memory`

#### Additional information
I have multiple identical nodes loaded as components, unloading them all at once. The error occurs only rarely, but if it does, it will occur for a large number of nodes at once.

eboasson, 2020-03-20:
Thanks for reporting this @fabmene. I think the error message comes from https://github.com/ros2/rclcpp/blob/3361e68bb966668791c56d64509669a2821c5e9c/rclcpp/include/rclcpp/service.hpp#L129 and that suggests an issue with rclcpp/rcl more than an issue with rmw and the levels below that. @dirk-thomas (apologies for always going to you) would you agree?

And @fabmene if you can distill this into a small reproducing example, that would help a lot in looking for the root cause.

fabmene, 2020-03-20:
The reason why I am reporting this issue here is that I am only seeing this error with CycloneDDS specifically. 

I am currently in the process of evaluating the suitability of the current RMWs for our specific usecase and in the process I run the same set of tests under different RMWs (mostly FastRTPS and CycloneDDS). Under FastRTPS I never see this error, while it is somewhat common with CycloneDDS.

Of course, it could still be the case that the issue is not caused by CycloneDDS, but rather points at an underlying issue in rclcpp, I am not arguing that.

I will try to find a reproducing example and get back to you.

eboasson, 2020-03-20:
> The reason why I am reporting this issue here is that I am only seeing this error with CycloneDDS specifically.

That's interesting, and so

> I will try to find a reproducing example and get back to you.

I really hope that will prove possible without too much effort. Thanks!

dirk-thomas, 2020-06-01:
> I will try to find a reproducing example and get back to you.

@fabmene Friendly ping.

clalancette, 2023-06-13:
Given the age of this, I am going to close this out.  If you are still having problems, please feel free to open a new issue.

closed 2023-06-13 by clalancette