ros2/rmw_cyclonedds #117: Newly created Liveliness Event is immediately true

rotu, 2020-03-19:
http://build.ros2.org/view/Fci/job/Fci__nightly-cyclonedds_ubuntu_focal_amd64/lastSuccessfulBuild/testReport/rclpy.src.ros2.rclpy.rclpy.test.test_qos_event/TestQoSEvent/test_call_subscription_rclpy_event_apis/

Not sure if this is a bug in CycloneDDS or not

rotu, 2020-03-19:
Could not reproduce locally on Focal + Foxy Master.

eboasson, 2020-03-20:
I've looked at that test a bit and ran it with Cyclone tracing enabled to double-check my interpretation of the test (on my mac) and just about the only way this can fail is if there's some process running in parallel with the same topic name: there simply can't be a "liveliness changed" if no writer is matched. That suggests ``test_waitable.py``, which also uses name space "/rclpy/test" and creates a writer for "test_topic" must have been running at the same time.

It's just that I thought the tests were run one at the time ...

rotu, 2020-03-22:
The tests *are* run sequentially. `Invoking '. /opt/ros/foxy/setup.sh && PYTHONIOENCODING=utf_8 PYTHONUNBUFFERED=1 colcon test --build-base build_isolated --install-base install_isolated --test-result-base test_results --event-handlers console_direct+ --executor sequential --test-result-base /tmp/ws/test_results' in '/tmp/ws'. But due to the setup and teardown being at the *class* level, lots of state is being persisted between test cases. That may be responsible

closed 2020-04-06 by clalancette