ros2/rmw_cyclonedds #118: rosbag2_transport CI test failure

rotu, 2020-03-19:
http://build.ros2.org/view/Fci/job/Fci__nightly-cyclonedds_ubuntu_focal_amd64/lastSuccessfulBuild/testReport/rosbag2_transport/RecordIntegrationTestFixture/published_messages_from_multiple_topics_are_recorded/

investigation needed

rotu, 2020-03-24:
Reproduced locally Focal / Foxy Master

rotu, 2020-03-24:
I can see various crashes - I think there may be issues in hashmap:
```
1585076737.466860 [42] test_recor: using network interface wlp0s20f3 (udp/10.0.2.85) selected arbitrarily from: wlp0s20f3, docker0

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'array_list is already initialized, at /opt/ros/master/src/ros2/rcutils/src/array_list.c:61, at /opt/ros/master/src/ros2/rcl/rcl/src/rcl/logging_rosout.c:188'

with this new error message:

  'Failed to add publisher to map., at /opt/ros/master/src/ros2/rcl/rcl/src/rcl/logging_rosout.c:190'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [1585076750.251731495] [rosbag2_transport]: Failed to record: failed to initialize rcl node: Failed to add publisher to map., at /opt/ros/master/src/ros2/rcl/rcl/src/rcl/logging_rosout.c:190
```
Stack trace
```
rcutils_array_list_init array_list.c:61
hash_map_insert_entry hash_map.c:165
rcutils_hash_map_set hash_map.c:399
rcl_logging_rosout_init_publisher_for_node logging_rosout.c:188
rcl_node_init node.c:362
...
```

rotu, 2020-03-24:
Sometimes I get a segfault when resizing
```
Signal: SIGSEGV (signal SIGSEGV: invalid address (fault address: 0x0))
```

```
hash_map_check_and_grow_map hash_map.c:209
rcutils_hash_map_set hash_map.c:410
rcl_logging_rosout_init_publisher_for_node logging_rosout.c:188
rcl_node_init node.c:362
rclcpp::node_interfaces::NodeBase::NodeBase(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<rclcpp::Context>, rcl_node_options_t const&, bool) node_base.cpp:66
rclcpp::Node::Node(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::NodeOptions const&) node.cpp:101
rclcpp::Node::Node(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::NodeOptions const&) node.cpp:93
rosbag2_transport::Rosbag2Node::Rosbag2Node(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) rosbag2_node.cpp:35
std::__shared_ptr<rosbag2_transport::Rosbag2Node, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<rosbag2_transport::Rosbag2Node>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::_Sp_alloc_shared_tag<std::allocator<rosbag2_transport::Rosbag2Node> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&) new_allocator.h:147
std::__shared_ptr<rosbag2_transport::Rosbag2Node, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<rosbag2_transport::Rosbag2Node>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::_Sp_alloc_shared_tag<std::allocator<rosbag2_transport::Rosbag2Node> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&) alloc_traits.h:484
std::__shared_ptr<rosbag2_transport::Rosbag2Node, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<rosbag2_transport::Rosbag2Node>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::_Sp_alloc_shared_tag<std::allocator<rosbag2_transport::Rosbag2Node> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&) shared_ptr_base.h:548
std::__shared_ptr<rosbag2_transport::Rosbag2Node, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<rosbag2_transport::Rosbag2Node>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::_Sp_alloc_shared_tag<std::allocator<rosbag2_transport::Rosbag2Node> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&) shared_ptr_base.h:680
std::__shared_ptr<rosbag2_transport::Rosbag2Node, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<rosbag2_transport::Rosbag2Node>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::_Sp_alloc_shared_tag<std::allocator<rosbag2_transport::Rosbag2Node> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&) shared_ptr_base.h:1344
rosbag2_transport::Rosbag2Transport::setup_node(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) shared_ptr.h:359
rosbag2_transport::Rosbag2Transport::setup_node(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) shared_ptr.h:701
rosbag2_transport::Rosbag2Transport::setup_node(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) shared_ptr.h:717
rosbag2_transport::Rosbag2Transport::setup_node(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) rosbag2_transport.cpp:87
rosbag2_transport::Rosbag2Transport::record(rosbag2_cpp::StorageOptions const&, rosbag2_transport::RecordOptions const&) rosbag2_transport.cpp:75
RecordIntegrationTestFixture::start_recording(rosbag2_transport::RecordOptions const&)::'lambda'()::operator()() const record_integration_fixture.hpp:55
std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<RecordIntegrationTestFixture::start_recording(rosbag2_transport::RecordOptions const&)::'lambda'()> >, void>::operator()() const invoke.h:60
std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<RecordIntegrationTestFixture::start_recording(rosbag2_transport::RecordOptions const&)::'lambda'()> >, void>::operator()() const invoke.h:95
std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<RecordIntegrationTestFixture::start_recording(rosbag2_transport::RecordOptions const&)::'lambda'()> >, void>::operator()() const thread:244
std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<RecordIntegrationTestFixture::start_recording(rosbag2_transport::RecordOptions const&)::'lambda'()> >, void>::operator()() const thread:251
std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<RecordIntegrationTestFixture::start_recording(rosbag2_transport::RecordOptions const&)::'lambd [... cut ...]
std::__future_base::_State_baseV2::_M_set_result(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>, bool) future:401
std::__future_base::_Async_state_impl<std::thread::_Invoker<std::tuple<RecordIntegrationTestFixture::start_recording(rosbag2_transport::RecordOptions const&)::'lambda'()> >, void>::_Async_state_impl(std::thread::_Invoker<std::tuple<RecordIntegrationTestFixture::start_recording(rosbag2_transport::RecordOptions const&)::'lambda'()> >&&)::'lambda'()::operator()() const future:1662
<unknown> 0x00007f6d02fa4cb4
start_thread 0x00007f6d032a7609
clone 0x00007f6d02c94153
```

rotu, 2020-03-24:
@nburek, could you please take a look at this? Is there a bug in our hashmap implementation? In `rcl`'s usage of it?

nburek, 2020-03-24:
The first stack trace you posted would indicate a thread race condition to me, since the hashmap checks if the list is initialized before trying to initialize it and then the list initialization function does the same check and determines that it's not. Possibly something weird with a messed up pointer or corrupted stack, but I need to dig more into what you're running to understand.

Is your test spinning up multiple threads or initializing multiple contexts within the same process? If so, thread safety checks in rcl would be the next place I would check.

rotu, 2020-03-24:
Ah. I think you're right that the tests are using `std::async` in an unsafe way. `rcl_node_init` is already marked as not threadsafe, and these crashes demonstrate why. So it looks to me like a defect in `rosbag2_transport`.

@zmichaels11, would you like to dig in?

Stack trace in thread 1:
```
rcl_node_init node.c:125
rclcpp::node_interfaces::NodeBase::NodeBase(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<rclcpp::Context>, rcl_node_options_t const&, bool) node_base.cpp:66
rclcpp::Node::Node(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::NodeOptions const&) node.cpp:101
rclcpp::Node::Node(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::NodeOptions const&) node.cpp:93
std::__shared_ptr<rclcpp::Node, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<rclcpp::Node>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, rclcpp::NodeOptions&>(std::_Sp_alloc_shared_tag<std::allocator<rclcpp::Node> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, rclcpp::NodeOptions&) new_allocator.h:147
std::__shared_ptr<rclcpp::Node, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<rclcpp::Node>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, rclcpp::NodeOptions&>(std::_Sp_alloc_shared_tag<std::allocator<rclcpp::Node> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, rclcpp::NodeOptions&) alloc_traits.h:484
std::__shared_ptr<rclcpp::Node, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<rclcpp::Node>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, rclcpp::NodeOptions&>(std::_Sp_alloc_shared_tag<std::allocator<rclcpp::Node> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, rclcpp::NodeOptions&) shared_ptr_base.h:548
std::__shared_ptr<rclcpp::Node, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<rclcpp::Node>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, rclcpp::NodeOptions&>(std::_Sp_alloc_shared_tag<std::allocator<rclcpp::Node> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, rclcpp::NodeOptions&) shared_ptr_base.h:680
std::__shared_ptr<rclcpp::Node, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<rclcpp::Node>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, rclcpp::NodeOptions&>(std::_Sp_alloc_shared_tag<std::allocator<rclcpp::Node> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, rclcpp::NodeOptions&) shared_ptr_base.h:1344
void rosbag2_test_common::PublisherManager::add_publisher<test_msgs::msg::Strings_<std::allocator<void> > >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<test_msgs::msg::Strings_<std::allocator<void> > >, unsigned long) shared_ptr.h:359
void rosbag2_test_common::PublisherManager::add_publisher<test_msgs::msg::Strings_<std::allocator<void> > >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<test_msgs::msg::Strings_<std::allocator<void> > >, unsigned long) shared_ptr.h:701
void rosbag2_test_common::PublisherManager::add_publisher<test_msgs::msg::Strings_<std::allocator<void> > >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<test_msgs::msg::Strings_<std::allocator<void> > >, unsigned long) shared_ptr.h:717
void rosbag2_test_common::PublisherManager::add_publisher<test_msgs::msg::Strings_<std::allocator<void> > >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<test_msgs::msg::Strings_<std::allocator<void> > >, unsigned long) publisher_manager.hpp:81
RecordIntegrationTestFixture_published_messages_from_multiple_topics_are_recorded_Test::TestBody() test_record_all.cpp:36
void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) gtest.cc:2447
void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) gtest.cc:2483
testing::Test::Run() gtest.cc:2522
testing::TestInfo::Run() gtest.cc:2703
testing::TestCase::Run() gtest.cc:2825
testing::internal::UnitTestImpl::RunAllTests() gtest.cc:5216
bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) gtest.cc:2447
bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) gtest.cc:2483
testing::UnitTest::Run() gtest.cc:4824
main gtest.h:2370
main gmock_main.cc:53
__libc_start_main 0x00007f9b22fce0b3
_start 0x000000000040fd9e
```

Stack trace in thread 2:
```
rcl_node_init node.c:125
[... cut ...]
std::__future_base::_State_baseV2::_M_set_result(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>, bool) future:401
std::__future_base::_Async_state_impl<std::thread::_Invoker<std::tuple<RecordIntegrationTestFixture::start_recording(rosbag2_transport::RecordOptions const&)::'lambda'()> >, void>::_Async_state_impl(std::thread::_Invoker<std::tuple<RecordIntegrationTestFixture::start_recording(rosbag2_transport::RecordOptions const&)::'lambda'()> >&&)::'lambda'()::operator()() const future:1662
<unknown> 0x00007f9b233d9cb4
start_thread 0x00007f9b236dc609
clone 0x00007f9b230c9153
```

zmichaels11, 2020-03-24:
Hey @rotu , thanks for bringing this up.
We're now aware of this and don't have the resources to act upon it immediately.

I've opened an issue on https://github.com/ros2/rosbag2/issues/329 which will be pulled into the tooling WG's backlog.

Can you please help fill in the missing information for the bug report?

rotu, 2020-03-24:
closing in favor of https://github.com/ros2/rosbag2/issues/329

closed 2020-03-24 by rotu