ros2/rmw_cyclonedds #120: crashed on Windows Debug

dirk-thomas, 2020-03-20:
See https://ci.ros2.org/job/ci_windows-metal/5/ for a build with only CycloneDDS testing `test_communication`. When invoking `test_publisher_cpp.exe Empty /topic` it exits with an assertion:

> vector subscription out of range

Overall it fails 88 tests. I assume they are all related to the same reason.

dirk-thomas, 2020-03-20:
The bug is related to this call: https://github.com/ros2/rmw_cyclonedds/blob/025762ac4f21c7d77b1e3cb09642debc7bb14193/rmw_cyclonedds_cpp/src/Serialization.cpp#L513-L514

The invocation of `sequence_contents()` tries to get the pointer to the data of a vector with size 0. Accessing `[0]` from that vector results in an out of bound exception. Since the `count` argument is zero this shouldn't try to access the vector data.

dirk-thomas, 2020-03-20:
@rotu FYI since you seem to have touched this part of the code recently.

rotu, 2020-03-20:
Thanks, @dirk-thomas! On it!

eboasson, 2020-03-20:
@dirk-thomas Hah, I had *just* managed to catch this crash in the debugger! It took me forever because Windows how much I hate it and does everything in its power to thwart my efforts.

@rotu Thanks!

PR #122 (2020-03-20): Prevent undefined behavior when serializing empty vector
Since m_get_const_function calls `std::vector<T>::operator[]`, accessing the zeroth element causes undefined behavior. Instead, return a null pointer to make the function behave sanely when vector is empty.
Fix #120
commit 3ea93f27 (2020-03-20): Prevent undefined behavior when serializing empty vector (#122)

Since m_get_const_function calls `std::vector<T>::operator[]`, accessing the zeroth element causes undefined behavior. Instead, return a null pointer to make the function behave sanely when vector is empty.
Fix #120
closed 2020-03-20 by rotu