ros2/rmw_cyclonedds #121: WString tests failing against Connext

dirk-thomas, 2020-03-20:
From https://github.com/ros2/build_farmer/issues/268#issuecomment-598926902

eboasson, 2020-03-25:
For background information and references: #43.

dirk-thomas, 2020-04-03:
@eboasson What is the status on this? I would like to get the failing tests resolved - one way or another.

rotu, 2020-04-03:
I started a fix but haven’t yet finished it. https://github.com/RoverRobotics-forks/rmw_cyclonedds/tree/wstring

Other things came up but I can make this a priority if need be

dirk-thomas, 2020-04-03:
Since the failing tests are a constant noise when determining if changes break anything or if anything regresses. So addressing them rather sooner than later would be helpful.

rotu, 2020-04-03:
Addressing them will exchange one set of test failures for another until FastRTPS also fixes their implementation.

dirk-thomas, 2020-04-03:
As mentioned before: please double check that the behavior of Connext is really the one being desired. If I recall correctly FastRTPS adheres to the newer version of the standard which I would consider the better future proof approach.

rotu, 2020-04-03:
Aww heck! It looks like between DDS-XTypes 1.1 and 1.2, OMG redefined the WString representations for CDR at the same time they started versioning CDR.

That is absolutely diabolical!

Yes, it looks like the behavior of Connext is probably the desirable behavior based on their docs. I do need to test it. We may come out of this with 3 different serialization formats rather than 2, and I think we should disable inter-vendor WString tests for now.

eboasson, 2020-04-03:
@rotu the standards body is called "OMG" for a reason ... and thank you for diving into this diabolical mess!

MiguelCompany, 2020-04-03:
@eboasson @rotu According to [this](https://github.com/ros2/system_tests/blob/68c495a58df2502c77ebc972aba87c73ac46adcc/test_communication/CMakeLists.txt#L198-L216), the exception for Fast RTPS is on Mac only, and only when Connext is subscribing.

This means that Fast RTPS is able to receive wstrings from Connext on Mac, and that they are interoperable on Windows and Linux. For me that seems more like a problem in Connext, and I guess that is the reason for [this issue](https://github.com/ros2/rmw_connext/issues/364) being open on rmw_connext.

Nevertheless, thanks @rotu for pointing out the subtle and diabolical change between the two versions of the standard. I guess the best way would be for Fast-CDR to support strings of both types (i.e. char32 and char16), so the user could choose which one to use.

rotu, 2020-04-06:
Closing duplicate of #43

closed 2020-04-06 by rotu

dirk-thomas, 2020-04-11:
@rotu What is the status of this? Is there any work towards addressing the failing test ongoing? If yes, in which PR and when is it expected to land? If not, I propose to disable the test with a reference to the issue.