ros2/rmw_cyclonedds #126: rmw_cyclonedds_cpp crashes when receiving a message on /rosout

pjreed, 2020-03-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Linux 18.04
- Installation type:
  - Binaries
- Version or commit hash:
  - Dashing:
    - ros-dashing-cyclonedds: 0.1.0-3bionic.20191205.182051
    - ros-dashing-rmw-cyclonedds-cpp: 0.4.2-1bionic.20200318.032607
  - Eloquent:
    - ros-eloquent-cyclonedds: 0.1.0-7bionic.20191213.041258
    - ros-eloquent-rmw-cyclonedds-cpp: 0.4.4-1bionic.20200121.221538
- DDS implementation:
  - rmw_cyclonedds_cpp
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

1. Create a rclcpp client that subscribes to `rcl_interfaces:msg::Log` messages on `/rosout` using `rmw_cyclonedds_cpp`
2. Publish a message to `/rosout`
    - `ros2 topic pub -1 /rosout rcl_interfaces/msg/Log`
3. The client will crash

Here's an example program:
```cpp
#include <rclcpp/rclcpp.hpp>
#include <rcl_interfaces/msg/log.hpp>
#include <stdio.h>

int main(int argc, char **argv)
{
  rclcpp::init(argc, argv);
  auto node = rclcpp::Node::make_shared("cyclone_log_test");

  auto sub = node->create_subscription<rcl_interfaces::msg::Log>("/rosout", 1,
      [node](rcl_interfaces::msg::Log::ConstSharedPtr msg) {
    std::cout << "rosout: " << msg->msg << std::endl;
  });
  auto sub2 = node->create_subscription<rcl_interfaces::msg::Log>("/rosout2", 1,
      [node](rcl_interfaces::msg::Log::ConstSharedPtr msg) {
    std::cout << "rosout2: " << msg->msg << std::endl;
  });

  rclcpp::spin(node);
  return 0;
}
```

#### Expected behavior

The above client should not crash, but instead print `rosout: ` followed by the log message.

#### Actual behavior

The client crashes.

#### Additional information

The problem does not happen when subscribing to other topics.  In the above example, it also creates a subscriber on `/rosout2`, and publishing a log message there will print as expected.

Crashes do not occur when using `rmw_fastrtps_cpp`, `rmw_opensplice_cpp`, or `rmw_connext_cpp`.

gdb catches this stack trace:

```
realloc(): invalid pointer

Thread 1 "cyclone_log_tes" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff68cf801 in __GI_abort () at abort.c:79
#2  0x00007ffff6918897 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff6a45b9a "%s\n") at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007ffff691f90a in malloc_printerr (str=str@entry=0x7ffff6a43efd "realloc(): invalid pointer") at malloc.c:5350
#4  0x00007ffff6927eba in __GI___libc_realloc (oldmem=0x555555809910, bytes=1) at malloc.c:3174
#5  0x00007ffff549c432 in rosidl_generator_c__String__assignn () from /opt/ros/eloquent/lib/librosidl_generator_c.so
#6  0x00007ffff3bc1fa5 in rmw_cyclonedds_cpp::StringHelper<rosidl_typesupport_introspection_c__MessageMembers>::assign (field=0x555555809900, deser=...)
    at ./include/rmw_cyclonedds_cpp/TypeSupport.hpp:86
#7  rmw_cyclonedds_cpp::deserialize_field<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > (call_new=false, deser=..., field=0x555555809900, 
    member=0x7ffff03b97e0) at ./include/rmw_cyclonedds_cpp/TypeSupport_impl.hpp:579
#8  rmw_cyclonedds_cpp::TypeSupport<rosidl_typesupport_introspection_c__MessageMembers>::deserializeROSmessage (this=0x5555557c8800, deser=..., members=0x7ffff03b8900, 
    ros_message=0x5555558098f0, call_new=false) at ./include/rmw_cyclonedds_cpp/TypeSupport_impl.hpp:732
#9  0x00007ffff3bdebc1 in rmw_cyclonedds_cpp::TypeSupport<rosidl_typesupport_introspection_c__MessageMembers>::deserializeROSmessage(cycdeser&, void*, std::function<void (cycdeser&)>) (
    this=0x5555557c8800, deser=..., ros_message=ros_message@entry=0x5555558098f0, prefix=...) at ./include/rmw_cyclonedds_cpp/TypeSupport_impl.hpp:907
#10 0x00007ffff3bdcc28 in serdata_rmw_to_sample (dcmn=<optimized out>, sample=0x5555558098f0, bufptr=<optimized out>, buflim=<optimized out>) at ./src/serdata.cpp:282
#11 0x00007ffff393879b in ddsi_serdata_to_sample (buflim=0x0, bufptr=0x0, sample=<optimized out>, d=<optimized out>) at ./src/core/ddsi/include/dds/ddsi/ddsi_serdata.h:199
#12 dds_rhc_take_w_qminv (cond=0x0, handle=<optimized out>, qminv=0, max_samples=1, info_seq=0x7fffffffabc0, values=0x7fffffffabb8, lock=<optimized out>, rhc=0x5555557fc430)
    at ./src/core/ddsc/src/dds_rhc_default.c:2016
#13 dds_rhc_default_take (cond=0x0, handle=<optimized out>, mask=<optimized out>, max_samples=<optimized out>, info_seq=<optimized out>, values=<optimized out>, lock=<optimized out>, 
    rhc=0x5555557fc430) at ./src/core/ddsc/src/dds_rhc_default.c:2640
#14 dds_rhc_default_take_wrap (rhc=0x5555557fc430, lock=<optimized out>, values=0x7fffffffabb8, info_seq=0x7fffffffabc0, max_samples=1, mask=<optimized out>, handle=<optimized out>, 
    cond=0x0) at ./src/core/ddsc/src/dds_rhc_default.c:379
#15 0x00007ffff394187e in dds_rhc_take (cond=<optimized out>, handle=0, mask=128, max_samples=1, info_seq=0x7fffffffabc0, values=0x7fffffffabb8, lock=true, rhc=<optimized out>)
    at ./src/core/ddsc/include/dds/ddsc/dds_rhc.h:83
#16 dds_read_impl (take=take@entry=true, reader_or_condition=<optimized out>, buf=buf@entry=0x7fffffffabb8, bufsz=<optimized out>, maxs=1, si=0x7fffffffabc0, mask=128, hand=0, lock=true, 
    only_reader=false) at ./src/core/ddsc/src/dds_read.c:120
#17 0x00007ffff3941e80 in dds_take (rd_or_cnd=<optimized out>, buf=buf@entry=0x7fffffffabb8, si=si@entry=0x7fffffffabc0, bufsz=bufsz@entry=1, maxs=<optimized out>, maxs@entry=1)
    at ./src/core/ddsc/src/dds_read.c:331
#18 0x00007ffff3bad013 in rmw_take_int (subscription=<optimized out>, ros_message=<optimized out>, taken=0x7fffffffac7f, message_info=0x7fffffffc130) at ./src/rmw_node.cpp:1610
#19 0x00007ffff7860748 in rcl_take () from /opt/ros/eloquent/lib/librcl.so
#20 0x00007ffff7b0fb1a in rclcpp::executor::Executor::execute_subscription(std::shared_ptr<rclcpp::SubscriptionBase>) () from /opt/ros/eloquent/lib/librclcpp.so
#21 0x00007ffff7b11045 in rclcpp::executor::Executor::execute_any_executable(rclcpp::executor::AnyExecutable&) () from /opt/ros/eloquent/lib/librclcpp.so
#22 0x00007ffff7b1771f in rclcpp::executors::SingleThreadedExecutor::spin() () from /opt/ros/eloquent/lib/librclcpp.so
#23 0x00007ffff7b14412 in rclcpp::spin(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>) () from /opt/ros/eloquent/lib/librclcpp.so
#24 0x00007ffff7b146cb in rclcpp::spin(std::shared_ptr<rclcpp::Node>) () from /opt/ros/eloquent/lib/librclcpp.so
#25 0x000055555555e36c in main (argc=1, argv=0x7fffffffd308) at /home/preed/src/ros2/cyclone_log_test/src/cyclone_log_test/src/main.cpp:19
```

eboasson, 2020-03-29:
Reproduced with the old versions (on dashing on macOS), confirmed fixed since Feb 4. This is ultimately a duplicate of #72. The 0.5.1 versions that should become available as apt packages in the near future will include the fix.

Current master is also fine, just make sure you also get a matching Cyclone version (anything from Feb 5, and to get https://github.com/ros2/rmw_cyclonedds/pull/127 to fix a recent mishap causing a build breakage for dashing & eloquent.

pjreed, 2020-03-30:
Ok, thanks!

rotu, 2020-04-15:
@pjreed I think the binary (apt) version of cyclone is out now. Could you confirm and try it out?

pjreed, 2020-04-15:
It looks like 0.5.1 is in the testing repository now:
```bash
$ apt-cache policy ros-eloquent-rmw-cyclonedds-cpp
ros-eloquent-rmw-cyclonedds-cpp:
  Installed: 0.5.1-1bionic.20200413.161250
  Candidate: 0.5.1-1bionic.20200413.161250
  Version table:
 *** 0.5.1-1bionic.20200413.161250 500
        500 http://packages.ros.org/ros2-testing/ubuntu bionic/main amd64 Packages
```

I tested it and that seems to have fixed this problem.  Thanks!

closed 2020-04-15 by pjreed