ros2/rmw_cyclonedds #128: Confusion over versioning

rotu, 2020-03-29:
Unlike most other ROS2 repos, rmw_cyclonedds shares a codebase between different ROS major versions. This causes frequent breakages when someone makes, e.g. a fix for Foxy that breaks Eloquent due to a changed header.

We need to make it more obvious which branch is for which release and prevent these breakages.

rotu, 2020-03-29:
Opinions (@eboasson, @ThijsSassen, @joespeed)? Here's mine.

For now:
1. Add a blurb in the readme or repo description about which branch is for which version
2. Add github actions to confirm that PRs don't break supported versions

For future:
1. A coherent strategy for which branches target which versions. I think it makes sense to only branch on an LTS. So Dashing-Foxy prerelease would be one branch. Foxy-(First LTS prerelease after Foxy) would be another.

joespeed, 2020-03-30:
@rotu great if folks were careful with ros2 changes🙄 @dirk-thomas is in favor of goal to keep bringing latest & greatest rmw to previous ros2 releases. Maybe there's something ros2 can do to facilitate that?

rotu, 2020-03-30:
@joespeed it’s a matter of expectations and feedback. Working on both in #130 and #131

rotu, 2020-03-30:
That wasn’t very clear so I’ll rephrase: developers (such as myself) come to new projects with all sorts of varied experience and assumptions. It’s totally reasonable (abd wring), based on other ROS projects, to assume master only needs to support Foxy. It’s also easy to make a code change that accidentally breaks compatibility. Rather than expect developers to juggle more things in their already addled brains, there are ways to gently steer them in the right direction.

CI is one way. When you submit a PR, if you get fast, actionable feedback about mistakes and breakages, it makes it easier to contribute good code.

Documentation and readmes are another way. If community expectation and the structure of a project are obvious, it’s easier to develop a project and not fight against it. As @dirk-thomas knows very well, I’m often guilty of “fixing“ things before I’ve understood how they work. It causes avoidable frustration for all parties.

It does well to *always* assume developers are capable and well-intentioned. But development is hard and we need all the help we can get :-).

dirk-thomas, 2020-03-30:
> @dirk-thomas is in favor of goal to keep bringing latest & greatest rmw to previous ros2 releases.

Just to clarify this: I am fine leaving it up to you as the maintainers to decide what changes you release into already released distros. The risk of bringing the "latest and greatest" to previous ROS 2 distros is that you potentially also introduce regressions (which is exactly what has happened in the past for CycloneDDS). That is why I advice you to be extra careful when doing this and do proper testing to ensure this doesn't happen.

dirk-thomas, 2020-04-07:
@eboasson Can you please reply what branching strategy this repository uses going forward and make sure it is documented in the README (if it diverges from the common branches used in ROS repos).

If you stick to using a single branch to support all active distros please make sure that pending PRs like #138 are being updated in a timely fashion to be mergable.

rotu, 2020-04-07:
I have an open PR to that PR's dev branch in order to preserve Dashing, Eloquent compatibility. https://github.com/ahcorde/rmw_cyclonedds/pull/4

My 2 cents:

* Moving forward, I think breaking PRs like #138 should be discouraged. It breaks the concept of tracking source code history if you need to manually manage cross-repo dependencies.
* If such PRs cannot be reworked to retain compatibility, then they should be submitted with a list of simultaneous PRs or branches needed to retain compatibility. Preferably CI can be made aware of this list in an automated way so such PRs are built/tested together.

dirk-thomas, 2020-04-07:
> I think breaking PRs like #138 should be discouraged.

The exact purpose of ROS distros is to allow breaking changes when necessary. And this is a very good example where such a break is necessary to make desired improvements. So no, such PRs will for sure happen in the future.

> If such PRs cannot be reworked to retain compatibility, then they should be submitted with a list of simultaneous PRs or branches needed to retain compatibility.

If it is the decision of the maintainers of this repo to aim for one branch being compatibility with multiple distros it will imp also be on them to spend the required effort to make this work. I don't think it should be the responsibility of the contributor to solve the needs-to-work-for-N-distros case i scenarios like the above PR.

rotu, 2020-04-07:
discouraged != forbidden. I'm sure they'll happen, and you're right that this is a case where avoiding breaking changes would have been messy, absent other enhancements (e.g. allowing package xml conditions to use dependency versions)

I'm not saying the package maintainers aren't responsible for maintaining compatibility. I'm saying that this is akin to making sure a PR does not break existing tests and/or coding style. The expectation should generally be that CI *remains green* not that red CI is the normal state of things (as is all too often the case).

eboasson, 2020-04-08:
The main (certainly the original) reason for having a single version of the RMW layer supporting multiple versions of ROS2 was to avoid the need for a lot of backporting. With the merging of #123 Cyclone's RMW layer is now feature complete and the relevance of ongoing work on the RMW implementation to dashing/eloquent will be significantly less.

Moreover, the change from 1 DDS participant per ROS2 node to 1 DDS participant per ROS2 context makes maintaining that compatibility well nigh impossible. Technically, it is possible, but I've started the work and trying to maintain that compatibility will result in a terrible mess. So big a mess that I'd be surprised if anybody would be willing to touch it when there are further RMW interface changes, in fact. That'd be a disaster.

Therefore, unless there are well-argued objections, ``rmw_cyclonedds`` will change to the standard pattern of having a branch per ROS2 release, with the master branch a development branch for the next release. In practice that means a ``dashing`` branch must be created, and the ``eloquent`` one updated. Those branches can be updated to the HEAD of master as far as I am concerned.

Shortly (I expect it this week) there'll be a PR for the changed use of participants. That'll then be a Foxy development only one.

@rotu can you please sort that out so I can use my brain cells for getting that rewrite somewhat correct?

dirk-thomas, 2020-04-08:
> `rmw_cyclonedds` will change to the standard pattern of having a branch per ROS2 release, with the master branch a development branch for the next release. In practice that means a `dashing` branch must be created, and the `eloquent` one updated.

Thanks for clarifying. That sounds like the most reasonable to me.

If the chances are high that Dashing and Eloquent will already get the same updates those two could share the same branch. Especially higher level packages often use a branch `x` for `x` and `y`and only have a separate branch for `z`. PR testing usually happens for x and y if the branch x is targeted. Also backing the branch `x-and-y` is obviously an option.

dirk-thomas, 2020-04-08:
> Shortly (I expect it this week) there'll be a PR for the changed use of participants. That'll then be a Foxy development only one.

It would be good to tag a new version before that split. Also it would be good to have at least separate minor versions for different branches to allow each of the to bump patch versions on changed without colliding (or alternating) version numbers.

rotu, 2020-04-08:
I don't think we should be creating a separate Dashing and Eloquent branch - since Dashing EOL is *after* Eloquent EOL, any bugfixes will need to be backported. The only question in my mind is should we cut a new branch at Foxy prerelease or at Foxy release?

If we need to make separate branches to accommodate compatibility-breaking changes, we need different MAJOR versions for different branches, not minor (https://semver.org/)

dirk-thomas, 2020-04-08:
Since this repo currently uses a version 0.x it can break anything anytime. The suggestion to need a minor bump is simply due to the fact that you need "room" for future patch releases on each branch.

rotu, 2020-04-08:
Even more reason to bump the major version number!

closed 2020-04-09 by rotu