ros2/rmw_cyclonedds #151: some WString tests are failing on Windows debug

dirk-thomas, 2020-04-13:
These are the tests which fail with CycloneDDS:
* https://ci.ros2.org/view/nightly/job/nightly_win_deb/1589/testReport/(root)/projectroot/test_publisher_subscriber__rclpy__rmw_cyclonedds_cpp/
* https://ci.ros2.org/view/nightly/job/nightly_win_deb/1589/testReport/(root)/projectroot/test_publisher_subscriber__rclpy__rclcpp__rmw_cyclonedds_cpp/

rotu, 2020-04-13:
Investigating.

rotu, 2020-04-13:
The failure is rather opaque. I know it's in here, but I can't figure out *where* UnicodeDecodeError is getting thrown.

```
test_subscriber_terminates_in_a_finite_amount_of_time[WStrings] (test_communication.TestPublisherSubscriber)
Test that the subscriber terminates after a finite amount of time. ... [INFO] [test_publisher-23]: process started with pid [3884]
[INFO] [test_subscriber-24]: process started with pid [3460]
[test_subscriber-24] exception in subscriber:
[test_subscriber-24] Traceback (most recent call last):
[test_subscriber-24]   File "C:/ci/ws/src/ros2/system_tests/test_communication/test/subscriber_py.py", line 81, in <module>
[test_subscriber-24]     listener(message_name=args.message_name, namespace=args.namespace)
[test_subscriber-24]   File "C:/ci/ws/src/ros2/system_tests/test_communication/test/subscriber_py.py", line 65, in listener
[test_subscriber-24]     rclpy.spin_once(node)
[test_subscriber-24]   File "C:\ci\ws\install\Lib\site-packages\rclpy\__init__.py", line 171, in spin_once
[test_subscriber-24]     executor.spin_once(timeout_sec=timeout_sec)
[test_subscriber-24]   File "C:\ci\ws\install\Lib\site-packages\rclpy\executors.py", line 685, in spin_once
[test_subscriber-24]     raise handler.exception()
[test_subscriber-24]   File "C:\ci\ws\install\Lib\site-packages\rclpy\task.py", line 239, in __call__
[test_subscriber-24]     self._handler.send(None)
[test_subscriber-24]   File "C:\ci\ws\install\Lib\site-packages\rclpy\executors.py", line 397, in handler
[test_subscriber-24]     arg = take_from_wait_list(entity)
[test_subscriber-24]   File "C:\ci\ws\install\Lib\site-packages\rclpy\executors.py", line 325, in _take_subscription
[test_subscriber-24]     msg = _rclpy.rclpy_take(capsule, sub.msg_type, sub.raw)
[test_subscriber-24] UnicodeDecodeError: 'utf-16-le' codec can't decode bytes in position 0-1: illegal encoding
[ERROR] [test_subscriber-24]: process has died [pid 3460, exit code 1, cmd 'C:\Python37\python_d.exe C:/ci/ws/src/ros2/system_tests/test_communication/test/subscriber_py.py WStrings /test_time_11_49_20'].
[WARNING] [test_publisher-23]: 'SIGINT' sent to process[test_publisher-23] not supported on Windows, escalating to 'SIGTERM'
[INFO] [test_publisher-23]: sending signal 'SIGTERM' to process[test_publisher-23]
[ERROR] [test_publisher-23]: process has died [pid 3884, exit code 1, cmd 'C:\Python37\python_d.exe C:/ci/ws/src/ros2/system_tests/test_communication/test/publisher_py.py WStrings /test_time_11_49_20'].
```

rotu, 2020-04-14:
Hmmm.... Fast-RTPS and Connext seems afflicted by the same failure. This also affects `test_serialize_deserialize`, which has this useful tidbit `serialized_message = b'\x00\x01\x00\x00\x00\x00\x00\x00\x0c\x00\x00\x00\xdd\xdd\x00\x00\xdd\xdd\x00\x00\xdd\xdd\x00\x00\xdd\xdd\x00\x00\xdd...cd\xcd\xcd\xcd\xcd\xcd\xcd\xcd\xcd\xcd\xcd\xcd\xcd\xcd\xcd\xcd\xcd\xcd\xcd\xcd\xcd\xcd\xcd\xcd\xcd\xcd\xcd\xcd\xcd\xcd'`

The first 4 bytes are the RTPS encaspulation header (PLAIN_CDR, LITTLE_ENDIAN). The next 4 bytes are string length = 0 (no data since the string is empty); The next 4 are string length = 10. My guess is `rclpy` is trying to interpret part of the buffer which is not string contents as UTF-16.

Here's the IDL: https://github.com/ros2/test_interface_files/commit/4a20196a92194ee4111f3877eaa975bcbfa98f8d#diff-3f029d743270e7e5fc3fd86e75fd4e02

```
139: test_subscriber_terminates_in_a_finite_amount_of_time[WStrings] (test_communication.TestPublisherSubscriber)
139: Test that the subscriber terminates after a finite amount of time. ... [INFO] [test_publisher-23]: process started with pid [5124]
139: [INFO] [test_subscriber-24]: process started with pid [2476]
139: [test_subscriber-24] exception in subscriber:
139: [test_subscriber-24] Traceback (most recent call last):
139: [test_subscriber-24]   File "C:/ci/ws/src/ros2/system_tests/test_communication/test/subscriber_py.py", line 81, in <module>
139: [test_subscriber-24]     listener(message_name=args.message_name, namespace=args.namespace)
139: [test_subscriber-24]   File "C:/ci/ws/src/ros2/system_tests/test_communication/test/subscriber_py.py", line 65, in listener
139: [test_subscriber-24]     rclpy.spin_once(node)
139: [test_subscriber-24]   File "C:\ci\ws\install\Lib\site-packages\rclpy\__init__.py", line 171, in spin_once
139: [test_subscriber-24]     executor.spin_once(timeout_sec=timeout_sec)
139: [test_subscriber-24]   File "C:\ci\ws\install\Lib\site-packages\rclpy\executors.py", line 685, in spin_once
139: [test_subscriber-24]     raise handler.exception()
139: [test_subscriber-24]   File "C:\ci\ws\install\Lib\site-packages\rclpy\task.py", line 239, in __call__
139: [test_subscriber-24]     self._handler.send(None)
139: [test_subscriber-24]   File "C:\ci\ws\install\Lib\site-packages\rclpy\executors.py", line 397, in handler
139: [test_subscriber-24]     arg = take_from_wait_list(entity)
139: [test_subscriber-24]   File "C:\ci\ws\install\Lib\site-packages\rclpy\executors.py", line 325, in _take_subscription
139: [test_subscriber-24]     msg = _rclpy.rclpy_take(capsule, sub.msg_type, sub.raw)
139: [test_subscriber-24] UnicodeDecodeError: 'utf-16-le' codec can't decode bytes in position 0-1: illegal encoding
139: [ERROR] [test_subscriber-24]: process has died [pid 2476, exit code 1, cmd 'C:\Python37\python_d.exe C:/ci/ws/src/ros2/system_tests/test_communication/test/subscriber_py.py WStrings /test_time_11_30_30'].
139: [WARNING] [test_publisher-23]: 'SIGINT' sent to process[test_publisher-23] not supported on Windows, escalating to 'SIGTERM'
139: [INFO] [test_publisher-23]: sending signal 'SIGTERM' to process[test_publisher-23]
139: [ERROR] [test_publisher-23]: process has died [pid 5124, exit code 1, cmd 'C:\Python37\python_d.exe C:/ci/ws/src/ros2/system_tests/test_communication/test/publisher_py.py WStrings /test_time_11_30_30'].
139: ok
139: 
139: ----------------------------------------------------------------------
139: Ran 1 test in 0.788s
139: 
139: OK
139: test_processes_finished_gracefully[WStrings] (test_communication.TestPublisherSubscriberAfterShutdown)
139: Test that both executables finished gracefully. ... FAIL
139: 
139: ======================================================================
139: FAIL: test_processes_finished_gracefully[WStrings] (test_communication.TestPublisherSubscriberAfterShutdown)
139: Test that both executables finished gracefully.
139: ----------------------------------------------------------------------
139: Traceback (most recent call last):
139:   File "C:\ci\ws\build\test_communication\test_publisher_subscriber__rclpy__rmw_fastrtps_cpp_Debug.py", line 77, in test_processes_finished_gracefully
139:     'test_subscriber'
139:   File "C:\ci\ws\install\Lib\site-packages\launch_testing\asserts\assert_exit_codes.py", line 64, in assertExitCodes
139:     info.returncode
139: AssertionError: Proc test_subscriber-24 exited with code 1
139: 
139: ----------------------------------------------------------------------
139: Ran 1 test in 0.002s
139: 
139: FAILED (failures=1)
139: -- run_test.py: return code 1
139: -- run_test.py: verify result file 'C:/ci/ws/build/test_communication/test_results/test_communication/test_publisher_subscriber__rclpy__rmw_fastrtps_cpp.xunit.xml'
```

```
15: test_subscriber_terminates_in_a_finite_amount_of_time[WStrings] (test_communication.TestPublisherSubscriber)
15: Test that the subscriber terminates after a finite amount of time. ... [INFO] [test_publisher-23]: process started with pid [3988]
15: [INFO] [test_subscriber-24]: process started with pid [3876]
15: [test_subscriber-24] exception in subscriber:
15: [test_subscriber-24] Traceback (most recent call last):
15: [test_subscriber-24]   File "C:/ci/ws/src/ros2/system_tests/test_communication/test/subscriber_py.py", line 81, in <module>
15: [test_subscriber-24]     listener(message_name=args.message_name, namespace=args.namespace)
[... cut ...]
15: Traceback (most recent call last):
15:   File "C:\ci\ws\build\test_communication\test_publisher_subscriber__rclpy__rmw_connext_cpp_Debug.py", line 77, in test_processes_finished_gracefully
15:     'test_subscriber'
15:   File "C:\ci\ws\install\Lib\site-packages\launch_testing\asserts\assert_exit_codes.py", line 64, in assertExitCodes
15:     info.returncode
15: AssertionError: Proc test_subscriber-24 exited with code 1
15: 
15: ----------------------------------------------------------------------
15: Ran 1 test in 0.009s
```

rotu, 2020-04-14:
Pretty sure the bug is in https://github.com/ros2/rosidl_python/blob/master/rosidl_generator_py/resource/_msg_support.c.em so moving this issue to https://github.com/ros2/rosidl_python/issues/104

closed 2020-04-14 by rotu