ros2/rmw_cyclonedds #152: reenabled WString cross vendor tests

dirk-thomas, 2020-04-13:
Once #43 has been resolved the tests which have been disabled in ros2/system_tests#421 should be reenabled.

rotu, 2020-04-13:
Thanks. Please put this issue in the ros2/system_tests where it properly belongs. It is not actionable here.

closed 2020-04-13 by rotu

dirk-thomas, 2020-04-14:
Since it is not actionable in `system_tests` (since the actual fix has to be made in this repo and the maintainers of this repository will have the responsibility to follow up on this once there blocking ticket is resolved) I won't move it there. If you don't want to keep this ticket here please remember to reenable the tests once they would pass.

rotu, 2020-04-14:
Yes, #43 is tracking the underlying issue already. I would suggest creating an issue there and tracking it with whatever methods you currently do for interdependent development.

dirk-thomas, 2020-04-14:
> the maintainers of this repository will have the responsibility to follow up on this once the blocking ticket is resolved

rotu, 2020-04-14:
Yes, you can transfer this issue to that repo and assign it to me if you like. Would you like the tests re-enabled as soon as Cyclone is compliant with the spec or when the other DDS's are as well?

dirk-thomas, 2020-04-14:
Since I don't expect CycloneDDS maintainers to check ticket in the `ros2/system_tests` repo and the the maintainers of that repo can't act on this I won't move the ticket into that repo (I thought I made that clear in the previous comments already). Please accept that decision.

Feel free to do whatever your prefer to track this task (reopen this ticket, make a note on #43, or track it somewhere outside of GitHub) - whatever works for your workflow.

> Would you like the tests re-enabled as soon as Cyclone is compliant with the spec or when the other DDS's are as well?

Whenever a disabled / skipped test will pass it should be reenabled.

rotu, 2020-04-14:
I'd prefer to track this ticket in the same repo as the code that needs changing. That way merging a fix PR can autoclose the issue.