ros2/rmw_cyclonedds #167: Implement received timestamp

iluetkeb, 2020-04-24:
Thanks @iluetkeb for helping with implementing this also for cyclone dds, it really makes a difference to me. I just wanted to add a suggestion on how to get a receive time stamp without having to rely on Cyclone to provide it.

The RMW layer relies on a custom sample implementation to perform direct ROS2-to-CDR conversion and to get access to the serialized form for `rmw_take_serialized_message` and any received message will be turned into a sample instance as soon as it is ready to be inserted in the reader history caches. If that’s what you mean by “the receive time stamp” (as I pointed out in https://github.com/ros2/design/issues/259#issuecomment-612440879 there are very many versions), then one way to do it would be to store that time stamp in this custom sample: just add it to https://github.com/ros2/rmw_cyclonedds/blob/2638f4abe87982f45e5301e90fc4200371a613c3/rmw_cyclonedds_cpp/src/serdata.hpp#L48-L61

It won’t be returned in the sample info because cyclone currently only implements the fields defined in the standard, but that can be worked around by using `dds_takecdr` instead of `dds_take`, like when taking a serialized message:
https://github.com/ros2/rmw_cyclonedds/blob/2638f4abe87982f45e5301e90fc4200371a613c3/rmw_cyclonedds_cpp/src/rmw_node.cpp#L2178
That gives you a *reference* to the sample that you can then deserialize by calling `ddsi_serdata_to_sample` on it. That way you get everything.

(I don’t know what the rules are for loaned samples, but with this sample implementation, and for those topics where the CDR representation matches the in-memory representation and where there are no endianness issues, the same trick would be feasible.)

_Originally posted by @eboasson in https://github.com/ros2/rmw_cyclonedds/pull/163#issuecomment-618826057_

iluetkeb, 2020-04-24:
Thanks @eboasson, I've created an issue from your suggestion for future reference.

ZhenshengLee, 2022-05-25:
@iluetkeb 

https://github.com/hsgwa/cyclonedds/tree/devel_timestamp_for_ros2 and 
https://github.com/hsgwa/rmw_cyclonedds/commits/devel_timestamp_for_ros2

do the work of this issue, but the commit will cause rviz2 crash on startup.