ros2/rmw_cyclonedds #183: test_parameter_server_cpp__rmw_cyclonedds_cpp is failing most of the time on linux

tfoote, 2020-05-12:
In triaging CI failures I've identified this test as failing most of the time. 

Linux Debug: https://ci.ros2.org/job/nightly_linux_debug/1550/testReport/(root)/projectroot/test_parameter_server_cpp__rmw_cyclonedds_cpp/history/

Linux Release: https://ci.ros2.org/job/nightly_linux_release/1539/testReport/(root)/projectroot/test_parameter_server_cpp__rmw_cyclonedds_cpp/history/

LInux Aarch64: https://ci.ros2.org/job/nightly_linux-aarch64_release/1159/testReport/(root)/projectroot/test_parameter_server_cpp__rmw_cyclonedds_cpp/history/

Here's an example of the failure.
<details>

```
41: Test command: /home/jenkins-agent/workspace/nightly_linux_debug/venv/bin/python3 "-u" "/home/jenkins-agent/workspace/nightly_linux_debug/ws/install/ament_cmake_test/share/ament_cmake_test/cmake/run_test.py" "/home/jenkins-agent/workspace/nightly_linux_debug/ws/build/test_rclcpp/test_results/test_rclcpp/test_parameter_server_cpp__rmw_cyclonedds_cpp.xunit.xml" "--package-name" "test_rclcpp" "--output-file" "/home/jenkins-agent/workspace/nightly_linux_debug/ws/build/test_rclcpp/launch_test/CHANGEME.txt" "--env" "RCL_ASSERT_RMW_ID_MATCHES=rmw_cyclonedds_cpp" "RMW_IMPLEMENTATION=rmw_cyclonedds_cpp" "--append-env" "LD_LIBRARY_PATH=/home/jenkins-agent/workspace/nightly_linux_debug/ws/build/test_rclcpp" "--command" "/home/jenkins-agent/workspace/nightly_linux_debug/venv/bin/python3" "-m" "launch_testing.launch_test" "/home/jenkins-agent/workspace/nightly_linux_debug/ws/build/test_rclcpp/test_parameter_server_cpp__rmw_cyclonedds_cpp_Debug.py" "--junit-xml=/home/jenkins-agent/workspace/nightly_linux_debug/ws/build/test_rclcpp/test_results/test_rclcpp/test_parameter_server_cpp__rmw_cyclonedds_cpp.xunit.xml" "--package-name=test_rclcpp"
41: Test timeout computed to be: 60
41: -- run_test.py: extra environment variables:
41:  - RCL_ASSERT_RMW_ID_MATCHES=rmw_cyclonedds_cpp
41:  - RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
41: -- run_test.py: extra environment variables to append:
41:  - LD_LIBRARY_PATH+=/home/jenkins-agent/workspace/nightly_linux_debug/ws/build/test_rclcpp
41: -- run_test.py: invoking following command in '/home/jenkins-agent/workspace/nightly_linux_debug/ws/build/test_rclcpp':
41:  - /home/jenkins-agent/workspace/nightly_linux_debug/venv/bin/python3 -m launch_testing.launch_test /home/jenkins-agent/workspace/nightly_linux_debug/ws/build/test_rclcpp/test_parameter_server_cpp__rmw_cyclonedds_cpp_Debug.py --junit-xml=/home/jenkins-agent/workspace/nightly_linux_debug/ws/build/test_rclcpp/test_results/test_rclcpp/test_parameter_server_cpp__rmw_cyclonedds_cpp.xunit.xml --package-name=test_rclcpp
41: [INFO] [launch]: All log files can be found below /home/rosbuild/.ros/log/2020-05-11-12-09-06-131831-b25d421f3bc4-3879
41: [INFO] [launch]: Default logging verbosity is set to INFO
41: test_parameter_server_cpp (test_rclcpp.TestTwoExecutables)
41: Test that the second executable terminates after a finite amount of time. ... [INFO] [test_parameters_server_cpp-1]: process started with pid [3882]
41: [INFO] [test_remote_parameters_cpp-2]: process started with pid [3884]
41: [test_remote_parameters_cpp-2] Running main() from /home/jenkins-agent/workspace/nightly_linux_debug/ws/install/gtest_vendor/src/gtest_vendor/src/gtest_main.cc
41: [test_remote_parameters_cpp-2] [==========] Running 6 tests from 2 test cases.
41: [test_remote_parameters_cpp-2] [----------] Global test environment set-up.
41: [test_remote_parameters_cpp-2] [----------] 3 tests from parameters__rmw_implementation
41: [test_remote_parameters_cpp-2] [ RUN      ] parameters__rmw_implementation.test_remote_parameters_async
41: [test_remote_parameters_cpp-2] Setting parameters
41: [test_remote_parameters_cpp-2] Got set_parameters result
41: [test_remote_parameters_cpp-2] Listing parameters with recursive depth
41: [test_remote_parameters_cpp-2] Listing parameters with depth 1
41: [test_remote_parameters_cpp-2] Listing parameters with depth 2
41: [test_remote_parameters_cpp-2] Getting parameters
41: [test_remote_parameters_cpp-2] Getting nonexistent parameters
41: [test_remote_parameters_cpp-2] Listing parameters with recursive depth
41: [test_remote_parameters_cpp-2] [       OK ] parameters__rmw_implementation.test_remote_parameters_async (427 ms)
41: [test_remote_parameters_cpp-2] [ RUN      ] parameters__rmw_implementation.test_remote_parameters_sync
41: [test_remote_parameters_cpp-2] Setting parameters
41: [test_remote_parameters_cpp-2] Got set_parameters result
41: [test_remote_parameters_cpp-2] Listing parameters with recursive depth
41: [test_remote_parameters_cpp-2] Listing parameters with depth of 1
41: [test_remote_parameters_cpp-2] Listing parameters with depth of 2
41: [test_remote_parameters_cpp-2] Getting parameters
41: [test_remote_parameters_cpp-2] Getting nonexistent parameters
41: [test_remote_parameters_cpp-2] Listing parameters with recursive depth
41: [test_remote_parameters_cpp-2] Listing parameters with depth 100
41: [test_remote_parameters_cpp-2] Listing parameters with depth 1
41: [test_remote_parameters_cpp-2] [       OK ] parameters__rmw_implementation.test_remote_parameters_sync (423 ms)
41: [test_remote_parameters_cpp-2] [ RUN      ] parameters__rmw_implementation.test_set_remote_parameters_atomically_sync
41: [test_remote_parameters_cpp-2] Setting parameters atomically
41: [test_remote_parameters_cpp-2] Got set_parameters_atomically result
41: [test_remote_parameters_cpp-2] Listing parameters with recursive depth
41: [test_remote_parameters_cpp-2] Listing parameters with depth of 1
41: [test_remote_parameters_cpp-2] Listing parameters with depth of 2
41: [test_remote_parameters_cpp-2] Getting parameters
41: [test_remote_parameters_cpp-2] Getting nonexistent parameters
41: [test_remote_parameters_cpp-2] Listing parameters with recursive depth
41: [test_remote_parameters_cpp-2] Listing parameters with depth 100
41: [test_remote_parameters_cpp-2] Listing parameters with depth 1
41: [test_remote_parameters_cpp-2] [       OK ] parameters__rmw_implementation.test_set_remote_parameters_atomically_sync (376 ms)
41: [test_remote_parameters_cpp-2] [----------] 3 tests from parameters__rmw_implementation (1226 ms total)
41: [test_remote_parameters_cpp-2] 
41: [test_remote_parameters_cpp-2] [----------] 3 tests from parameters_must_declare__rmw_implementation
41: [test_remote_parameters_cpp-2] [ RUN      ] parameters_must_declare__rmw_implementation.test_remote_parameters_async
41/81 Test #41: test_parameter_server_cpp__rmw_cyclonedds_cpp ...................***Timeout  60.05 sec
```
</details>

tfoote, 2020-05-14:
I have disabled this test for our standard builds using the xfail tag. This can be closed once that's removed. 

https://github.com/ros2/system_tests/blob/20f85e7395f9771671918a6e868013ef93442f03/test_rclcpp/CMakeLists.txt#L299

eboasson, 2020-05-16:
I've been really swamped the past few days, but I did see it and I will look into it as soon as I can manage to do so. Service invocation has been a bit of an issue for some time (e.g., see #74) and it annoys me greatly ...

sloretz, 2021-04-02:
It looks like this issue has been solved. The xfail tag has been removed in https://github.com/ros2/system_tests/pull/467

I'll close this for now.

closed 2021-04-02 by sloretz