ros2/rmw_cyclonedds #188: Fix serialization on non-32-bit, big-endian systems - dashing-eloquent

dodsonmg, 2020-05-21:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 18.04

- Installation type:
Source installation

- Version or commit hash:
`dashing-eloquent` branch provided with ROS2 dashing

- DDS implementation:
rmw_implementation = cyclonedds

- Client library (if applicable):
N/A

#### Steps to reproduce issue
This is an issue with serialisation on big-endian systems and is discussed here:

[eclipse-cyclonedds/cyclonedds#488 ](https://github.com/eclipse-cyclonedds/cyclonedds/issues/488)

The fix for this has already been merged with `master` (#159), and I was hoping it could be pulled into `dashing-eloquent`.

I think the patch used for #159 on `master` can be merged directly into `dashing-eloquent`.  I'm happy to submit the PR myself.

eboasson, 2020-05-21:
I'm in favour. Please do submit a PR!

dodsonmg, 2020-05-21:
Thanks!  PR submitted.  #189.

closed 2020-05-21 by dodsonmg