ros2/rmw_cyclonedds #191: Improve service discovery

hidmic, 2020-05-29:
## Feature request

#### Feature description
Follow-up issue after #187. Prior to that PR, services' discovery would rely on built-in request/reply topics discovery, which can and did lead to races (see ros2/ros2#922). A short-term workaround was then introduced by #187, but not without caveats i.e. blocking behavior of rmw_send_response, non-standard exchange IDs.

[OMG DDS-RPC 1.0 spec](https://www.omg.org/spec/DDS-RPC/1.0/PDF) describes both the issues associated with independent discovery of request/reply topics (see Basic Service Mapping under section 7.6.1) and an algorithm to prevent them (see Enhanced Service Mapping under section 7.6.2). Services' implementation in `rmw_cyclonedds_cpp` could and probably should move in that direction -- solving #184 in the process if https://github.com/ros2/rmw_fastrtps/issues/392 is simultaneously addressed.

eboasson, 2020-05-29:
Just FYI: there are currently no plans for adding DDS-RPC to Cyclone DDS — the project has to make choices and this is simply not high up on the prriority list. (One never knows what the future may bring, perhaps someone decides to do the work and contribute it, and that of course would completely change the situation.) What is high on the priority list is providing an interface that allows for an elegant solution to avoiding the race conditions in service invocation.

hidmic, 2020-05-29:
Even if compliance with DDS-RPC cannot be achieved, the discovery algorithm it suggests would provide for the behavior we expect. Some of your suggestions in #187 do not fall far from the tree either.