ros2/rmw_cyclonedds #212: failing test checking reported depth of QoS profile

dirk-thomas, 2020-07-27:
This nightly test is failing for CycloneDDS: http://build.ros2.org/view/Rci/job/Rci__nightly-cyclonedds_ubuntu_focal_amd64/29/testReport/rclcpp/TestNodeGraph/get_info_by_topic/

eboasson, 2020-07-28:
I'm tempted to say Cyclone is correct (the publisher history depth is indeed 1) and the test expectations are incorrect. However, in reality it is not that simple, because the spec'd DDS discovery data only contains QoS settings relevant for matching & interpreting data. The history setting is purely local and hence not part of the spec'd discovery data. Cyclone uses a vendor-specific extensions for the QoS settings left out of the standard precisely because it makes sense to have all the settings when troubleshooting or analyzing system performance.

Given that it can't always work, always setting the history to `RMW_QOS_POLICY_HISTORY_UNKNOWN` with depth = 0 is therefore a sensible option, but I do think the test should then check the policy as well. Another option would be to set it to the true value if known.

dirk-thomas, 2020-07-28:
Thanks for the feedback / context.

I will go ahead and close this in favor of patching the test in ros2/rclcpp#1245.

closed 2020-07-28 by dirk-thomas