ros2/rmw_cyclonedds #216: Installation instructions for ddsperf on Mac?

mikeferguson, 2020-08-10:
## Bug report

**Required Info:**

- Operating System:
  - MacOSX Catalina
- Installation type:
  - Source
- Version or commit hash:
  - e0bfe464741eaa1691a5ed5a1f4c814ad7164062
- DDS implementation:
  - rmw_cyclonedds_cpp
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
I've installed from source - and the debugging instructions in the README say that ddsperf is included - but it doesn't seem to be installed/built.

#### Expected behavior
ddsperf would be available.

How do we go about installing ddsperf tool? (since it appears to need some config to build, this should probably be in the README)

eboasson, 2020-08-10:
Something needs to be done, I agree. Having a readme that says ddsperf is there when it isn't won't do.

The probable cause is that you haven't got a JDK or Maven available in your path: it is possible to build Cyclone sans IDL compiler and, because of that, without ddsperf. A colcon-based build sets the `BUILD_IDLC` variable to `AUTO`, which causes it to build everything if Maven and a JDK are available, but excluding the IDL compiler it they aren't. So `brew install maven` should do the trick.

We did this this after discovering the Java code caused some trouble for people building ROS 2 while not actually contributing much to a ROS 2 installation, as the Cyclone RMW layer has a fully dynamic serializer that relies on the introspection type support.

So how does one go about fixing this?
* Remove the reference to ddsperf from the README — quite the cop-out;
* Point out more clearly in the README that this dependency exists;
* Include the (very small) amount of IDL compiler output in the ddsperf sources;
* Replace the Java-based IDL compiler by a C-based one that is always available.

A new IDL compiler is under active development and will eliminate the Java dependency and it should come online within a few weeks. Until then, a bit more information in the README would be wise, but for now, please give `brew install maven` a try. Beware that CMake caches a lot, the easiest way to be sure that it reconfigures from scratch is to delete Cyclone DDS' build directory (e.g., `ros2_ws/build/cyclonedds`).

mikeferguson, 2020-08-11:
Thanks! So, I can confirm the following worked:

 * remove the build directory as mentioned
 * brew install maven
 * brew cask install java (first attempt to rebuild didn't work - even though openjdk was installed, and I followed the notes to add to path, it didn't seem to be found by CMake - this version was found).

And in getting _ddsperf_ to work, I found what the issue with my underlying system was - but I'll open a separate issue to discuss that.

eboasson, 2020-08-11:
I'm glad you solved the problems, but I am somewhat surprised that you still had to jump through hoops to get there. I use Catalina myself and simply have the `openjdk` package installed with homebrew. No options, nothing special.

I guess you don't feel like going back to the non-working situation, but if you (or anyone else who happens to be reading this while running into the same or a similar problem) happen to have some logs from CMake and some information on where the JDK is installed, or know what to change in the CMake files, then we may be able to update the CMake files so it finds it.

> And in getting ddsperf to work, I found what the issue with my underlying system was - but I'll open a separate issue to discuss that.

Please do!

mikeferguson, 2020-08-11:
Ok - I removed the java package to rebuild and get the error message, but now it's not failing. I've updated my comment above to note that it's probably not needed (perhaps adding/removing "java" linked something with openjdk somewhere...)

mikeferguson, 2024-12-11:
The info here is now quite out of date - and Mac is no longer a well supported ROS platform - going to close this as I'm not even sure if it is still relevant