ros2/rmw_cyclonedds #219: Memory leak about not to call `placement delete` after calling `placement new`

iuhilnehc-ynos, 2020-08-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - source
- Version or commit hash:
  - 49a994e41808b2ebab37b4fc841792dbf6194aa4
- DDS implementation:
  - rmw_cyclonedds_cpp
- Client library (if applicable):
  - rcl

#### Steps to reproduce issue

```
$ colcon build --cmake-args -DCMAKE_BUILD_TYPE=Debug --packages-up-to rcl
$ . install/setup.bash
$ valgrind --leak-check=full ./build/rcl/test/test_node__rmw_cyclonedds_cpp --gtest_filter=*test_rcl_node_accessors*
```

#### Expected behavior

No memory leak about [x = std::string(data + pos, sz - 1);](https://github.com/ros2/rmw_cyclonedds/blob/49a994e41808b2ebab37b4fc841792dbf6194aa4/rmw_cyclonedds_cpp/include/rmw_cyclonedds_cpp/serdes.hpp#L165) in `cycdeser::deserialize`

#### Actual behavior

```
==668117== 29 bytes in 1 blocks are definitely lost in loss record 2 of 25
==668117==    at 0x483DE63: operator new(unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==668117==    by 0x255608: void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char const*>(char const*, char const*, std::forward_iterator_tag) (basic_string.tcc:219)
==668117==    by 0x5176382: cycdeser::deserialize(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) (serdes.hpp:165)
==668117==    by 0x5175CC4: cycdeser::operator>>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) (serdes.hpp:113)
==668117==    by 0x517724A: void rmw_cyclonedds_cpp::deserialize_field<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(rosidl_typesupport_introspection_cpp::MessageMember const*, void*, cycdeser&, bool) (TypeSupport_impl.hpp:281)
==668117==    by 0x518220C: rmw_cyclonedds_cpp::TypeSupport<rosidl_typesupport_introspection_cpp::MessageMembers>::deserializeROSmessage(cycdeser&, rosidl_typesupport_introspection_cpp::MessageMembers const*, void*, bool) (TypeSupport_impl.hpp:514)
==668117==    by 0x5182350: rmw_cyclonedds_cpp::TypeSupport<rosidl_typesupport_introspection_cpp::MessageMembers>::deserializeROSmessage(cycdeser&, rosidl_typesupport_introspection_cpp::MessageMembers const*, void*, bool) (TypeSupport_impl.hpp:542)
==668117==    by 0x517BB3E: rmw_cyclonedds_cpp::TypeSupport<rosidl_typesupport_introspection_cpp::MessageMembers>::deserializeROSmessage(cycdeser&, void*, std::function<void (cycdeser&)>) (TypeSupport_impl.hpp:669)
==668117==    by 0x51D10CF: serdata_rmw_to_sample(ddsi_serdata const*, void*, void**, void*) (serdata.cpp:288)
==668117==    by 0x57EDAD8: ddsi_serdata_to_sample (ddsi_serdata.h:230)
==668117==    by 0x586E9EC: read_take_to_sample (dds_rhc_default.c:1956)
==668117==    by 0x586F15B: take_w_qminv_inst (dds_rhc_default.c:2091)

... <and another type of memory leak about `create_message_type_support` that not destroyed which I've already known how to fix it.>

```

#### Additional information

- backtrace of main steps:

    <details><summary>new(field) std::string()</summary>
    <p>

    ```shell
    (gdb) bt
    #0  rmw_cyclonedds_cpp::deserialize_field<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > (
        member=0x7ffff754a5c0 <rmw_dds_common::msg::rosidl_typesupport_introspection_cpp::NodeEntitiesInfo_message_member_array>, field=0x7fffdc0018d0, deser=..., call_new=true)
        at /home/ros2_ws/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/include/rmw_cyclonedds_cpp/TypeSupport_impl.hpp:279
    #1  0x00007ffff76a420d in rmw_cyclonedds_cpp::TypeSupport<rosidl_typesupport_introspection_cpp::MessageMembers>::deserializeROSmessage (this=0x55555570d800, deser=..., 
        members=0x7ffff7549d40 <rmw_dds_common::msg::rosidl_typesupport_introspection_cpp::NodeEntitiesInfo_message_members>, ros_message=0x7fffdc0018d0, call_new=true)
        at /home/ros2_ws/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/include/rmw_cyclonedds_cpp/TypeSupport_impl.hpp:514
    #2  0x00007ffff76a4351 in rmw_cyclonedds_cpp::TypeSupport<rosidl_typesupport_introspection_cpp::MessageMembers>::deserializeROSmessage (this=0x55555570d800, deser=..., 
        members=0x7ffff7549d80 <rmw_dds_common::msg::rosidl_typesupport_introspection_cpp::ParticipantEntitiesInfo_message_members>, ros_message=0x7ffff1bb5ad0, call_new=false)
        at /home/ros2_ws/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/include/rmw_cyclonedds_cpp/TypeSupport_impl.hpp:542
    #3  0x00007ffff769db3f in rmw_cyclonedds_cpp::TypeSupport<rosidl_typesupport_introspection_cpp::MessageMembers>::deserializeROSmessage(cycdeser&, void*, std::function<void (cycdeser&)>) (this=0x55555570d800, deser=..., ros_message=0x7ffff1bb5ad0, prefix=...)
        at /home/ros2_ws/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/include/rmw_cyclonedds_cpp/TypeSupport_impl.hpp:669
    #4  0x00007ffff76f30d0 in serdata_rmw_to_sample (dcmn=0x555555721210, sample=0x7ffff1bb5ad0, bufptr=0x0, buflim=0x0)
        at /home/ros2_ws/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/serdata.cpp:288
[... 111 lines cut ...]
    #27 0x00007ffff7402cb4 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
    #28 0x00007ffff72f7609 in start_thread (arg=<optimized out>) at pthread_create.c:477
    #29 0x00007ffff721e103 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
    ```

    </p>
    </details>

- main sequence

    ```
        handle_ParticipantEntitiesInfo
            ParticipantEntitiesInfo msg;
            rmw_take->...->take_w_qminv_inst->
                to_sample
                  serdata_rmw_to_sample   // call `new(field) std::string()` inside
                free_sample
                  serdata_rmw_free        // It seems it's the best place to call `operator delete` here, but `msg` will be used after it.
                                          // NOTE: I had tried to save the `field` pointer into a list, and delete it. 
            impl->common.graph_cache.update_participant_entities(msg);
    ```

Could somebody give me a suggestion?

This issue is related to https://github.com/ros2/rcl/issues/721, but I think it belongs to rmw_cyclonedds.

eboasson, 2020-08-13:
@iuhilnehc-ynos thank you for looking into this!

I guess I had assumed that:
1. Leaving the scope containing `msg` would run a destructor that frees the memory;
2. If two (or more) messages get processed in the loop the deserializer would reuse the memory allocated for the preceding message.

I may have been thinking of how Cyclone DDS's native code for copying samples out to C applications handles things in making the second assumption. For some background: I borrowed the deserializer code from the dynamic Fast-RTPS RMW implementation some two years ago, hacking it to fit my needs and without ever really diving into the details, and especially without understanding what/when `call_new` is needed. It is definitely possible that I either screwed up in that process or that I copied a bug that's been fixed a long time ago.

If the destructor does free the memory, the easy solution would be to change the scope of `msg` so it gets destructed after each successful call to `take`. It could even be that that's the intent behind the RMW interface without my realizing it.

The calls to `serdata_rmw_free` release the internal message representation in Cyclone DDS after the message has been deserialized in the ROS message, so that's not the place to free this memory.

iuhilnehc-ynos, 2020-08-13:
@eboasson 
thanks for your quick reply.

> deserializer code from the dynamic Fast-RTPS RMW implementation

I quickly checked the [rmw_fastrtps](https://github.com/ros2/rmw_fastrtps/blob/0d039e6d0f944448ba2d26a6d0b83795745f60fc/rmw_fastrtps_dynamic_cpp/include/rmw_fastrtps_dynamic_cpp/TypeSupport_impl.hpp#L761), and found it still uses `new(field) std::string()`. The next thing I need to do is check how/where `dynamic Fast-RTPS RMW implementation` to free it.

> The calls to serdata_rmw_free release the internal message representation in Cyclone DDS after the message has been deserialized in the ROS message, so that's not the place to free this memory.

I agree with this.

iuhilnehc-ynos, 2020-08-24:
@eboasson

> If the destructor does free the memory, the easy solution would be to change the scope of msg so it gets destructed after each successful call to take

Sorry, I lost this important information. And after checking, I think it's the reason why string not deleted.
I have made a [PR](https://github.com/ros2/rmw_cyclonedds/pull/224), could you help me review it, please?

ivanpauno, 2020-08-26:
Summary of discussion in https://github.com/ros2/rmw_cyclonedds/pull/224, which works around the issue but doesn't fix the problem. The problems are in the typesupport implementation:
- Doing a `reinterpret_cast` of a `std::vector<MessageT>` to a `std::vector<unsigned char> *` is extremely unsafe and may not work in many stl implementations (I'm even surprised this is working at all, isn't size/capacity corrupted? when resizing).
- Placement new should never be called on a passed message (used after resizing a vector because of the unsafe reinterpret_cast above, see [here](https://github.com/ros2/rmw_cyclonedds/blob/217259a9d76357b345b04194556a75824fd8c009/rmw_cyclonedds_cpp/include/rmw_cyclonedds_cpp/TypeSupport_impl.hpp#L538)), that corrupts the memory of previously allocated objects and causes leaks.

ivanpauno, 2020-08-26:
The correct way of resizing and iterating is: https://github.com/ros2/rosidl/blob/279731bf72ab1f60553eb381fba7ba3f38947d44/rosidl_typesupport_introspection_cpp/include/rosidl_typesupport_introspection_cpp/message_introspection.hpp#L42-L44

iuhilnehc-ynos, 2020-08-27:
@ivanpauno 

Thanks for your explanation.
I believe you're right. At https://github.com/ros2/rmw_cyclonedds/blob/217259a9d76357b345b04194556a75824fd8c009/rmw_cyclonedds_cpp/include/rmw_cyclonedds_cpp/TypeSupport_impl.hpp#L430, it should call resize_function and get_function of MessageMember if these functions exist.

I checked the `participant_entities_info__type_support.cpp`
```
static const ::rosidl_typesupport_introspection_cpp::MessageMember ParticipantEntitiesInfo_message_member_array[2] = {
  ...
  {
    "node_entities_info_seq",  // name
    ::rosidl_typesupport_introspection_cpp::ROS_TYPE_MESSAGE,  // type
    0,  // upper bound of string
    ::rosidl_typesupport_introspection_cpp::get_message_type_support_handle<rmw_dds_common::msg::NodeEntitiesInfo>(),  // members of sub message
    true,  // is array
    0,  // array size
    false,  // is upper bound
    offsetof(rmw_dds_common::msg::ParticipantEntitiesInfo, node_entities_info_seq),  // bytes offset in struct
    nullptr,  // default value
    size_function__ParticipantEntitiesInfo__node_entities_info_seq,  // size() function pointer
    get_const_function__ParticipantEntitiesInfo__node_entities_info_seq,  // get_const(index) function pointer
    get_function__ParticipantEntitiesInfo__node_entities_info_seq,  // get(index) function pointer
    resize_function__ParticipantEntitiesInfo__node_entities_info_seq  // resize(index) function pointer
  }
```
Currently, these functions(such as, get_function__ParticipantEntitiesInfo__node_entities_info_seq and resize_function__ParticipantEntitiesInfo__node_entities_info_seq) are just defined. There is no function to call them.
I'll try this way.

iuhilnehc-ynos, 2020-08-27:
@ivanpauno 

I have checked it. It's working. (revert my previous [commit](https://github.com/ros2/rmw_cyclonedds/pull/224) and just use the following patch)

```
diff --git a/rmw_cyclonedds_cpp/include/rmw_cyclonedds_cpp/TypeSupport_impl.hpp b/rmw_cyclonedds_cpp/include/rmw_cyclonedds_cpp/TypeSupport_impl.hpp
index bf2d229..d4e8782 100644
--- a/rmw_cyclonedds_cpp/include/rmw_cyclonedds_cpp/TypeSupport_impl.hpp
+++ b/rmw_cyclonedds_cpp/include/rmw_cyclonedds_cpp/TypeSupport_impl.hpp
@@ -436,14 +436,19 @@ inline size_t get_submessage_array_deserialize(
   size_t sub_members_size,
   size_t max_align)
 {
-  (void)member;
+//  (void)member;
   uint32_t vsize = deser.deserialize_len(1);
-  auto vector = reinterpret_cast<std::vector<unsigned char> *>(field);
-  if (call_new) {
-    new(vector) std::vector<unsigned char>;
+  if (member->resize_function && member->get_function) {
+    member->resize_function(field, vsize);
+    subros_message = member->get_function(field, 0);
+  } else {
+    auto vector = reinterpret_cast<std::vector<unsigned char> *>(field);
+    if (call_new) {
+      new(vector) std::vector<unsigned char>;
+    }
+    vector->resize(vsize * align_int_(max_align, sub_members_size));
+    subros_message = reinterpret_cast<void *>(vector->data());
   }
-  vector->resize(vsize * align_int_(max_align, sub_members_size));
-  subros_message = reinterpret_cast<void *>(vector->data());
   return vsize;
 }

```

I think this fix belongs to you.

fujitatomoya, 2020-08-27:
@iuhilnehc-ynos 
CC: @ivanpauno 

Do we actually need `if (member->resize_function && member->get_function)` ? these will be defined as followings. (just curious)

```
void * get_function__ParticipantEntitiesInfo__node_entities_info_seq(void * untyped_member, size_t index)
{
  auto & member =
    *reinterpret_cast<std::vector<rmw_dds_common::msg::NodeEntitiesInfo> *>(untyped_member);
  return &member[index];
}

void resize_function__ParticipantEntitiesInfo__node_entities_info_seq(void * untyped_member, size_t size)
{
  auto * member =
    reinterpret_cast<std::vector<rmw_dds_common::msg::NodeEntitiesInfo> *>(untyped_member);
  member->resize(size);
}
```

iuhilnehc-ynos, 2020-08-27:
@fujitatomoya 

> Do we actually need if (member->resize_function && member->get_function) ?

Me too. I think I need to check the [template](https://github.com/ros2/rosidl_typesupport/blob/cc5dfe82bb282fe91e6070a0edc4e1ca8b9613c4/rosidl_typesupport_cpp/resource/idl__type_support.cpp.em) if all members will define these functions if member is (type_id_=`ROS_TYPE_MESSAGE`, is_array_=true, array_size_=0 or is_upper_bound_=true).

iuhilnehc-ynos, 2020-08-27:
From 
https://github.com/ros2/rosidl/blob/279731bf72ab1f60553eb381fba7ba3f38947d44/rosidl_typesupport_introspection_cpp/resource/msg__type_support.cpp.em#L168

https://github.com/ros2/rosidl/blob/279731bf72ab1f60553eb381fba7ba3f38947d44/rosidl_typesupport_introspection_cpp/resource/msg__type_support.cpp.em#L178

https://github.com/ros2/rosidl/blob/279731bf72ab1f60553eb381fba7ba3f38947d44/rosidl_typesupport_introspection_cpp/resource/msg__type_support.cpp.em#L185

, if there is a `vector<bool>`, these functions will be nullptr.

iuhilnehc-ynos, 2020-08-27:
> if there is a vector<bool>, these functions will be nullptr.

My mistake. bool is `ROS_TYPE_BOOLEAN`, not a `ROS_TYPE_MESSAGE`.

ivanpauno, 2020-08-27:
> I have checked it. It's working(not to test more, maybe it's not the final patch.). (reverted my previous commit and just use the following patch)

That patch is a good start but:

- The `else` part shouldn't reinterpret the vector as `std::vector<unsigned char>`, instead it should fail (all `ROS_TYPE_MESSAGE` members must provide it).
  Reinterpreting a `std::vector<X>` as `std::vector<unsigned char>` is not allowed, and I'm amazed it was working in four different platforms :exploding_head: :exploding_head: :exploding_head:. I even started reading libstdc++ implementation to understand why it was working, they [store only pointers](https://github.com/gcc-mirror/gcc/blob/eb68d9d828f94d28afa5900fbf3072bbcd64ba8a/libstdc%2B%2B-v3/include/bits/stl_vector.h#L93-L95) and not the size, so the reinterpret_cast works. Of course that's not guaranteed by the standard and shouldn't be done.
- The `resize` function provided by the introspection typesupport calls the correct [vector resize](https://github.com/ros2/rosidl/blob/279731bf72ab1f60553eb381fba7ba3f38947d44/rosidl_typesupport_introspection_cpp/resource/msg__type_support.cpp.em#L124) method, and thus all the placement news are unneeded. It would be great if we could get rid of all the `call_new` arguments and the placement news (that will fix the leaks and https://github.com/ros2/rmw_cyclonedds/pull/224 can be reverted after that).

I probably won't have time to work on a fix til next week, so if somebody else wants to open a PR that would be great.

PS:
```cpp
auto * data_vec = reinterpret_cast<std::vector<unsigned char> *>(data); // where data is a `void *` pointing to a `std::vector<X>`
data_vec->resize(WEIRD_CALCULATIONS_HERE);
```
is the unsafest line of cpp code I have ever seen :exploding_head: . I'm not quite sure if it's going to be beaten.

ivanpauno, 2020-08-27:
> Do we actually need if (member->resize_function && member->get_function) ? these will be defined as followings. (just curious)

A sanity check isn't bad, but in that case the code should fail, as the function pointer being NULL would be unexepected.

iuhilnehc-ynos, 2020-08-28:
> I probably won't have time to work on a fix til next week, so if somebody else wants to open a PR that would be great.

I'd like to make a PR for it.

ivanpauno, 2020-09-03:
I thinks this was fixed by #228, thanks @iuhilnehc-ynos !!

closed 2020-09-03 by ivanpauno