ros2/rmw_cyclonedds #220: rmw_cyclonedds GitHub actions broken

joespeed, 2020-08-19:
@hidmic @ThijsSassen Please help, thanks! It was pointed out to me today that rmw_cyclonedds GitHub actions are broken https://github.com/ros2/rmw_cyclonedds/actions

christophebedard, 2020-08-19:
Might be related/useful: https://github.com/ros2/ros2/issues/942

ivanpauno, 2020-08-28:
See https://github.com/ros2/rosidl_typesupport_connext/pull/57/ for an example fix

joespeed, 2020-08-29:
@ivanpauno thx! might you be able to? Most contributors in Netherlands have holiday. Let me know if not and we'll figure it out

rotu, 2020-08-29:
@ivanpauno That looks like a workaround, not a fix. Why doesn't cppcheck find it in the first place?

rotu, 2020-08-31:
Discovered this bug in trying to work around the issue: https://github.com/rotu/rmw_cyclonedds/actions/runs/232035117

It seems strange to me that rcutils *is* found in Linux but not on Mac/Windows. Is this caused by cppcheck 2.0 vs <2.0?

ivanpauno, 2020-08-31:
> @ivanpauno That looks like a workaround, not a fix. Why doesn't cppcheck find it in the first place?

Include paths of dependencies are not being passed automatically to cppcheck per discussion in https://github.com/ament/ament_lint/pull/117.
The discussion was revisited and this patch was applied in https://github.com/ros2/rclpy/pull/577.

It doesn't look ideal, but it doesn't look to bad either.
If you have any other proposal please open an issue in `ament_lint` repo.

ivanpauno, 2020-08-31:
> It seems strange to me that rcutils is found in Linux but not on Mac/Windows. Is this caused by cppcheck 2.0 vs <2.0?

I guess the version of cppcheck installed in both platforms is different.
I think that cppcheck <2.0 wasn't finding rcutils either, but it didn't require all macros to be configured (thus, there was no problem).

rotu, 2020-08-31:
Problems at the interfaces of packages are the thorniest. If cppcheck can’t hope to detect such problems then it’s not very useful as a static analysis tool.

What value does cppcheck provide? In the current configuration does it detect anything cpplint doesn’t?

I think we should consider disabling it.

rotu, 2020-08-31:
Problems at the interfaces of packages are the thorniest. If cppcheck can’t hope to detect such problems then it’s not very useful as a static analysis tool.

What value does cppcheck provide? In the current configuration does it detect anything cpplint doesn’t?

I think we should consider just disabling it.

ivanpauno, 2020-08-31:
> What value does cppcheck provide? In the current configuration does it detect anything cpplint doesn’t?

AFAIU, cpplint only does style checking and cppcheck is intended to detect bugs.
I haven't check how we're configuring both, but in general `cpplint` cannot replace `cppcheck` (clang-tidy could be a replacement, though some people combine both cppcheck and clang-tidy).

I wouldn't deactivate a linter only because of a one line workaround.
I think that opening an issue upstream to see how the problem can be solved in a better way is worth doing.

rotu, 2020-09-01:
@ivanpauno alternate solution: https://github.com/ament/ament_lint/pull/268

> AFAIU, cpplint only does style checking and cppcheck is intended to detect bugs.

Yes, in theory. But that only holds true if it has enough context to detect such bugs.

I didn't recommend deactivating cppcheck because of a one-line workaround. I recommend it because including dependencies makes performance unacceptably bad and not including dependencies makes it unable to perform static analysis in a meaningful way.

I'm not sure why the performance hit was deemed unacceptably bad, but I suspect it's due to misuse of the tool (e.g. not passing definitions to cppcheck so it checks a combinatoric explosion of macro combinations) rather than its inherent limitations.

iuhilnehc-ynos, 2020-09-08:
Hi all,

After using cppcheck(v2.1), there seems a workaround (by using `ament_cmake_cppcheck_ADDITIONAL_INCLUDE_DIRS`) to set '--include-dirs' for ament_cppcheck. But I think it's not good because there is already 'INCLUDE_DIRECTORYS' exist by `find_package`, why should I set `ament_cmake_cppcheck_ADDITIONAL_INCLUDE_DIRS`?

I think the following source code might be unreasonable, 

https://github.com/ament/ament_lint/blob/58cd08c9f671eba22c9efc222b6f4476b11eec48/ament_cmake_cppcheck/cmake/ament_cmake_cppcheck_lint_hook.cmake#L69-L72

PR #237 (2020-09-08): Suppress cppcheck unknownMacro
Fixes #220

Fast forward to https://github.com/ament/ament_lint/pull/268

This suppresses a defect in `ament_cppcheck` - namely that if a macro is missing at checking time, it fails loudly. This is compounded by fact that `ament_cppcheck` does not include all dependent headers, so macros are likely to be missing.
closed 2020-09-08 by rotu