ros2/rmw_cyclonedds #251: poor performance with pub/sub of pointclouds between computers

Jconn, 2020-09-29:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 18.04
- Installation type:
foxy - from source
- Version or commit hash:
cyclonedds hash:
5e54fa511
rmw_cyclonedds hash:
e0bfe4647

I have a 96x128 pointcloud where each point is 16 bytes. I am publishing this with Sensor QoS at 15 hz using foxy, built from source last week. 

When I am on the same computer as the publisher, and I check the message rate with:
`ros2 topic hz /camera/pointcloud `
I receive messages at 15 hz. 

I have another computer on the same ethernet network as this publisher, also ubuntu 18.04, with foxy built from source. When I check the subscription rate on this second computer using the same `ros2 topic hz /camera/pointcloud `, I receive messages at 
anywhere from .1-.3 hz. 

Doing a ping test between these computers, I get about 30MB/s between them. If my math is right, my pointcloud at 15 Hz should be 15x96x128x16 = 3MB/s of traffic, so I don't think my connection speed is hurting me. I also ran this same test using fastrtps, and my receive rate on computer two sits at ~15hz (although I do have occasional periods where a message is not received for 10 seconds using fastrtps). 

I have other topics being sent between these two computers with much smaller message sizes, e.g. my /tf topic is being published at 50hz between the two computers without any problems.

eboasson, 2020-09-29:
Hi @Jconn, my guess is that you're regularly losing some fragments of the ~200kB point clouds that then don't get retransmitted because the "sensor" QoS setting uses best-effort for the DDS reliability setting. The assumption then would be that this simply occurs more often with Cyclone DDS than with Fast-RTPS, and furthermore that those 10s gaps with Fast really are the same thing.

There are three details worth looking at:
1. If the machines are using the wired Ethernet interface, Cyclone DDS will detect the use of wired interfaces and allow multicast by default. Whether it then uses multicast depends on the number of different subscribed IP:PORT parts. If there is a WiFi in between (a plausible explanation for the max. throughput of 300Mb/s), the use of multicast will likely cause trouble because WiFi multicasting is terribly unreliable.
You can disable multicast use by doing `export CYCLONEDDS_URI='<General><AllowMulticast>spdp</></>'` (the "spdp" leaves multicast enabled for initial discovery only). (This is a quick way for changing Cyclone's configuration, the "proper" way involves files: https://github.com/eclipse-cyclonedds/cyclonedds/#configuration)

2. Another possible explanation for the low throughput is the use of slow machiens (e.g., some old Pis). I do find it plausible that with slow machines you could be getting UDP socket receive buffer overruns if the limit is left at Linux's default of ~400kB. You can monitor those drops by the way if you want to analyze things a bit more, see e.g., https://linux-tips.com/t/udp-packet-drops-and-packet-receive-error-difference/237. That link also contains information on how to raise the limit.

3. A detail that might play a role is that C. uses UDP datagrams of ~14kB by default, whereas F. will make them something close to 64kB — both then let the IP stack fragment/reassemble them to fit in Ethernet's MTU of 1500 bytes. The downside of larger datagrams is of course that more is lost when (some of) the datagram is lost and picking a default is simply a guessing game of what trade-off between best-case performance and cost of error recovery — which doesn't exist for best-effort! — works out best for most people ... Maybe I guessed wrong.
If you want to check whether the size of the UDP datagrames affects things: do `export CYCLONEDDS_URI='<General><MaxMessageSize>62kB</></>'` and try again.

If there is a WiFi involved, I expect that you're dealing with (1) and that that change will solve the problem. If it is the size limit on the UDP socket receive buffers, you can also make things work properly. But in general, sending larg-ish messages over best-effort is just asking for trouble and you're usually *much* better off using "reliable" communications for those.

Jconn, 2020-09-29:
@eboasson you were right that (1) was the problem. 

Both computers are multi-core i5s, connected through the wired Ethernet interface to a network. Neither of these machine have a wireless network interface active. There are several other computers connected to this network through WiFi. None of the machines connected through WiFi have ros nodes running. 

When I add `export CYCLONEDDS_URI='<General><AllowMulticast>spdp</></>'` my subscription frequency is the same on both computers, with no dropouts on either computer. I didn't realize that my WiFi network was playing any role in my message passing. 

Thanks so much for the fast resolution.

Jconn, 2020-10-16:
My dropouts are now less common but still happen once every ~10 minutes (9k pointcloud messages). I'll go through your other suggestions and see what happens.

@eboasson your suggestion to use reliable communication is interesting. What do you consider the message size that my sensors should stop using best effort, sensor QoS and start using reliable QoS?

Jconn, 2020-10-23:
I ended up tuning params as detailed in the [ros2 tuning page](https://index.ros.org/doc/ros2/Troubleshooting/DDS-tuning/) and my issue disappeared.

closed 2020-10-23 by Jconn

eboasson, 2020-10-27:
@Jconn I was offline for a bit and I am really happy to hear that you solved the issue.

The question:
> your suggestion to use reliable communication is interesting. What do you consider the message size that my sensors should stop using best effort, sensor QoS and start using reliable QoS?

is an interesting one, and, as with so many interesting questions, there is no simple answer.

Packet loss has a number of possible causes. Self-evidently, there is noise. The maximum allowed bit error rate (due to A(W)GN, I'd suspect) for GbE is pretty low, I'm not sure I remember the number correctly but I think it is 1e-15. That would mean 1 error every 1e6s on average for a link carrying 1Gbps. That would typically be negligible for loss-tolerant applications. I suspect that in practice it is much lower still, and so that can generally be ignored.

Interference from noisy electromagnetic bursts can be an issue, too. I once worked on DSL technology, there the BER < 1e7, but that is then dominated by error bursts from refrigerators, microwaves and other junk. (AM radio interference is fortunately more easy to deal with.) I'd imagine that the general quality of the CAT 5 (and higher) wiring is better than that of phone lines, so presumably the issue would be less severe, but I really don't know.

Then there is the case of switches/routers where multiple flows come together. If they don't fit on the outgoing channel, you'll get massive packet loss. It really depends on the switch/router, its buffer sizes and whether the flows are sustained or bursty. For multicast packets, the problems multiply: it is not uncommon for multicast packets to be dropped whenever the slightest "inconvenience" is encountered. It is also not uncommon for high-end switches to deliberately block incoming multicasts for a while if they see a high rate of incoming multicast traffic on a port. Such problems often lead to large bursts of lost packets, but they are rare in practice.

On normal machines, the kernel will manage to get the packets from the network fast enough not to lose any in the network hardware or kernel network stack. However, it then needs to deliver the data to the (UDP) sockets, and if the application is not draining those buffers fast enough, it'll throw away the packets. That again often leads to a large burst of lost packets. This is in my experience the most common source of loss.

The nasty thing is that these latter causes tend to worsen under load. For the switches, all you can do is buy good hardware, check your data flows and, possibly, utilise traffic shaping to smooth out bursts and reduce the likelihood of packet loss.

Inside the machine, you'd have to look at the combination of socket buffer size, scheduling priorities and the distribution of the latency from data-in-socket to retrieving it from the socket. Cyclone by default delivers the data to the reader histories from the thread reading from the network (avoiding a context switch), but this then means that the operations the application performs on the reader history caches ultimately affect the likelihood of packet loss. Cyclone can use a separate thread instead but that doesn't fundamentally resolve the problem because then there is a queue of samples in between, and so in the extreme case it could still end up dropping data, albeit in a different spot and for different reasons.

(For a proper worst-case analysis, you then also need to take into account the effects of discovery, lease expiration and whatnot.)

That alone is enough to mean there is not a single correct answer.

Small messages end up as single packets, smallish ones as single datagrams (but multiple packets), and the large ones as multiple datagrams each split into multiple packets. The IP stack does a good job of reassembling the datagrams and the DDS stack does a good job of reassembling the datagrams, but if there is a single lost packet, you lose the sample in best-effort mode. (In reliable mode, it requires a retransmit of the lost fragments, not of the entire sample.)

My mental model is that, in practice, the likelihood of datagram loss (and, to a lesser extent, packet loss) goes up with the size of the messages, with a very modest increase as long as the messages are significantly smaller than the socket receive buffer, and rapidly increasing once you get to roughly the size of that buffer. At some point, the tail end of large samples gets chopped off regularly, at which point the receive rate drops precipitously. This matches your remark that you fixed it by tuning those buffers.

Still, the fundamental question remains: if loss is a problem, then you should not be using best-effort mode, because that makes loss a normal event. (Just so you know: the RTPS protocol in combination with a fast writer using a keep-last history setting has a nasty failure mode that makes loss an equally normal event in reliable mode.)

What I generally advocate is that for data where some loss is acceptable (e.g., I imagine a lot of the processing performed on point clouds can deal with the occasional missing data, especially if the loss can be modelled as a random process), the application is designed to minimise the impact of packet loss. There are various ways of doing that, with the two most common being forward error correction (add parity packets so you can reconstruct in the presence of some loss), and permuting the data (so that packet loss changes from an error burst to a uniform distribution of errored contents in the message, where you do know which parts are missing).

DDS/RTPS don't provide this (I think it would be good to do something about that), and so you would have to do it at application-level by splitting the large messages into smaller ones, and recombining them on the receiving side.

TL;DR version: I'd probably avoid relying best-effort message to arrive if they are larger than 100kB. But that's just a gut feeling.

Jconn, 2020-10-28:
This gives me a lot to think about for future improvements and validation. Thanks for going into so much detail on something that wasn't strictly related to this ticket