ros2/rmw_cyclonedds #261: fatal error: rcpputils/scope_exit.hpp: No such file or directory

marcoag, 2020-10-30:
## Bug report

**Required Info:**

- Operating System: Ubuntu 20.04
- Installation type: Compilation from source
- Version or commit hash: tried on master and foxy branches
- ROS 2 version: Foxy (from debian binaries)

#### Steps to reproduce issue

Strictly following instructions on: 

[Working with Eclipse Cyclone DDS](https://index.ros.org/doc/ros2/Installation/DDS-Implementations/Working-with-Eclipse-CycloneDDS/)

```
mkdir -p ros2_ws/src
cd ros2_ws/src
git clone https://github.com/ros2/rmw_cyclonedds ros2/rmw_cyclonedds
git clone https://github.com/eclipse-cyclonedds/cyclonedds eclipse-cyclonedds/cyclonedds
cd ..
rosdep install --from src -i
colcon build --symlink-install
```

#### Expected behavior

Success compilation

#### Actual behavior

```
$ colcon build --symlink-install
Starting >>> cyclonedds
Finished <<< cyclonedds [8.18s]                     
Starting >>> rmw_cyclonedds_cpp
--- stderr: rmw_cyclonedds_cpp                              
/home/marco/cyc/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/rmw_node.cpp:57:10: fatal error: rcpputils/scope_exit.hpp: No such file or directory
   57 | #include "rcpputils/scope_exit.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/rmw_cyclonedds_cpp.dir/build.make:82: CMakeFiles/rmw_cyclonedds_cpp.dir/src/rmw_node.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:97: CMakeFiles/rmw_cyclonedds_cpp.dir/all] Error 2
make: *** [Makefile:160: all] Error 2
---
Failed   <<< rmw_cyclonedds_cpp [4.46s, exited with code 2]
```

ahcorde, 2020-10-30:
This feature was released https://github.com/ros/rosdistro/pull/26893

Maybe it was not included in the last sync. it will be available in the next one. You can use the rolling packages.

clalancette, 2020-10-30:
That's right; that version of rcpputils didn't make the last sync.  It should be in the next one.

In the meantime, you can set up your apt repositories to pull from the testing repository:

```
$ echo "deb http://packages.ros.org/ros2-testing/ubuntu focal main" | sudo tee -a /etc/apt/sources.list.d/ros2-latest.list
```

marcoag, 2020-11-02:
Perfect, works great with the foxy branch. 

Thanks a lot for the help!

closed 2020-11-02 by marcoag