ros2/rmw_cyclonedds #263: Running realsense node crashes when using cyclonedds

DronistB, 2020-11-01:
## Bug report

**Required Info:**

Trying to run rs_camera.launch.py [ros2 realsense](https://github.com/intel/ros2_intel_realsense/tree/eloquent) with cyclone DDS RMW but it crashs with this error - on fastrtps there is no crash

[INFO] [launch]: All log files can be found below /home/xtend/.ros/log/2020-11-01-17-58-02-352492-tx2-bionic-20659
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense_node-1]: process started with pid [20676]
[realsense_node-1] 
[realsense_node-1] >>> [rcutils|error_handling.c:107] rcutils_set_error_state()
[realsense_node-1] This error state is being overwritten:
[realsense_node-1] 
[realsense_node-1]   'type support not from this implementation, at /tmp/binarydeb/ros-eloquent-rmw-cyclonedds-cpp-0.5.1/src/rmw_node.cpp:1054'
[realsense_node-1] 
[realsense_node-1] with this new error message:
[realsense_node-1] 
[realsense_node-1]   'type_support is null, at /tmp/binarydeb/ros-eloquent-rmw-cyclonedds-cpp-0.5.1/src/rmw_node.cpp:1311'
[realsense_node-1] 
[realsense_node-1] rcutils_reset_error() should be called after error handling to avoid this.
[realsense_node-1] <<<
[realsense_node-1] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[realsense_node-1]   what():  could not create publisher: type_support is null, at /tmp/binarydeb/ros-eloquent-rmw-cyclonedds-cpp-0.5.1/src/rmw_node.cpp:1311, at /tmp/binarydeb/ros-eloquent-rcl-0.8.4/src/rcl/publisher.c:173

- Operating System:
  - 
- Installation type: Ubuntu 18.04
  - 
 sudo apt install ros-eloquent-cyclonedds-cpp
- DDS implementation: 
CycloneDDS
- Client library (if applicable):
rclcpp

#### Steps to reproduce issue

```
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
ros2 launch realsense_examples rs_camera.launch.py

```

eboasson, 2020-11-01:
In the ROS 2 build process message definitions get compiled for a variety of RMW implementations into different "type support" libraries. This particular error occurs when the right one (in this case, Cyclone) wasn't included at build time.

Which ones get included is determined at some point in the build process by looking at the list of available RMW implementations. Perhaps you enabled Cyclone later on or something like that. I've received the same error a few times and solved it by just rebuilding everything, but I am sure that there is a smarter way to do it!

With Fast-RTPS you don't have the problem because that is the default RMW implementation and so the necessary stuff is always generated for it.

DronistB, 2020-11-03:
> In the ROS 2 build process message definitions get compiled for a variety of RMW implementations into different "type support" libraries. This particular error occurs when the right one (in this [...]

Thanks, it worked!

closed 2020-11-03 by DronistB