ros2/rmw_cyclonedds #264: Assertion `0 == std::memcmp(context, &zero_context, sizeof(rmw_context_t))' failed.

marcoag, 2020-11-02:
## Bug report

**Required Info:**

- Operating System: Ubuntu 20.04
- Installation type: source 
- Version or commit hash: foxy branch (latest)
- ROS foxy from deb binaries

#### Steps to reproduce issue

```
mkdir -p ros2_ws/src
cd ros2_ws/src
git clone https://github.com/ros2/rmw_cyclonedds ros2/rmw_cyclonedds
git clone https://github.com/eclipse-cyclonedds/cyclonedds eclipse-cyclonedds/cyclonedds
cd ros2/rmw_cyclonedds/
git checkout foxy
cd ../../..
rosdep install --from src -i
colcon build --symlink-install
source install/setup.bash
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
ros2 run demo_nodes_cpp talker
```

#### Expected behavior

Run the node and start publishing

#### Actual behavior

`talker: /home/marco/ros2_ws/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/rmw_node.cpp:1151: rmw_ret_t rmw_init(const rmw_init_options_t*, rmw_context_t*): Assertion 0 == std::memcmp(context, &zero_context, sizeof(rmw_context_t))' failed.`

#### Additional information

It works fine if I checkout `0.7.4`

hidmic, 2020-11-02:
[Line 1156 in rmw_node.cpp](https://github.com/ros2/rmw_cyclonedds/blame/foxy/rmw_cyclonedds_cpp/src/rmw_node.cpp#L1156) was removed in `master` by https://github.com/ros2/rmw_cyclonedds/pull/208. We have to backport it.

ivanpauno, 2020-11-03:
I really don't understand why https://github.com/ros2/rmw_cyclonedds/pull/202 was backported, backporting that kind of things is likely to introduce a bug and doesn't have any real value.

marcoag, 2020-11-04:
Working great now, thanks for the fix!

closed 2020-11-04 by marcoag