ros2/rmw_cyclonedds #268: Action client throws "no free slot" exception when handling more than 45 goals in parallel

schornakj, 2020-11-25:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04 LTS
- Installation type:
  - Binaries
- Version or commit hash:
  - Eloquent
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - rclcpp

#### Summary

I have an action client/server pair that handles a large number of goals in parallel. When using `rmw_cyclonedds_cpp`, my action client node throws an exception when it reaches 45 in-progress goal requests.

#### Steps to reproduce issue

I've created a standalone package (written for Eloquent but possible to modify for later releases) that reproduces this issue: https://github.com/schornakj/rclcpp_action_exception_test/tree/feature/parallel-client

I set the environment variable `RMW_IMPLEMENTATION` to `rmw_cyclonedds_cpp`. I decided to post this issue here because I do not experience this issue when using `rmw_fastrtps_cpp` so I think it may be specific to `rmw_cyclonedds_cpp`.

#### Expected behavior

I was expecting my action client to be able to handle an arbitrary number of goal requests at once.

#### Actual behavior

An exception is thrown after 45 requests are in progress (I'm not sure if that number is meaningful but it's consistent).

```
$ ros2 launch action_exception_test run_nodes_parallel.launch.py 
[INFO] [launch]: All log files can be found below /home/joe/.ros/log/2020-11-25-14-49-36-730852-joe-29145
[INFO] [launch]: Default logging verbosity is set to INFO
1606337376.736461 [2]       ros2: using network interface eno1 (udp/192.168.1.176) selected arbitrarily from: eno1, docker0, tun0
[INFO] [action_exception_test_server_node-1]: process started with pid [29170]
[INFO] [action_exception_test_client_node_parallel-2]: process started with pid [29171]
[action_exception_test_client_node_parallel-2] 1606337376.764799 [2] action_exc: using network interface eno1 (udp/192.168.1.176) selected arbitrarily from: eno1, docker0, tun0
[action_exception_test_server_node-1] 1606337376.766412 [2] action_exc: using network interface eno1 (udp/192.168.1.176) selected arbitrarily from: eno1, docker0, tun0
[action_exception_test_client_node_parallel-2] [INFO] [client_node]: N_IN_PROGRESS=0

...

[action_exception_test_client_node_parallel-2] [INFO] [client_node]: N_IN_PROGRESS=44
[action_exception_test_client_node_parallel-2] [INFO] [client_node]: N_IN_PROGRESS=45
[action_exception_test_client_node_parallel-2] 1606337384.950955 [] action_exc: create_thread: action_exceptio: no free slot
[ERROR] [action_exception_test_client_node_parallel-2]: process has died [pid 29171, exit code -6, cmd '/home/joe/workspaces/cb_group_test_ws/install/action_exception_test/lib/action_exception_test/action_exception_test_client_node_parallel --ros-args -r __node:=client_node'].
```

eboasson, 2020-11-25:
Hi @schornakj, this is indeed a Cyclone-specific thing. Despite about 40 years of evidence of the foolishness behind the alleged claim that "640kB ought to be enough for anybody" (or, indeed, the much older one that there would be a market for perhaps 5 computers, world-wide), Cyclone still has its own "640kB": it tracks threads going through the API, and for that it still simply allocates space for 64 threads.

PR https://github.com/eclipse-cyclonedds/cyclonedds/pull/646 is the "well, then certainly 1280kB ought to be enough" response. It is not a proper fix, but it might give you a bit of breathing room.

schornakj, 2020-11-25:
Good to know! I had a suspicion that I was running into some kind of hard-coded limit like that.

Fortunately I don't have an urgent business need for 40+ action requests running in parallel. I was running some test code to explore other `rclcpp_action`-related issues (which sends an unusually large number of parallel requests to surface failures faster) and I was surprised to get a different exception than the one I was expecting.

I'd be interested in the dynamic thread allocation approach you mention in your PR -- seems like a good long-term solution.

brawner, 2020-12-01:
I noticed a similar failure with `test_multithreaded__rmw_cyclonedds_cpp` on a 32-core machine. That test always runs `2 * max_concurrency threads`. https://github.com/ros2/system_tests/blob/02a3972276535714b03ea24d2ffaf2f5be13033f/test_rclcpp/test/test_multithreaded.cpp#L154-L255

Perhaps there could be a way to pass the requested number of threads to cyclonedds in the `rmw_init_options`?