ros2/rmw_cyclonedds #273: arm64-specific bug with invalid node handles

JWhitleyWork, 2020-12-18:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04 or Ubuntu 20.04
- Architecture:
  - `arm64`
- Installation type:
  - Binaries
- Version or commit hash:
  - 0.7.0-1bionic for ROS Dashing and 0.7.6-1focal for ROS Foxy
- DDS implementation:
  - CycloneDDS w/`rmw_cyclonedds_cpp`

#### Steps to reproduce issue
```
$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ git clone https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto.git
$ cd AutowareAuto
$ colcon build --packages-up-to state_estimation_node
$ colcon test --packages-select state_estimation_node
$ colcon test-result --verbose
```

#### Expected behavior
Tests pass.

#### Actual behavior
First test fails with the following error on Dashing:
```
- test_state_estimation_node_bad.test.py
  <<< failure message
    -- run_test.py: invoking following command in '/home/whitleysoftwareservices/autoware.auto.dashing/src/prediction/state_estimation_node':
     - ros2 test /home/whitleysoftwareservices/autoware.auto.dashing/src/prediction/state_estimation_node/test/state_estimation_node_bad.test.py --junit-xml=/home/whitleysoftwareservices/autoware.auto.dashing/build/state_estimation_node/test_results/state_estimation_node/test_state_estimation_node_bad.test.py.xunit.xml --package-name=state_estimation_node
    [INFO] [launch]: All log files can be found below /home/whitleysoftwareservices/.ros/log/2020-12-18-12-30-02-235681-ade-12015
    [INFO] [launch]: Default logging verbosity is set to INFO
    1608316203.982884 [1]       ros2: ddsi_udp_create_conn: set IP_MULTICAST_IF failed: Bad Parameter
    [ERROR] [rmw_cyclonedds_cpp]: rmw_create_node: failed to create DDS participant
    
    >>> [rcutils|error_handling.c:106] rcutils_set_error_state()
    This error state is being overwritten:
    
      'error not set, at /tmp/binarydeb/ros-dashing-rcl-0.7.9/src/rcl/node.c:326'
    
    with this new error message:
    
      'rcl node's rmw handle is invalid, at /tmp/binarydeb/ros-dashing-rcl-0.7.9/src/rcl/node.c:464'
    
    rcutils_reset_error() should be called after error handling to avoid this.
    <<<
    [ERROR] [rcl]: Failed to fini publisher for node: 1
    [ERROR] [launch]: Caught exception in launch (see debug for traceback): Unknown error creating node: rcl node's rmw handle is invalid, at /tmp/binarydeb/ros-dashing-rcl-0.7.9/src/rcl/node.c:464
    Processes under test stopped before tests completed
    -- run_test.py: return code 1
    -- run_test.py: verify result file '/home/whitleysoftwareservices/autoware.auto.dashing/build/state_estimation_node/test_results/state_estimation_node/test_state_estimation_node_bad.test.py.xunit.xml'
  >>>
```

And the following on Foxy:
```
- test_state_estimation_node_bad.test.py
  <<< failure message
    -- run_test.py: invoking following command in '/home/whitleysoftwareservices/autoware.auto.foxy/build/state_estimation_node':
     - ros2 test /home/whitleysoftwareservices/autoware.auto.foxy/src/prediction/state_estimation_node/test/state_estimation_node_bad.test.py --junit-xml=/home/whitleysoftwareservices/autoware.auto.foxy/build/state_estimation_node/test_results/state_estimation_node/test_state_estimation_node_bad.test.py.xunit.xml --package-name=state_estimation_node
    Running with ROS_DOMAIN_ID 1
    ROS_DOMAIN_ID 1
    [INFO] [launch]: All log files can be found below /home/whitleysoftwareservices/.ros/log/2020-12-18-12-49-08-678370-ade-4228
    [INFO] [launch]: Default logging verbosity is set to INFO
    Warning: Passing ready_fn as an argument to generate_test_description will be removed in a future release.  Include a launch_testing.actions.ReadyToTest action in the LaunchDescription instead.
[... 38 lines cut ...]
      File "/home/whitleysoftwareservices/autoware.auto.foxy/src/prediction/state_estimation_node/test/state_estimation_node_bad.test.py", line 76, in test_exit_code
        self.assertIn(ref_stderr, full_stderr)
    AssertionError: "what():  Please provide either 'output_frequency' setting or 'data_driven' one, not both." not found in "1608317348.993236 [1] state_esti: ddsi_udp_create_conn: set IP_MULTICAST_IF failed: Bad Parameter\n[ERROR] [1608317348.996267111] [rmw_cyclonedds_cpp]: rmw_create_node: failed to create domain, error Error\n\n>>> [rcutils|error_handling.c:108] rcutils_set_error_state()\nThis error state is being overwritten:\n\n  'error not set, at /tmp/binarydeb/ros-foxy-rcl-1.1.9/src/rcl/node.c:276'\n\nwith this new error message:\n\n  'rcl node's rmw handle is invalid, at /tmp/binarydeb/ros-foxy-rcl-1.1.9/src/rcl/node.c:428'\n\nrcutils_reset_error() should be called after error handling to avoid this.\n<<<\n[ERROR] [1608317348.998946065] [rcl]: Failed to fini publisher for node: 1\nterminate called after throwing an instance of 'rclcpp::exceptions::RCLError'\n  what():  failed to initialize rcl node: rcl node's rmw handle is invalid, at /tmp/binarydeb/ros-foxy-rcl-1.1.9/src/rcl/node.c:428\nqemu: uncaught target signal 6 (Aborted) - core dumped\n"
    
    ----------------------------------------------------------------------
    Ran 1 test in 0.007s
    
    FAILED (failures=1)
    -- run_test.py: return code 1
    -- run_test.py: verify result file '/home/whitleysoftwareservices/autoware.auto.foxy/build/state_estimation_node/test_results/state_estimation_node/test_state_estimation_node_bad.test.py.xunit.xml'
  >>>
build/state_estimation_node/test_results/state_estimation_node/test_state_estimation_node_bad.test.py.xunit.xml: 2 tests, 0 errors, 1 failure, 0 skipped
- state_estimation_node.TestProcessOutput test_exit_code
  <<< failure message
    Traceback (most recent call last):
      File "/home/whitleysoftwareservices/autoware.auto.foxy/src/prediction/state_estimation_node/test/state_estimation_node_bad.test.py", line 76, in test_exit_code
        self.assertIn(ref_stderr, full_stderr)
    AssertionError: "what():  Please provide either 'output_frequency' setting or 'data_driven' one, not both." not found in "1608317348.993236 [1] state_esti: ddsi_udp_create_conn: set IP_MULTICAST_IF failed: Bad Parameter\n[ERROR] [1608317348.996267111] [rmw_cyclonedds_cpp]: rmw_create_node: failed to create domain, error Error\n\n>>> [rcutils|error_handling.c:108] rcutils_set_error_state()\nThis error state is being overwritten:\n\n  'error not set, at /tmp/binarydeb/ros-foxy-rcl-1.1.9/src/rcl/node.c:276'\n\nwith this new error message:\n\n  'rcl node's rmw handle is invalid, at /tmp/binarydeb/ros-foxy-rcl-1.1.9/src/rcl/node.c:428'\n\nrcutils_reset_error() should be called after error handling to avoid this.\n<<<\n[ERROR] [1608317348.998946065] [rcl]: Failed to fini publisher for node: 1\nterminate called after throwing an instance of 'rclcpp::exceptions::RCLError'\n  what():  failed to initialize rcl node: rcl node's rmw handle is invalid, at /tmp/binarydeb/ros-foxy-rcl-1.1.9/src/rcl/node.c:428\nqemu: uncaught target signal 6 (Aborted) - core dumped\n"
  >>>
```

#### Additional information
This has been tested both on `arm64`-native hardware and using `arm64` binary translation with QEMU. The test fails consistently in both Dashing and Foxy when using QEMU and occasionally on `arm64`-native hardware depending on the system load indicating that this is timing-dependent. Using `rmw_fastrtps_cpp`, this failure does not occur. We have also only seen this error on this one test indicating it's a pretty corner-case scenario.

For some context to aide in troubleshooting, this test intentionally provides the `state_estimation_node` with conflicting parameters `data_driven: true` and `output_frequency: 30.0`. Using the free function `time_between_publish_requests` in `src/prediction/state_estimation_node/src/state_estimation_node.cpp` with both of these parameters should throw a `std::logic_error` [on line 74](https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/blob/master/src/prediction/state_estimation_node/src/state_estimation_node.cpp#L74) before any publishers/subscribers are created but after the node has been constructed/initialized. In Dashing, there is no indication that this exception is ever thrown but Foxy does show the output with the correct exception, only _after_ the unexpected `rcl` error.

eboasson, 2021-01-06:
I'd say this:
```
    [state_estimation_node_exe-1] 1608317348.993236 [1] state_esti: ddsi_udp_create_conn: set IP_MULTICAST_IF failed: Bad Parameter
```
is the weird bit: then the DDS domain doesn't start, the participant can't be created, nor can the node be created, &c. So one would thing everything else is collateral damage. That error really means:
```C
  return ddsrt_setsockopt (sock, IPPROTO_IP, IP_MULTICAST_IF, gv->ownloc.address + 12, 4);
```
failed, or really
```C
  if (setsockopt(sock, level, optname, optval, optlen) == -1) {
    goto err_setsockopt;
  }
```
Unfortunately, many errors get mapped to `BAD_PARAMETER`, so we don't know exactly what it is. However, `sock` must be ok or it wouldn't have made it to this point. So it is most likely that the kernel is unhappy with the address.

There are some things that should be fairly simple to do and that ought to provide a bit more information:
* Getting a trace from Cyclone should tell us what interfaces it thinks exist, and then one can infer the parameters;
* Run it under `strace` to trace the inputs and outputs of the networking related system calls;
* We could patch Cyclone to print the details (that might be easier than trying to catch it in a debugger).

I suspect that will show it always tries to set the multicast interface to use for outgoing packets, also when the socket isn't used for transmitting and even when the interface doesn't support multicast. Perhaps that is what the kernel doesn't like ... The good news is that the fix will be pretty straightforward if this is the case.

eboasson, 2021-03-01:
You might want to give this:

https://github.com/eboasson/cyclonedds/tree/mc-options-only-when-needed

a try. It is based on the possibility that you are not actually using multicast and that there is therefore no need to set this particular option. Otherwise, I really need to know a bit more about what it actually passes into the kernel and which network interfaces exist (like I mentioned). I have no reason to believe the interface numbers Cyclone uses are incorrect, but I've been caught by stupid bugs before.

As another data point, it doesn't happen on Ubuntu 20.04 on qemu on macOS on an M1 ... that's not emulation, of course, but still emulated devices.

JWhitleyWork, 2021-03-01:
@eboasson Do you know how to easily replace the installed CycloneDDS installation with this one, built from source? Just adding the repository from my workspace, `colcon` doesn't seem to recognize that it's a deep-down dependency when building with `--packages-up-to`.

eboasson, 2021-03-02:
@JWhitleyWork I used to do `--packages-up-to rmw_cyclonedds_cpp` but for rolling/galatic that's no longer needed. The easiest should be to just do `colcon build --packages-select cyclonedds`, all you need is the new `libddsc.so.0.7.0` (or whatever the number is).

What version of `rmw_cyclonedds_cpp` are you using, by the way? Because the RMW layer uses some things that aren't part of the stable API to paper over differences between ROS2 and DDS and some changes happened that are binary compatible but not quite source compatible. The patch I did should cherry-pick cleanly to just about any version of Cyclone if needed.

jpace121, 2021-09-30:
I'm seeing a similar error message (`ddsi_udp_create_conn: set IP_MULTICAST_IF failed: Bad Parameter`) when trying to use a publisher in a arm64 Ubuntu container that's running on a 64-bit Debian 11 host using qemu-user-static for the emulation.

I was able to replicate the error with a DDS publisher without ros 2 (using cyclonedds-cxx). I tried applying the commit @eboasson linked to in his branch above to cyclonedds, but the commit didn't apply cleanly to the latest version of cyclonedds.

eboasson, 2021-09-30:
@jpace121, I rebased https://github.com/eboasson/cyclonedds/tree/mc-options-only-when-needed and would appreciate it very much if you gave it another go.

jpace121, 2021-10-01:
That didn't seem to fix it. With that being said, I noticed tonight that I'm also getting the following warning out stdout:
```
qemu: uncaught target signal 6 (Aborted) - core dumped
Aborted (core dumped)
```
so I'm guessing my bug may be caused by something going on in qemu, not cyclone.

Thanks for the assistance.

eboasson, 2021-10-01:
@jpace121 I imagine any `qemu` crash would be rather serious, I'm sure they'd like to hear of it.

Anyway, the point of that branch was to avoid doing `setsockopt(... IP_MULTICAST_IF ...)` when multicast is not used. One reason was that it seemed plausible (however remotely) that it might give an error if multicast wasn't enabled somewhere in the kernel or below; the other reason is that it should at least give a workaround by disabling multicast in Cyclone.

One thing I have wondered about is what it is actually passing into the kernel. As these calls happen really early on in the process and it sounds like it is fairly easily reproduced (especially now you don't even need to involve ROS), perhaps it would be possible to log the system calls with `strace -v` (I think that's the option that gives all the gory details; possibly `-e network` would be sensible too). That would at least give us some more understanding of what goes on when it fails.

tfoote, 2022-02-04:
I can reproduce this in a much shorter invocation using the example rclcpp executables. Below is on Galactic, I saw it first on foxy. 

```
$ docker run --platform=linux/aarch64 -ti ros bash
Unable to find image 'ros:latest' locally
latest: Pulling from library/ros
Digest: sha256:dec83e0668b43f0f7734935d91a1c9c7a89fba6dd878cd2bd6aedffd196132c3
Status: Downloaded newer image for ros:latest
root@66357d5ba0db:/# apt-get update -qq && apt-get install -qqy ros-galactic-examples-rclcpp-minimal-subscriber ros-galactic-examples-rclcpp-minimal-publisher
## INSTALL LOGS TRUNCATED
root@66357d5ba0db:/# . /opt/ros/galactic/setup.bash
root@66357d5ba0db:/# export ROS_DOMAIN_ID=84
root@66357d5ba0db:/# ros2 run examples_rclcpp_minimal_publisher publisher_member_function
Unsupported setsockopt level=0 optname=32
1643936991.678175 [84] publisher_: ddsi_udp_create_conn: set IP_MULTICAST_IF failed: Bad Parameter
[ERROR] [1643936991.681629437] [rmw_cyclonedds_cpp]: rmw_create_node: failed to create domain, error Error

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'error not set, at /tmp/binarydeb/ros-galactic-rcl-3.1.2/src/rcl/node.c:261'

with this new error message:

  'rcl node's rmw handle is invalid, at /tmp/binarydeb/ros-galactic-rcl-3.1.2/src/rcl/node.c:413'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [1643936991.685166614] [rcl]: Failed to fini publisher for node: 1
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  failed to initialize rcl node: rcl node's rmw handle is invalid, at /tmp/binarydeb/ros-galactic-rcl-3.1.2/src/rcl/node.c:413
qemu: uncaught target signal 6 (Aborted) - core dumped
```

eboasson, 2022-02-04:
Hi @tfoote, one would think the patch in linked to in https://github.com/ros2/rmw_cyclonedds/issues/273#issuecomment-930915050 will fix it by not doing `IP_MULTICAST_IF` when multicast is not enabled. It is not in the 0.8.x because I didn't want to change that branch without having confirmed that it solved the problem, but if it works I'll merge it.

But I'd still like to know what's going on, and so if you have a chance to run reproduce with Cyclone tracing enabled (i.e., `export CYCLONEDDS_URI='<Tr><C>trace</C><Out>cdds.log.${CYCLONEDDS_PID}</></>'` and with `strace` to get the exact parameters passed to the system call, I'd be grateful. The latter is the only way I can think of to be absolutely certain that it gets passed the right parameters (I think that's likely, I just want to be sure), and the former has everything to tell me if it ends up with multicast enabled or disabled internally.

tfoote, 2022-04-08:
Here's an example of the cdds.log generated. 

```
root@caa9f7853d37:/# cat cdds.log.2608 
1649459589.202664 [84] publisher_: config: Domain/General/NetworkInterfaceAddress/#text: auto {}
1649459589.204019 [84] publisher_: config: Domain/General/MulticastRecvNetworkInterfaceAddresses/#text: preferred {}
1649459589.204106 [84] publisher_: config: Domain/General/ExternalNetworkAddress/#text: auto {}
1649459589.204146 [84] publisher_: config: Domain/General/ExternalNetworkMask/#text: 0.0.0.0 {}
1649459589.204184 [84] publisher_: config: Domain/General/AllowMulticast/#text: default {}
1649459589.204226 [84] publisher_: config: Domain/General/PreferMulticast/#text: false {}
1649459589.204257 [84] publisher_: config: Domain/General/MulticastTimeToLive/#text: 32 {}
1649459589.204276 [84] publisher_: config: Domain/General/DontRoute/#text: false {}
1649459589.204336 [84] publisher_: config: Domain/General/Transport/#text: udp {}
1649459589.204367 [84] publisher_: config: Domain/General/EnableMulticastLoopback/#text: true {}
1649459589.204495 [84] publisher_: config: Domain/General/MaxMessageSize/#text: 14720 B {}
1649459589.204516 [84] publisher_: config: Domain/General/MaxRexmitMessageSize/#text: 1456 B {}
1649459589.204549 [84] publisher_: config: Domain/General/FragmentSize/#text: 1344 B {}
1649459589.204568 [84] publisher_: config: Domain/General/RedundantNetworking/#text: false {}
1649459589.204626 [84] publisher_: config: Domain/Sizing/ReceiveBufferSize/#text: 1 MiB {}
1649459589.204646 [84] publisher_: config: Domain/Sizing/ReceiveBufferChunkSize/#text: 128 KiB {}
1649459589.204703 [84] publisher_: config: Domain/Compatibility/StandardsConformance/#text: lax {}
1649459589.204727 [84] publisher_: config: Domain/Compatibility/ExplicitlyPublishQosSetToDefault/#text: false {}
1649459589.204772 [84] publisher_: config: Domain/Compatibility/ManySocketsMode/#text: single {}
1649459589.204791 [84] publisher_: config: Domain/Compatibility/AssumeRtiHasPmdEndpoints/#text: false {}
1649459589.204809 [84] publisher_: config: Domain/Discovery/Tag/#text:  {}
1649459589.204846 [84] publisher_: config: Domain/Discovery/ExternalDomainId/#text: 84 {}
1649459589.204885 [84] publisher_: config: Domain/Discovery/DSGracePeriod/#text: 30 s {}
1649459589.204924 [84] publisher_: config: Domain/Discovery/ParticipantIndex/#text: none {}
1649459589.204943 [84] publisher_: config: Domain/Discovery/MaxAutoParticipantIndex/#text: 9 {}
1649459589.204960 [84] publisher_: config: Domain/Discovery/SPDPMulticastAddress/#text: 239.255.0.1 {}
1649459589.204979 [84] publisher_: config: Domain/Discovery/SPDPInterval/#text: 30 s {}
1649459589.204997 [84] publisher_: config: Domain/Discovery/DefaultMulticastAddress/#text: auto {}
1649459589.205030 [84] publisher_: config: Domain/Discovery/Ports/Base/#text: 7400 {}
1649459589.205050 [84] publisher_: config: Domain/Discovery/Ports/DomainGain/#text: 250 {}
1649459589.205069 [84] publisher_: config: Domain/Discovery/Ports/ParticipantGain/#text: 2 {}
1649459589.205089 [84] publisher_: config: Domain/Discovery/Ports/MulticastMetaOffset/#text: 0 {}
1649459589.205108 [84] publisher_: config: Domain/Discovery/Ports/UnicastMetaOffset/#text: 10 {}
1649459589.205134 [84] publisher_: config: Domain/Discovery/Ports/MulticastDataOffset/#text: 1 {}
1649459589.205154 [84] publisher_: config: Domain/Discovery/Ports/UnicastDataOffset/#text: 11 {}
1649459589.205699 [84] publisher_: config: Domain/Tracing/Category/#text: trace {0}
1649459589.205760 [84] publisher_: config: Domain/Tracing/OutputFile/#text: cdds.log.2608 {0}
1649459589.205780 [84] publisher_: config: Domain/Tracing/AppendToFile/#text: false {}
1649459589.205799 [84] publisher_: config: Domain/Tracing/PacketCaptureFile/#text:  {}
1649459589.205818 [84] publisher_: config: Domain/Internal/DeliveryQueueMaxSamples/#text: 256 {}
1649459589.205836 [84] publisher_: config: Domain/Internal/PrimaryReorderMaxSamples/#text: 128 {}
1649459589.205854 [84] publisher_: config: Domain/Internal/SecondaryReorderMaxSamples/#text: 128 {}
1649459589.205873 [84] publisher_: config: Domain/Internal/DefragUnreliableMaxSamples/#text: 4 {}
1649459589.205890 [84] publisher_: config: Domain/Internal/DefragReliableMaxSamples/#text: 16 {}
1649459589.205960 [84] publisher_: config: Domain/Internal/BuiltinEndpointSet/#text: writers {}
1649459589.205980 [84] publisher_: config: Domain/Internal/MeasureHbToAckLatency/#text: false {}
1649459589.205998 [84] publisher_: config: Domain/Internal/UnicastResponseToSPDPMessages/#text: true {}
1649459589.206016 [84] publisher_: config: Domain/Internal/SynchronousDeliveryPriorityThreshold/#text: 0 {}
1649459589.206038 [84] publisher_: config: Domain/Internal/SynchronousDeliveryLatencyBound/#text: inf {}
1649459589.206057 [84] publisher_: config: Domain/Internal/MaxParticipants/#text: 0 {}
1649459589.206075 [84] publisher_: config: Domain/Internal/AccelerateRexmitBlockSize/#text: 0 {}
1649459589.206120 [84] publisher_: config: Domain/Internal/RetransmitMerging/#text: never {}
1649459589.206140 [84] publisher_: config: Domain/Internal/RetransmitMergingPeriod/#text: 5 ms {}
1649459589.206158 [84] publisher_: config: Domain/Internal/HeartbeatInterval/#text: 100 ms {}
1649459589.206198 [84] publisher_: config: Domain/Internal/HeartbeatInterval[@min]: 5 ms {}
1649459589.206218 [84] publisher_: config: Domain/Internal/HeartbeatInterval[@minsched]: 20 ms {}
1649459589.206236 [84] publisher_: config: Domain/Internal/HeartbeatInterval[@max]: 8 s {}
[... cut ...]
1649459526.676448 [84] publisher_: extmask: invalid/0 (not applicable)
1649459526.676473 [84] publisher_: SPDP MC: udp/239.255.0.1
1649459526.676496 [84] publisher_: default MC: udp/239.255.0.1
1649459526.676513 [84] publisher_: SSM support included
1649459526.679645 [84] publisher_: failed to increase socket receive buffer size to 1048576 bytes, continuing with 425984 bytes
1649459526.679913 [84] publisher_: ddsi_udp_create_conn: set IP_MULTICAST_IF failed: Bad Parameter
1649459526.681850 [84] publisher_: udp finalized
root@caa9f7853d37:/# 
```