ros2/rmw_cyclonedds #281: 👩‍🌾 TestExecutor.spin_some_fail_wait_set_clear test failure

chapulina, 2021-02-08:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Focal 
- Installation type:
  -  source
- Version or commit hash:
  - `master/ros2.repos`
- DDS implementation:
  - default
- Client library (if applicable):
  - `rclcpp`

#### Steps to reproduce issue

The test failed once on the repeated nightly CI, see [test history](https://ci.ros2.org/job/nightly_linux-aarch64_repeated/1479/testReport/projectroot.test/rclcpp/test_executor/history/).

This could be a spurious failure, but I'll ticket the issue to keep record for the coming weeks.

#### Expected behavior

Test passes 100% of the time.

#### Actual behavior

1 failure in several months

#### Additional information

Test failure output:

```
[ RUN      ] TestExecutor.spin_some_fail_wait_set_clear
1611048853.258930 [108] test_execu: ddsi_udp_create_conn: failed to bind to ANY:34401: address in use
[ERROR] [1611048853.258965495] [rmw_cyclonedds_cpp]: rmw_create_node: failed to create domain, error Error

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'error not set, at /home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/src/ros2/rcl/rcl/src/rcl/node.c:261'

with this new error message:

  'rcl node's rmw handle is invalid, at /home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/src/ros2/rcl/rcl/src/rcl/node.c:413'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [1611048853.259004337] [rcl]: Failed to fini publisher for node: 1
unknown file: Failure
C++ exception with description "failed to initialize rcl node: rcl node's rmw handle is invalid, at /home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/src/ros2/rcl/rcl/src/rcl/node.c:413" thrown in the test body.
[  FAILED  ] TestExecutor.spin_some_fail_wait_set_clear (1 ms)
```

---

https://github.com/osrf/buildfarmer/issues/145

eboasson, 2021-02-09:
> because we don't specify the address to bind to in our code, so we have no choice over it

You indirectly specify it via the domain id and the domain id-to-port number mapping (which is configurable in itself). In the default mapping, 34401 is the application data multicast port for domain id 108. This error only ever happens for a multicast socket in Cyclone, so that detail fits, too.

The port number also happens in the range where the kernel may have bound some other socket. I don't know if there's a way of finding out whether that happened, but it at least seems like it might have happened.

I don't know why test used domain id 108. If it is to keep different runs separate, then there are some tricks that can be used. E.g., the mapping can be adjusted, or you could run everything on the default domain and use the "domain tag" (a recent addition to the standard) to keep them apart instead (the multicasts would then reach all processes, but it would only be interpreted for a matching "domain tag"). That way you are not restricted to a few integers.

mjcarroll, 2021-03-18:
Similar failures have showed up in a few other places in nightlies.  I have seen it exclusively happen in the `_repeated` suite.

* https://ci.ros2.org/job/nightly_linux_repeated/2229/testReport/(root)/projectroot/test_destruction/
* https://ci.ros2.org/job/nightly_linux_repeated/2228/testReport/junit/projectroot.test/rclcpp/test_static_executor_entities_collector/

ivanpauno, 2021-03-22:
> The port number also happens in the range where the kernel may have bound some other socket. I don't know if there's a way of finding out whether that happened, but it at least seems like it might have happened.

Just to double check, do you mean that the linux kernel assigned that port for another thing as it's in the "ephemeral port" range? That range seems to be 32768-60999.

The CI scripts are setting domain id 108 for linux machines: https://github.com/ros2/ci/blob/43059ca88b1f44677a7b7d60ce0fa8250abe9bab/ros2_batch_job/linux_batch/__init__.py#L33.
I think that logic can be deleted, as actually it doesn't help (ros2 CI jobs doesn't use local machines in the same network anymore).

The only weird thing is that this was noticed several times with cyclonedds, and not with other dds vendors.
Maybe it was just bad luck, but I would expect the problem to equally happen in all implementations.

ivanpauno, 2021-03-22:
> The only weird thing is that this was noticed several times with cyclonedds, and not with other dds vendors.

At least in one of the other vendors a `bind()` error is ignored and only a log is printed in that case, maybe that's why we never saw it happening (?).

A colision in a multicast address seems pretty strange to me though, I'm not sure what else could be using multicast address in our CI machines.

eboasson, 2021-03-23:
> Just to double check, do you mean that the linux kernel assigned that port for another thing as it's in the "ephemeral port" range? That range seems to be 32768-60999

Yes, that was the only (somewhat) plausible explanation I could come up with

> A colision in a multicast address seems pretty strange to me though, I'm not sure what else could be using a multicast address in our CI machines.

I would rather think it would be an ordinary unicast port. Multicast ones typically have the `SO_REUSEADDR` flag set, and so wouldn't cause the problem.

> At least in one of the other vendors a bind() error is ignored and only a log is printed in that case, maybe that's why we never saw it happening (?).

At least that means it is still a possibility 🙂 You would only notice anything if there are multiple processes that need to communicate (and there are not that many tests that try to do that), otherwise there's nothing to discover and so being unable to receive discovery messages via multicast becomes a non-issue. Moreover, it is possible to some other vendors always do unicast discovery (I don't know, but they might ...) even when multicast is a possibility.

ivanpauno, 2021-03-23:
> I would rather think it would be an ordinary unicast port. 

Ah yes, I forgot that was even possible

> At least that means it is still a possibility :slightly_smiling_face: You would only notice anything if there are multiple processes that need to communicate (and there are not that many tests that try to do that), otherwise there's nothing to discover and so being unable to receive discovery messages via multicast becomes a non-issue

Yes, I thought exactly the same

> Moreover, it is possible to some other vendors always do unicast discovery (I don't know, but they might ...) even when multicast is a possibility.

A little debugging tells me that's exactly the case, so I guess that's why we only see the failure with cyclonedds.

clalancette, 2021-03-31:
As an FYI, today I moved all of our CI infrastructure so that it uses domain IDs that don't collide with the OS ephemeral port ranges.  In particular:

Linux - Made it use domain ID 1 instead of domain ID 108 by default
macOS - Made the domain IDs on the various CI machines < 166
Windows - Already was using a domain ID of 1

I also wrote up https://github.com/ros2/ros2_documentation/pull/1287 , which is an article describing best practices for choosing a domain ID.

*Hopefully* that will fix this set of issues, but only time will tell.  I'll leave this open for now and we can close it if we stop seeing these issues after some time.

clalancette, 2021-04-14:
Since I've made this change, I don't think we've seen any instances of this problem.  So I'm going to go ahead and close this issue for now.  If it pops up again, we can reopen and reinvestigate.

closed 2021-04-14 by clalancette