ros2/rmw_cyclonedds #284: Implement CycloneDDS across different networks

destkk, 2021-02-15:
- Operating System:
Ubuntu 20.04

- DDS implementation:
rmw_implementation used: rmw_cyclonedds

Hi there. May I know what are the methods that I could implement CycloneDDS across different networks? Current method could only be able to establish in the same network for discovery and communication of topics and services.

eboasson, 2021-02-15:
Please see https://github.com/eclipse-cyclonedds/cyclonedds/issues/688 but the short version is:

* multi-network support is being worked on
* you are likely running into a low default limit of the number of Cyclone processes on a machine when not using multicast

MitchGaleaAbyss, 2023-02-26:
Is there an update on this, is this possible?