ros2/rmw_cyclonedds #285: Program termination with Ctrl+C stops in the middle.

hsgwa, 2021-02-17:
## Bug report
- CPU:
  Intel(R) Core(TM) i5-3340M CPU @ 2.70GHz
- Operating System:
  - Ubuntu 20.04
- Installation type:
  -  from source
- Version or commit hash:
  -  https://github.com/ros2/rcl/commit/2883ce4e282d97572b402d4810d10e26acef2064 (master)
  - https://github.com/eclipse-cyclonedds/cyclonedds/commit/314d3fc19f2320bbb6008a562bf657f41ffbbe10 (master)
  - https://github.com/ros2/rmw_cyclonedds/commit/d024823043504ea40af24bf22365a21cd203df55 (master)
- DDS implementation:
  - cyclonedds ONLY. rmw_fastrtps_cpp did not cause.
- Client library (if applicable):
  - rclcpp or rclpy

#### Steps to reproduce issue
This issue occurred in both rclcpp and rclpy.
```
$ ros2 run demo_nodes_cpp talker --ros-args --log-level debug
# send Ctrl+C
```
```
$ ros2 run demo_nodes_py talker --ros-args --log-level debug
# send Ctrl+C
```

#### Expected behavior
terminate safely.

#### Actual behavior
stop during finalizing state.
It seems like node finalizing has some problem.

```
$ ros2 run demo_nodes_cpp talker --ros-args --log-level debug
^C[INFO] [1611636125.400290516] [rclcpp]: signal_handler(signal_value=2)
[DEBUG] [1611636125.400375072] [rclcpp]: signal_handler(): SIGINT received, notifying deferred signal handler
[DEBUG] [1611636125.400436017] [rclcpp]: deferred_signal_handler(): woken up due to SIGINT or uninstall
[DEBUG] [1611636125.400486808] [rclcpp]: deferred_signal_handler(): SIGINT received, shutting down
[DEBUG] [1611636125.400651593] [rclcpp]: deferred_signal_handler(): shutting down rclcpp::Context @ 0x55d466e8e830, because it had shutdown_on_sigint == true
[DEBUG] [1611636125.400687545] [rcl]: Shutting down ROS client library, for context at address: 0x55d466e8eab0
[DEBUG] [1611636125.400782572] [rcl]: Guard condition in wait set is ready
[DEBUG] [1611636125.400830210] [rcl]: Finalizing publisher
[DEBUG] [1611636125.401084234] [rcl]: Timer canceled
[DEBUG] [1611636125.401109610] [rcl]: Timer canceled
[DEBUG] [1611636125.401343531] [rcl]: Publisher finalized
[DEBUG] [1611636125.401495158] [rclcpp]: deferred_signal_handler(): waiting for SIGINT or uninstall
[DEBUG] [1611636125.401520824] [rcl]: Finalizing event
[DEBUG] [1611636125.401541437] [rcl]: Event finalized
[DEBUG] [1611636125.401579052] [rcl]: Finalizing publisher
[DEBUG] [1611636125.401832104] [rcl]: Publisher finalized
[DEBUG] [1611636125.402000585] [rcl]: Finalizing event
[DEBUG] [1611636125.402013763] [rcl]: Event finalized
[DEBUG] [1611636125.402026546] [rcl]: Finalizing subscription
[DEBUG] [1611636125.402194399] [rcl]: Subscription finalized
[DEBUG] [1611636125.402357121] [rcl]: Finalizing service
[DEBUG] [1611636125.404965197] [rcl]: Service finalized
[DEBUG] [1611636125.405047754] [rcl]: Finalizing service
[DEBUG] [1611636125.405435803] [rcl]: Service finalized
[DEBUG] [1611636125.405483505] [rcl]: Finalizing service
[DEBUG] [1611636125.405815296] [rcl]: Service finalized
[DEBUG] [1611636125.405853974] [rcl]: Finalizing service
[DEBUG] [1611636125.406083684] [rcl]: Service finalized
[DEBUG] [1611636125.406120386] [rcl]: Finalizing service
[DEBUG] [1611636125.406316801] [rcl]: Service finalized
[DEBUG] [1611636125.406351774] [rcl]: Finalizing service
[DEBUG] [1611636125.406559456] [rcl]: Service finalized
[DEBUG] [1611636125.406635210] [rcl]: Finalizing event
[DEBUG] [1611636125.406648754] [rcl]: Event finalized
[DEBUG] [1611636125.406664502] [rcl]: Finalizing publisher
[DEBUG] [1611636125.406853173] [rcl]: Publisher finalized
[DEBUG] [1611636125.407209588] [rcl]: Finalizing node
```

```
$ ros2 run demo_nodes_py talker --ros-args --log-level debug
^C[DEBUG] [1611636186.002509447] [rcl]: Guard condition in wait set is ready
[DEBUG] [1611636186.002854722] [rcl]: Finalizing event
[DEBUG] [1611636186.002893463] [rcl]: Event finalized
[DEBUG] [1611636186.002922664] [rcl]: Finalizing publisher
[DEBUG] [1611636186.003449009] [rcl]: Publisher finalized
[DEBUG] [1611636186.003530384] [rcl]: Finalizing event
[DEBUG] [1611636186.003552872] [rcl]: Event finalized
[DEBUG] [1611636186.003578862] [rcl]: Finalizing publisher
[DEBUG] [1611636186.004997261] [rcl]: Publisher finalized
[DEBUG] [1611636186.005102081] [rcl]: Finalizing service
[DEBUG] [1611636186.006543735] [rcl]: Service finalized
[DEBUG] [1611636186.006641986] [rcl]: Finalizing service
[DEBUG] [1611636186.006956990] [rcl]: Service finalized
[DEBUG] [1611636186.007011764] [rcl]: Finalizing service
[DEBUG] [1611636186.008383436] [rcl]: Service finalized
[DEBUG] [1611636186.008476722] [rcl]: Finalizing service
[DEBUG] [1611636186.009434476] [rcl]: Service finalized
[DEBUG] [1611636186.009525846] [rcl]: Finalizing service
[DEBUG] [1611636186.010021857] [rcl]: Service finalized
[DEBUG] [1611636186.010073573] [rcl]: Finalizing service
[DEBUG] [1611636186.011492397] [rcl]: Service finalized
[DEBUG] [1611636186.011578826] [rcl]: Timer canceled
[DEBUG] [1611636186.011635771] [rcl]: Finalizing node
[DEBUG] [1611636186.011656284] [rcl]: Finalizing publisher
[DEBUG] [1611636186.011833613] [rcl]: Publisher finalized
```

ivanpauno, 2021-02-17:
Transferred the issue from `rcl` to `rmw_cyclonedds`.

eboasson, 2021-02-18:
@ivanpauno @hsgwa The way I read it, the problem was solved with changing the firewall settings, and it is not really a bug in Cyclone if it can't receive data because of a firewall. Hanging during shutdown is a consequence of that.

I do want everything to work fine within a machine if at least the loopback interface works, and so I am working on supporting using multiple network interfaces simultaneously.

The interesting options as I see it are:

* If hanging during shutdown because of a firewall is a bigger problem than not communicating because of a firewall, then we can change the default to `Internal/multipleReceiveThread=false` to prevent the hang.
* We could try to autodetect a firewall blocking network traffic on start-up and by default not use network interfaces that are blocked. I don't know exactly how reliable that would be, though.
* We can just wait for Cyclone to use the loopback interface and the ethernet in parallel. That should also fix it.

ivanpauno, 2021-02-18:
> @ivanpauno @hsgwa The way I read it, the problem was solved with changing the firewall settings, and it is not really a bug in Cyclone if it can't receive data because of a firewall. Hanging during shutdown is a consequence of that.

Sorry, I hadn't followed the previous comments too closely, I only confirmed it was a cyclonedds related issue and moved the ticket.
But yes, this seems to only be a firewall configuration problem.

About the other options, using the ethernet and loopback interface in parallel by default seems interesting.

I'm closing as the original question has been answered, @hsgwa feel free to comment/reopen if you think there's something else to be done.

closed 2021-02-18 by ivanpauno