ros2/rmw_cyclonedds #302: How to deal with serialized messages and iceoryx shared memory transport?

budrus, 2021-04-13:
## Feature request

#### Feature description
It seems that the combination of publishing and taking serialized messages and a zero-copy shared memory transport is currently not supported. This should also be possible

#### Implementation considerations
1. Option A
- Sending the serialized message via the Cyclone shared memory transport
- The subscription needs to know whether a serialized or non-serialized message was received
- The subscription just can pass the received serialized message when  `rmw_take_serialized_message()` is used
- The subscription has to do a de-serialization if the normal `rmw_take()` is used
- Cyclone DDS has to know the size of the serialized message when write() is called (maybe different from the topic size)
- One copy when sending the message as we have to copy it into the shared memory
- No copy when the subscription takes a serialized message
- One copy when the subscription takes the non-serialized message as we have to de-serialize it

2. Option B  
- The rmw layer always passes non-serialized messages to Cyclone
- One copy with a de-serialization step for sending the message. This is done when copying the message into the shared memory and is needed anyway as there is no combination of loaned and serialized messages, right?
- No copy when the subscription takes the non-serialized message
- One copy when the subscription takes a serialized message as we have to serialize it again

budrus, 2021-04-13:
@sumanth-nirmal @eboasson @MatthiasKillat Are the listed options correct? Do you see other ones. They seem to have equal costs and it depends on whether the subscriber wants to consume a serialized or non-serialized message. If that's the case Option B is maybe the easier one

sumanth-nirmal, 2021-04-25:
@eboasson @budrus @MatthiasKillat I think the cost for both options are more or less the same (depending on the `take` path). However, with option-A it is tricky to determine if the received message is serialized or de-serialized. So I went ahead and implemented the following (Option-B) in #308, where the loaned memory is always initialized with the de-serialized sample and passed to Cyclone DDS

Publishing serialized messages with SHM enabled
1. Loan the memory chunk from Cyclone DDS
2. De-serialize the message and copy it into loaned memory
3. Pass the loaned memory to Cyclone DDS for publishing

Taking serialized message
1. Serialize the received sample and copy it into the serialized message buffer 
2. Free the loaned memory chunk

Taking message 
1. Return the message from the loaned memory chunk
2. The memory will be eventually freed when `return_loaned_message` is called