ros2/rmw_cyclonedds #316: New cyclonedds release for rolling

ivanpauno, 2021-06-02:
After https://github.com/ros2/rmw_cyclonedds/pull/309, the rolling PR checker is failing, e.g.: https://build.ros2.org/job/Rpr__rmw_cyclonedds__ubuntu_focal_amd64/239/console#console-section-6.

We will need a new `cyclonedds` realease that includes the `dds_data_allocator_init_heap()`.

ivanpauno, 2021-06-02:
@eboasson is it possible to do a new release?

eboasson, 2021-06-02:
Absolutely, I’ll do it when I’m home again. (I went for a bike ride and saw your note when I stopped to take a picture 🙂.)

eboasson, 2021-06-02:
@ivanpauno I just created `0.8.0beta6`.

ivanpauno, 2021-06-02:
Thanks @eboasson !!!
I will do a `bloom-release` with the new tag so it gets updated in `rosdistro`.

> I went for a bike ride and saw your note when I stopped to take a picture

Sorry to disturb the ride, hope you have enjoyed it!!

ivanpauno, 2021-06-03:
I have opened the rosdistro PR: https://github.com/ros/rosdistro/pull/29767

closed 2021-06-03 by ivanpauno