ros2/rmw_cyclonedds #318: Unable to communicate between a FastDDS application and ROS2 using CycloneDDS when ROS_LOCALHOST_ONLY is set to 1

TSC21, 2021-06-07:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - `0.22.2-1focal.20210519.013430`
- DDS implementation:
  - CycloneDDS in the ROS2 side; FastDDS application in the other
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
I am trying to connect a FastDDS participant, in an external application, which without using `ROS_LOCALHOST_ONLY`, is able to connect to a ROS2 domain using CycloneDDS as it's rmw. The FastDDS application whitelists the localhost network and sets up a custom transport using UDP with:

```c++
auto localhostUdpTransport = std::make_shared<UDPv4TransportDescriptor>();
localhostUdpTransport->interfaceWhiteList.emplace_back("127.0.0.1");

// Disable the built-in Transport Layer
PParam.rtps.useBuiltinTransports = false;

// Add the descriptor as a custom user transport
PParam.rtps.userTransports.push_back(localhostUdpTransport);
```

`ROS_LOCALHOST_ONLY=1` is set on the environment, with Galactic using `rmw_cyclonedds_cpp` as its rmw. `ros2 topic` tools are used to try to get information from the topics published from the FastDDS application, which work quite well when not whitelisting the localhost.

#### Expected behavior
Be able to publish and subscribe between a FastDDS app (configured with the same qos and policies used on ROS2) and a ROS2 node using CycloneDDS, where FastDDS whitelists the localhost network and ROS_LOCALHOST_ONLY is set on the ROS2 side (I am not sure what happens in this case for CycloneDDS).

#### Actual behavior
The opposite behavior of the above, i.e., not able to get or send data to/from the FastDDS app.

Note that the above works well when the RMW_IMPLEMENTATION is set to `rmw_fastrtps_cpp`, which means that apparently the same configuration for both rmws results on incompatibilities between the DDS implementations, i.e., two DDS implementation apparently are not able to communicate when the locahost network is whitelisted.

TSC21, 2021-06-07:
Further note about https://github.com/ros2/rmw_cyclonedds/blob/master/rmw_cyclonedds_cpp/src/rmw_node.cpp#L245-L248: I have `127.0.0.1` set to `localhost` in `/etc/hosts`, so this should not be the problem.

eboasson, 2021-06-08:
It looks to me like setting `ROS_LOCALHOST_ONLY=1` causes `rmw_fastrtps_cpp` to use the same configuration you describe (disable built-in transports, whitelist localhost) and so that should be a reasonable approximation of what you are doing. I've tried that and it works as expected for me locally. While that proves it can work, it is also clearly a different experiment because the exact application code, ROS 2 version (slightly newer in my case, as it is a source build), Cyclone DDS version (somewhat newer in my case), Fast-DDS version, RMW versions (Cyclone one is probably .., you get the point), operating system (macOS, in my case) and hardware (Apple M1, in my case) are all different ...

There are two ways one can proceed:
* one is for you try the current version of Cyclone and its RMW layer because the recent changes may have affected the behaviour in this configuration (see https://github.com/ros2/ros2/issues/1131);
* the other is for you to re-run your test with Cyclone tracing enabled so we get a better picture of what is going on. The latter can be done by setting `CYCLONEDDS_URI='<Tr><C>trace</><Out>cdds.log</></>'` and re-running your experiment.

The trace files written by cyclone are somewhat cryptic, but I can help with that.