ros2/rmw_cyclonedds #319: Interfacing ROS2 CycloneDDS with OpenSplice

makolele12, 2021-06-07:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu 20.04
- Installation type:
  source
- Version or commit hash:
  ROS2 Foxy `cylclonedds: 457a07132d394e6461ca23a5c27ddaefbb8ab868`
- DDS implementation:
  CycloneDDS with ROS2 and a native OpenSplice application

#### Steps to reproduce issue
I am trying to communicate a native OpenSplice application with ROS2/CycloneDDS. My first test has been to replicate the "HelloWorld" example provided in the OpenSplice Installation. (The exact same example is also provided with the CycloneDDS installation).

I slightly edited the .idl message in the sample directory such that the topic is keyless (my understanding is that ROS2 only works with keyless topics). I named the topic rt/HelloWorldDataMsg, such that ROS2 is able to see it.

Additionally, I created a ROS2 package containing the same IDL message I used with OpenSplice (but deleted the #pragma statement from OpenSplice), compiled it and built it successfully.

I publish to the mentioned topic with my OpenSplice Application. The publishing is successful as with the "Tester" tool from OpenSplice, I can read the HelloWorld Message.

#### Expected behavior
 I expect to be able to read the HelloWorldMessage on ROS2, however I can't.

#### Actual behavior

When executing `ros2 topic list -t` I get:

> /helloworlddata_msg [helloworlddatakeyless::idl::Msg]

If I execute: `ros2 topic echo /helloworlddata_msg` I get:

> ValueError: Empty Module name

I noticed that the topic type is formatted different as ROS2 does. I guess ROS2 expects something like `/helloworlddatakeyless/idl/Msg`. Therefore, I tried subscribing to the topic by explicitly giving the topic message type:

`ros2 topic echo /helloworlddata_msg /helloworlddatakeyless/idl/Msg `

However after executing this command I don't get any output. The terminal is just blank and ROS2 is waiting for a message to be published. 

I am stuck at this point. What am I omitting in order to be able to read the messages with ROS2? Any help would be deeply appreciated, thanks.

#### Additional information

The .idl I used to create my ROS2 interface:

```
module helloworlddatakeyless
{
module idl{
   struct Msg
   {
      long userid;
      string message;
   };   
};
};
```

The .idl I used to create my OpenSplice application:

```
module helloworlddatakeyless
{
 module idl
{
 struct Msg
   {
      long userid;
      string message;
   };
   #pragma keylist Msg
};
};

```

eboasson, 2021-06-08:
Hi @makolele12, I tried to have a look but I get some inscrutable errors from CMake trying to use a `.idl` files with the ROS 2 build system. I am sure it is because I am doing something wrong.

Could you provide a set of sources for the ROS 2 side that'll allow me reproduce the problem (the OpenSplice side is no issue, I know (almost) all the the tricks there) without having to figure out how to translate your description into a compiling ROS 2 program?

makolele12, 2021-06-09:
Hi @eboasson thanks a lot for your answer. Below is the zip file containing my small ROS2 package. You just need to extract it in your src folder in your current ROS2 environment and build it with `colcon build --packages-select helloworlddatakeyless`

[helloworlddatakeyless.zip](https://github.com/ros2/rmw_cyclonedds/files/6621924/helloworlddatakeyless.zip)

Afterwards, if you run `ros2 interface show helloworlddatakeyless/idl/Msg` you should get:

```
module helloworlddatakeyless
{
module idl{
   struct Msg
   {
      long userid;
      string message;
   };   

};
};
```
As I mentioned before, I am doing this test locally in the same machine. I start the OpenSplice Launcher and choose `ospl_shmem.xml` as configuration.

Afterwards I start publishing to the rt/helloworlddata_msg topic. Using the OpenSplice Tester I can create a Default Reader and see the messages being posted.

[image]

However, (as I mentioned before) executing `ros2 topic echo /helloworlddata_msg /helloworlddatakeyless/idl/Msg` does not print any message. It is worth mentioning I am using the default .xml configuration provided by CycloneDDS.

Is this a topic type mismatch? Or maybe a problem with the used .IDL files on both sides? Or maybe a config problem between OpenSplice and CycloneDDS? I am kind of lost here.

eboasson, 2021-06-09:
This is all quite interesting to me, I'm learning a bit more about the dark corners of ROS 2 ...

Anyway, this:
```
ros2 interface show helloworlddatakeyless/idl/Msg
```
works in Foxy but not in (a source build that is roughly) Galactic where it fails with:
```
# ros2_ws ros2 interface show helloworlddatakeyless/idl/Msg
Error processing 'module helloworlddatakeyless' of 'helloworlddatakeyless/Msg': ''module' is an invalid message name. It should have the pattern '^[A-Z][A-Za-z0-9]*$''
Traceback (most recent call last):
  File "/Users/erik/ros2_ws/install/ros2cli/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.14.0', 'console_scripts', 'ros2')())
  File "/Users/erik/ros2_ws/install/ros2cli/lib/python3.9/site-packages/ros2cli/cli.py", line 67, in main
    rc = extension.main(parser=parser, args=args)
  File "/Users/erik/ros2_ws/install/ros2interface/lib/python3.9/site-packages/ros2interface/command/interface.py", line 35, in main
    return extension.main(args=args)
  File "/Users/erik/ros2_ws/install/ros2interface/lib/python3.9/site-packages/ros2interface/verb/show.py", line 201, in main
    _show_interface(
  File "/Users/erik/ros2_ws/install/ros2interface/lib/python3.9/site-packages/ros2interface/verb/show.py", line 148, in _show_interface
    for line in _get_interface_lines(interface_identifier):
  File "/Users/erik/ros2_ws/install/ros2interface/lib/python3.9/site-packages/ros2interface/verb/show.py", line 116, in _get_interface_lines
    yield InterfaceTextLine(
  File "/Users/erik/ros2_ws/install/ros2interface/lib/python3.9/site-packages/ros2interface/verb/show.py", line 43, in __init__
    msg_spec = parse_message_string(
  File "/Users/erik/ros2_ws/install/rosidl_adapter/lib/python3.9/site-packages/rosidl_adapter/parser.py", line 516, in parse_message_string
    Type(type_string, context_package_name=pkg_name),
  File "/Users/erik/ros2_ws/install/rosidl_adapter/lib/python3.9/site-packages/rosidl_adapter/parser.py", line 276, in __init__
    super(Type, self).__init__(
  File "/Users/erik/ros2_ws/install/rosidl_adapter/lib/python3.9/site-packages/rosidl_adapter/parser.py", line 204, in __init__
    raise InvalidResourceName(
rosidl_adapter.parser.InvalidResourceName: 'module' is an invalid message name. It should have the pattern '^[A-Z][A-Za-z0-9]*$'
```

On Foxy, I needed to change the type name in the `ros2 topic echo` command to be `helloworlddatakeyless/idl/Msg` (i.e., without the initial slash), else I also got an error. No worries if it was just a typo on your side, I'm just letting you know because these are potentially things that someone may want to look into.

With that, I can make it work if I change the IDL on the DDS side to:
```
module helloworlddatakeyless {
 module idl {
  module dds_ {
   struct Msg_ {
    long userid;
    string message;
   };
   #pragma keylist Msg_
}; }; };
```
that is, the name mangling by ROS 2 inserts a `dds_` after the modules, and it appends an underscore. There is an option in ROS 2 somewhere to disable the ROS name mangling, but I don't see that in `ros2 echo topic -h`.

Does that get you any further?

makolele12, 2021-06-09:
The initial slash you mention was in fact a typo while I wrote the post.

Modifying the IDL message on the DDS Side worked perfectly, I can read the message. Also if I run `ros2 topic list -t` I get to see the topic type the way ROS2 formats it:

`/helloworldata_msg [helloworlddatakeyless/idl/Msg]`

 Is this ROS2 behavior anywhere documented? If it does I overlooked it. Thanks for the help.

eboasson, 2021-06-09:
> Is this ROS2 behavior anywhere documented? If it does I overlooked it

I don't know of any design document that covers it — indeed, I'd have liked to have one, too. The one I know of that covers similar matter is https://design.ros2.org/articles/topic_and_service_names.html, but it doesn't seem to cover type names.

I had to add a line of output to the `ddsls` tool in the Cyclone python binding (see https://github.com/eclipse-cyclonedds/cyclonedds-python/pull/12), but with that in place
```
# python3 ./cyclonedds/tools/ddsls.py -w -t dcpssubscription
...
--- New SUBSCRIPTION ------------
SUBSCRIPTION
 key:d0631001-9670-9e73-0228-b3cc00000804
 participant_key:d0631001-9670-9e73-0228-b3cc000001c1
 topic_name:rt/helloworlddata_msg
 type_name:helloworlddatakeyless::idl::dds_::Msg_
 qos:
  ('Deadline', {'deadline': 9223372036854775807})
  ('DestinationOrder', {'kind': 'ByReceptionTimestamp'})
  ('Durability', {'kind': 'Volatile'})
  ('History', {'kind': 'KeepLast', 'depth': 5})
  ('IgnoreLocal', {'kind': 'Nothing'})
  ('LatencyBudget', {'budget': 0})
  ('Liveliness', {'kind': 'Automatic', 'lease_duration': 9223372036854775807})
  ('Ownership', {'kind': 'Shared'})
  ('PresentationAccessScope', {'kind': 'Instance', 'coherent_access': False, 'ordered_access': False})
  ('ReaderDataLifecycle', {'autopurge_nowriter_samples_delay': 9223372036854775807, 'autopurge_disposed_samples_delay': 9223372036854775807})
  ('Reliability', {'kind': 'BestEffort', 'max_blocking_time': 0})
  ('ResourceLimits', {'max_samples': -1, 'max_instances': -1, 'max_samples_per_instance': -1})
  ('TimeBasedFilter', {'filter_time': 0})
  ('TransportPriority', {'priority': 0})
```
it is quite easy to find out what ROS 2 is actually doing.